SUSTAINA-OP/SUSTAINA-OP #80: サーボの穴のズレについて

hiroki-0001, 2021-06-20:
サーボを固定するネジ穴が一部ずれている。
[image]

CAD上ではあっているため、穴の位置関係を実測で確認し修正する必要がある。

MasatoKubotera, 2021-06-20:
メーカHPページからダウンロードした寸法は，inventorCADと同一であった．
実測で変更する前に，メーカ側に問い合わせをする必要があると考える．

KiyoshiroKawanabe, 2021-08-24:
[image]

【近藤さんの方に、「赤丸のついている穴がずれているのでは？」と問い合わせたところ、お電話いただきお話しましたので報告します。】

アッパーケース(画像あり)とミドルケースがビスで止められており、それらは別々に製造されている。そのためアッパーケースとミドルケースを重ねたときに穴がずれていないのであれば、製品に問題はない。
と、回答頂きました。
[image]
↑アッパーケース

hiroki-0001, 2021-09-17:
こちらの調査方法として、当面は以下の流れをで進めていきたいと思います。

1. 現在使用しているサーボモータのverをすべて調べる
2. 調査した各verの寸法をメーカ側から出している寸法図と照らし合わせチェックする

NaokiIwasawa, 2021-09-26:
# 調査結果
サーボのverに関しては @hiroki-0001 が調査したので詳細はお願いしたいが、自分と @MasatoKubotera @hiroki-0001 が実測による調査を数個で実施したところ重大な問題が発覚したので調査途中であるが報告する。

## 調査方法
近藤科学のサイトよりB3Mの[参考寸法図](https://kondo-robot.com/w/wp-content/uploads/B3M-11xx-Dimensions.pdf)を見て調査用器具を作成する。作成した器具のスケッチと寸法図は画像の通りである。
[image]

器具を切削しサーボの固定穴と重ねて図面とズレが無いか確認する。器具の精度を指摘される恐れがあるのでサーボの両面で確認は行い穴のサイズはφ2.6で器具を2つ切削した。そして一つはネジを通すためにわずかに穴をやすりで削り他方はネジが通らないがそのままの状態にしてある。画像や動画では削ったものを使用。

## 結果
[image]写真はフリーホーン側で反対側から照らしている

[確認中の様子（動画）](https://drive.google.com/open?id=1LQRHWVi6raedHLDxUVB2RoHPypjFKs0H&authuser=s20c1016ct%40s.chibakoudai.jp&usp=drive_fs)

確認したサーボモータは6個程度だがそのほとんどでズレが目視確認できた。ただし、いずれも器具とのズレがあったのはサーボモータのフリーホーン側であった。

## 考察
今回の調査は時間が限られていたため検査数が十分とは言えないが新品・ver違いも含めて確認したため相当数の個体で出力ホーンとフリーホーン側での穴ズレ問題が生じていると考えられる。被締結部材の穴径をある程度大きくすれば何とかなりそうではあるが、再設計を強いられるため対策は検討中である。

NaokiIwasawa, 2021-10-01:
# 今後の方針
現在あるサーボ全てで調査を行い修理するのは現実的でないので以下の通り対応をする
## すでにある部品
再設計・再切削をするリソースはないので、問題があればやすりで削って対応する。
## ver4_0
設計の段階で問題があった穴は長穴にするなどで対応をする。ただし、ミラーパーツが増えないようにする。
穴ズレの程度は個体によって差がある為大きい方に合わせるが、目測でおおよそ0.5mm～0.8mm程度は余裕を持った方が良さそうである。

yasuohayashibara, 2021-10-03:
追加工で対応するので良いと思いますが，サーボモータの穴の精度次第では，組み合わせに依存関係が生まれます．
依存関係がある場合は，組立のときに部品を入れ替えないように徹底したほうが良いと思います．
あと，サーボモータ交換時にもしかしたらさらに追加工が必要になるかもしれません．
（それを知らない人が組み立てると無茶をしてネジ穴をだめにするおそれもあります．）
そこまでずれてはいないと思いますが，念のため指摘しておきます．

MasatoKubotera, 2021-10-03:
# Ver4.0での穴ずれ対策

上記のコメント
>設計の段階で問題があった穴は長穴にするなどで対応をする。ただし、ミラーパーツが増えないようにする。
穴ズレの程度は個体によって差がある為大きい方に合わせるが、目測でおおよそ0.5mm～0.8mm程度は余裕を持った方が良さそうである。

にもある通り，該当箇所について長穴として設計する．
穴ずれにも個体差があり，やすりで削ることによる対応も可能であるが，167機製作時に切削部品や既製品での対応に時間を要しているため，設計段階で対応するものとする．

設計変更となったのは，下図に示す赤色で着色された，7種類19個の部品である．
現状，長穴のずれを**最大1.0mmと仮定して**モデルを作成している．
本格的に切削に入る前に，最大の穴ずれが何mmであるか検証する必要がある．
@NaokiIwasawa 

---
[image]

[image]「Trunk_chest_horizontal_frame.ipt」

[image]「Trunk_chest_side_frame.ipt」

[image]「Trunk_upper_waist_frame.ipt」

[image]「Leg_joint_roll_frame.ipt」

[image]「Knee_side_frame.ipt」

[image]「Ankle_side_frame.ipt」

[image]「Ankle_middle_frame.ipt」

yasuohayashibara, 2021-10-06:
> 本格的に切削に入る前に，最大の穴ずれが何mmであるか検証する必要がある．

位置決めに使用していないネジは大丈夫ですが，ネジで位置決めをしている場合は穴を大きくしすぎると衝撃力で位置がずれることになりますので，お気をつけ下さい．
（そもそもネジ穴をできるだけ位置決めに使用しない設計にしたほうが良いといえば良いのですが．．．）

MasatoKubotera, 2021-10-19:
対応が終了したためcloseします。

closed 2021-10-19 by MasatoKubotera