SUSTAINA-OP/SUSTAINA-OP #82: 間にPOMなどのパーツが挟まれている部位の適正トルクの算出

hiroki-0001, 2021-06-20:
165・163の整備中に間にPOMなどのパーツが挟まれている部位の締め付けトルクが適正でないため、以下の写真のように間のパーツに余計な負荷を与えてしまっている。

[image]
[足付け根ロールの一部　ネジでの締め付けが強すぎて間のスペーサの形が歪になっている]

このままではパーツの寿命を早めてしまったり、余計な負荷を与えてしまうため、適切な締め付けトルクの算出をおこないたい。

MasatoKubotera, 2021-06-20:
適正トルクの算出には非締結部の部材厚や素材による．
締め付けの適切なトルクの算出も必要な事項ではあるが，TPUや上記のパーツでは構造的欠陥が見られる．
非締結部品が締結に対し弱すぎると締め付けトルクが低くなりすぎる恐れがあるため，部品の再設計や締結方法の見直しが必要であると考えられる．
[image]

MasatoKubotera, 2021-09-15:
上記で指摘されている，間にPOMなどのパーツがは挟まれている部位についてver4.0では廃止されました．
そのため，議論するべきは非締結部品がTPUであるRearBumper部分のみとなりました．

MasatoKubotera, 2021-12-12:
トルクの算出よりも，設計を見直すことが大前提になるため，このissueは一旦closeします．

closed 2021-12-12 by MasatoKubotera