SUSTAINA-OP/SUSTAINA-OP #83: 各パーツのフィレット加工

hiroki-0001, 2021-06-20:
胴体_サブフレーム_1_4とそのMIRパーツにはフィレット加工されていない、怪我を防ぐため修正しておく。

[image] 

<br/>
このようにフィレット加工されていないパーツが多くあるため、それらの調査、修正を行う必要がある

MasatoKubotera, 2021-06-20:
フィレット加工のR寸法を決める必要性がある．
また，外側だけではなく内側のフィレット加工も修正する必要性がある．

FUKUCHIGarbaldigus, 2021-08-30:
胴体部の調査
メインフレーム1-1,1-12,1-13
サブフレーム1-1,1-2,1-3,1-4,1-6,1-7,1-9,1-11,1-12
にR加工が不足

FUKUCHIGarbaldigus, 2021-08-30:
コックピットの調査
メインフレーム1,2
サブフレーム1,2,3,4,5,6,7,8

FUKUCHIGarbaldigus, 2021-08-30:
頭の調査
頭メインフレーム1,2
頭サブフレーム1,2
カメラメインフレーム1,2

FUKUCHIGarbaldigus, 2021-08-30:
腕の調査
肩メイン1-1
肩サブ1-1
上腕メイン1-1
前腕サブ1-1

FUKUCHIGarbaldigus, 2021-08-30:
上半身はほぼすべてにR加工が不足しているようです。どこまで厳しく見るかというのはありますが...

MasatoKubotera, 2021-09-02:
@FUKUCHIGarbaldigus 
部品名は省略せずに正式名称で記載してください．

FUKUCHIGarbaldigus, 2021-09-02:
R寸法基準を仮にR2として修正を入れていきたいと思います。

MasatoKubotera, 2021-09-02:
@FUKUCHIGarbaldigus 
現段階では，修正はできないので注意してください．

hiroki-0001, 2021-09-02:
@FUKUCHIGarbaldigus 
R寸法を基準としてR2にするのはいいと思いますがGithubでバイナリファイルを使用する都合上、各自がバラバラになって作業すると、[このような事件が起きて](https://github.com/citbrains/OpenPlatform_ver3_0/issues/58#issuecomment-868473265)再度作業しなければならず、二度手間になってしまうため、気を付けてください。
またパーツの再設計等が終わっていないので、今回はフィレット加工の基準をR2とした場合、R加工が不足しているところの調査をしていただけると助かります。
久保寺君が指摘してくれていますが、あとから部品の検索がかけられるように、対象パーツは正式名称で記載してください。

FUKUCHIGarbaldigus, 2021-09-06:
9月6日2時時点でのフィレット加工不足パーツ一覧です(.ipt拡張子省略)
- 上半身
    - 胴体
        - 胴体_メインフレーム_1_1
        - 胴体_サブフレーム_1_1
        - 胴体_サブフレーム_1_3
        - 胴体_サブフレーム_1_4
        - 胴体_サブフレーム_1_12
    - 頭
        - 頭_メインフレーム
- 下半身
    - 足首
        - 足首_メインフレーム_1_1
    - リンク_6_1
        - リンク_メインフレーム_6_1

FUKUCHIGarbaldigus, 2021-09-13:
9月13日15:23分現在のフィレット加工不足パーツ一覧です(.ipt拡張子省略)
- Head
  - Head_vertical_frame_01
- Ankle
  - Ankle_side_frame:1

MasatoKubotera, 2021-09-13:
上記指摘部分に関しては対応しました．

MasatoKubotera, 2021-09-17:
ミーティングでの共有が終了したのでcloseします．

closed 2021-09-17 by MasatoKubotera