SUSTAINA-OP/SUSTAINA-OP #84: コックピット容量の拡大

MasatoKubotera, 2021-06-20:
現在，頭サーボのフリーホーン軸側の凸部によりコックピットスペースが小さくなっている．
カメラの軸，位置を動かさずともサーボを上方にあげることでコックピットスペースが確保できる．

MasatoKubotera, 2021-07-08:
関連性があるissue：[胴体_サブフレーム_1_9の再設計 #89](https://github.com/citbrains/OpenPlatform_ver3_0/issues/89)

KentaTsukino, 2021-09-05:
今回作ったものです
大まかなはめ合わせなどを形にしただけなので、細かい見た目に関しては後から修正します
全体
[image]

正面
[image]

胴体_サブフレーム_1_9:2
[image]

胴体_サブフレーム_1_11:1
[image]

NaokiIwasawa, 2021-09-05:
再設計お疲れ様です
いくつか気になった部分があるので下記にまとめました。（詳細はまだ取り掛かってないようなので大きな点）

1. 胴体_サブフレーム_1_11について、上部と下部で異なる形状の物を利用すつもりでしょうか？
全体と正面の画像を見る限り上部は現行を、下部には新たに再設計したものを使用しているように見受けられますが、出来れば部品点数を減らしたいので上下で共通の設計にしていただけると嬉しいです。
2. 肉抜き部分について
CADで計測したところ胴体_サブフレーム_1_9の十字部は1.4mm程度と他の部分(2~mm)と比べ細い気がします。
3. はめ合いの箇所について（1と内容はかぶりますが）
上部は前方側、下部は中央のように違うのが気になります。軽量化の観点で首の付けに近い部分にのみ柱を設けたのなら現行のモデル同様に前方のみでも良いかもしれませんが。
4. 胴体_サブフレーム_1_9の上部はめ合い形状について
一部のみ延長した形状なのでエンドミルの逃げは最低限に修正お願いします。

自分がパッと見て気になった大きな点は以上です。詳細（エンドミルの逃げ・フィレット等）については後程修正お願いします。

KentaTsukino, 2021-09-06:
岩澤先輩のコメントを改善したものです。
胴体_サブフレーム_1_11:1を上下とも同じものを使い対応しました。その結果胴体_サブフレーム_1_9の形が変わりました。
まだ細かい修正は行えていません。

胴体_サブフレーム_1_9
[image]

Inventorのデータです
胴体_サブフレーム_1_9:2
https://drive.google.com/file/d/1h-0kyBWetEQ0-hqQIrJ4mifKrHv0qmi6/view?usp=sharing

KentaTsukino, 2021-09-08:
自分的に完成形になります。
胴体_サブフレーム_1_9
[image]
胴体_サブフレーム_1_1の形に関しては上にあるものです。ですが胴体_サブフレーム_1_9とのはめ合わせの部分の幅が1.5mmで、エンドミルが通らないので部品全体の厚さを1.5mmから2mmに変更しました。
Inventorのデータです
胴体_サブフレーム_1_9:2
https://drive.google.com/file/d/1EVkkn6f5n6gk1mVwk10B2kjZLPlJAIHm/view?usp=sharing

KentaTsukino, 2021-09-09:
名前に関しては統合するときに変えてください。
[image]
https://drive.google.com/file/d/1EVkkn6f5n6gk1mVwk10B2kjZLPlJAIHm/view?usp=sharing

[image]

https://drive.google.com/file/d/17b-cB2b-ZXfdNXtiZvWHr9TGDfHGuk2R/view?usp=sharing

[image]

https://drive.google.com/file/d/1W-Wd3dcVc9HLwkeHbTJQ9V0KMzNCoNSg/view?usp=sharing

MasatoKubotera, 2021-09-09:
@KentaTsukino 
コックピットの容量拡大に取り組んでいただきありがとうございます．
上記Googleドライブで共有されたモデルについて，エンドミルの逃げ加工やフィレットなどミスを修正し，ver4_0ブランチに反映しました．
確認お願いします．

MasatoKubotera, 2021-09-11:
上記で共有されたモデルから横幅についてもさらに改良を加えた．

最終的な大きさは，下記の通りである．
上部空間：横幅75mm(+9mm)，高さ71mm(+8.75mm)
下部空間：横幅46mm(+6mm)，高さ37mm(+0mm)

下図は左:：ver4.0，右：167機の比較である．
[image]「コックピット容量の比較　」

変更に当たって，肩・腰のサーボ配置に関しては触れていない．首サーボについては上方に移動しているが，カメラ位置が変動しないように首フレームを短くすることなどで対応している．

横幅の拡張のために，肩と胴体間の隙間が片側-4.5mm狭くなっている．これにより，現状の胸バンパーについては再利用が出来ない状況となった．この問題については，[FreundBumper_standの再設計 #150](https://github.com/citbrains/OpenPlatform_ver3_0/issues/150)で対処する．

この変更により，改善されたのはコックピット容量だけではない．
Trunk_vertical_frame_02(旧名：胴体_サブフレーム_1_2)の厚みが8mmから4.75mmの2枚重ねに変更になったことで，使用するエンドミルが異なるPOMT8の廃止に貢献している．
[image]

また，胴体を接続する六角スペーサは，167機では[ALSFB6-66](https://jp.misumi-ec.com/vona2/detail/110300206620/?PNSearch=ALSFB6-66&HissuCode=ALSFB6-66&searchFlow=suggest2products&Keyword=ALSFB6-66&list=SuggestPreview)：￥5357/10本を使用していたが，Ver4.0では[ASU-375H](https://www.hirosugi-net.co.jp/shop/g/g7617/)￥1958/10本までにコストカットできている．

[image]

コックピットについては，ver4.0機体において現存のものでも挿入できるため動作確認などは行える．

MasatoKubotera, 2021-09-11:
上記のモデルについては，すでにver4.0ブランチに統合されている．
このissueについては，共有ができ次第closeしたいと考える．

MasatoKubotera, 2021-09-17:
ミーティングでの共有が終了したのでcloseします．

MasatoKubotera, 2021-09-26:
上記コメントで

>Trunk_vertical_frame_02(旧名：胴体_サブフレーム_1_2)の厚みが8mmから4.75mmの2枚重ねに変更になったことで，使用するエンドミルが異なるPOMT8の廃止に貢献している．
[image]「Trunk_vertical_frame_02」

としたが，T5部材から0.25mmだけポケット加工をする必要があり，CAMでのパス作成時に忘れる恐れがあり不向きであると考えた．
そのため，下部のスペース幅を45mm(-0.5mm)とすることでT5部材厚をそのまま使用できるようにした．

[image]
[image]

MasatoKubotera, 2021-09-27:
下図の青色で選択された二つのねじで干渉が見つかった．
[image]

MasatoKubotera, 2021-10-04:
干渉しないように形状を変更した．
[image]
[image]

closed 2021-10-12 by MasatoKubotera