SUSTAINA-OP/SUSTAINA-OP #85: 頭バンパーの必要性

MasatoKubotera, 2021-06-20:
胸にバンパーが搭載されているが，前方に倒れた際にカメラの外装が床面に接触することが見受けられる．
そのため，頭にバンパー的存在が必要なのではないだろうか．
カメラ外装は，PLA素材で製作されているが，TPU素材で作成することも視野に入れて考えたい．

ちなみに，頭サーボの故障率は低いようだ．
また，カメラの故障率も調査する必要がありそうだ．

yasuohayashibara, 2021-06-20:
以下の資料にありますが，前の世代のロボットには首に衝撃吸収のメカニズムがありましたが，試合中に動くことがあり，画像処理が影響を受けるということで最終的に固定されました．
（少し変形するだけでも画像処理が大きく影響を受けます．）
あと，カメラの前にバンパを付ける案も時々でますが，視野を塞がないように取り付ける方法が見つからず現在に至っています．
念のためお伝えしておきます．

https://docs.google.com/a/p.chibakoudai.jp/viewer?a=v&pid=sites&srcid=cC5jaGliYWtvdWRhaS5qcHxjaXQtYnJhaW5zfGd4OjEyZmYyM2Y5Mjk5MDk1MjU

MasatoKubotera, 2021-08-15:
[カメラ_外装の再設計 #67](https://github.com/citbrains/OpenPlatform_ver3_0/issues/67)のissueで試作を含めて制作することになりました．

そのため，この提案issueは閉じます．

closed 2021-08-15 by MasatoKubotera