SUSTAINA-OP/SUSTAINA-OP #86: 結束バンドの固定箇所確保

MasatoKubotera, 2021-06-20:
胴体などでは，サーボケーブルの結束バンドを固定する箇所がなく腕や他機体に干渉する恐れがあるため，固定箇所を確保する必要がある．

Keita1Mori, 2021-08-01:
# 報告
## 164機体右足付け根のサーボケーブル
昨日の模擬試合後に164機体の右足の付け根から下のサーボがうまく起動していなかったため調べたところサーボケーブルの付け方に問題があることが分かった。（サーボケーブルの根本が曲がりすぎていて負荷がかかっている）

- 起動時(hr)
[image]
[image]

- 取り付け部
[image]
[image]

MasatoKubotera, 2021-10-03:
この対処については，時間的都合から1機体目切削時までの再設計で行わない方針となった．
そのた，このissueについては保留とする．

MasatoKubotera, 2021-11-19:
3Dプリンタパーツででケーブルガイドを作っている例です．
https://twitter.com/shiba_8ro/status/1461486568615686149?s=20

[image]

MasatoKubotera, 2022-03-07:
Ankle_side_frameに結束バンド締結用の凹凸を作成しました．

[image]
[image]

MasatoKubotera, 2022-03-08:
Knee_side_frameに結束バンド締結用の凹凸を作成しました．

設計変更後の足首ユニット
[image]

実ロボットの様子
[image]

closed 2022-03-16 by MasatoKubotera