SUSTAINA-OP/SUSTAINA-OP #87: 治具の設計

MasatoKubotera, 2021-06-20:


MasatoKubotera, 2021-07-08:
側面タップ加工をするときに，治具の作成が必要となる．
167機製作の際に，治具の設計/作成を忘れていた過去があったため立ち上げた．

作成する際は，できる限り少ない治具で行えるように作るべきである．
部品ごとに専用を用意してたが共通化できるものも多々ある．

167機に作成した治具に関しては，Google ドライブの[210226_OpenPlatform_ver3_0.zip](https://drive.google.com/file/d/1q-2IgAKyCoR74i8XxF0yLV_I0pqHaSqT/view?usp=sharing)で保管している．
ver4.0の治具に関しては，データの分散を防ぐためにGitHubで管理したいと考えている．

Keita1Mori, 2021-09-08:
# 治具の設計について

## [治具の設計の仕方](https://docs.google.com/presentation/d/1KkJ71DvfgqWCIM54uNBP45d5g5OOJJgyyO5Z-IPVUew/edit#slide=id.geebdde1293_1_64)

## 報告
- 胴体_メインフレーム_1_13と胴体_メインフレーム_1_12を派生した際にスケッチのボタンが押せなかった。

Keita1Mori, 2021-09-10:
# 報告
- 治具の検証が終わりました。
[image]

Keita1Mori, 2021-09-21:
全ての治具をcommitとしました。

Keita1Mori, 2021-10-14:
治具の更新してpushしました。

Keita1Mori, 2021-10-17:
@MasatoKubotera 
了解です。

Keita1Mori, 2021-11-12:
cockpit upper side frameの設計に問題あり

MasatoKubotera, 2021-12-12:
ver4.0での作業が終了したため，closeします．

closed 2021-12-12 by MasatoKubotera