SUSTAINA-OP/SUSTAINA-OP #88: 胴体サブフレーム1_4の修正

hiroki-0001, 2021-06-22:
コックピットの再設計により、リアバンパーの取り付け位置が異なるのでそれに対応させる。

[image]

hiroki-0001, 2021-07-23:
リアバンパーを固定する際のナットを廃止し、カレイナットを使用するようにする

MasatoKubotera, 2021-09-15:
胴体サブフレーム1_4について，英名化によりTrunk_vertical_frame_01に名称が変更されている．

MasatoKubotera, 2021-09-15:
Trunk_vertical_frame_01（旧名：胴体サブフレーム1_4）について下記の項目を変更している．
- 埋め込みナット（カレイナット）に対応．[埋め込みナットの見直し #16](https://github.com/citbrains/OpenPlatform_ver3_0/issues/16#issuecomment-916316693
)
- コックピット容量を確保するために全体的にポケットしていた部分を廃止．切削時間の短縮にもつながる．[コックピット容量の拡大 #84](https://github.com/citbrains/OpenPlatform_ver3_0/issues/84)
- 胴体フレームを繋ぐスペーサの面幅変更による六角切り抜き寸法と間隔変更．[コックピット容量の拡大 #84](https://github.com/citbrains/OpenPlatform_ver3_0/issues/84)
- 胴体フレームから派生しスケッチを作成．

下図はVer3.0とVer4.0の比較である．

Ver3.0
[image]

[image]「胴体サブフレーム1_4」

Ver4.0
[image]

[image]「Trunk_vertical_frame_01」

[image]「Trunk_vertical_frame_01_MIR」

MasatoKubotera, 2021-09-17:
ミーティングでの共有が終了したのでcloseします．

closed 2021-09-17 by MasatoKubotera