SUSTAINA-OP/SUSTAINA-OP #90: パーツを再設計する時の適切な設計手順の検討

AD58-3104, 2021-06-23:
再設計を行うべきパーツが沢山出てきたが、それらの再設計時の方針について決める。
・パーツの有限要素解析の方法
・重くなる事を許容するのかどうか。許容する場合はどのような基準で行うのか

AD58-3104, 2021-06-23:
個人的な意見としてはパーツの有限要素解析はパーツ単体で行うべきと考える。理想としてはアセンブリをしっかり行った後にそのパーツにどのような力が掛かるのかも含めて解析が行えると良いが、現状のモデルのアセンブリ状況などを鑑みるとそれは難しいからである。

yasuohayashibara, 2021-06-27:
どこかで述べましたが，重くすることは負けにつながりますのでおすすめしません．
もしこのチームに経験者が豊富にいて，最小限のモータの負荷でロボットを運用できるとしたら，重くしてもあまりダメージがないかもしれませんが，現状では多く動かして経験値を上げていくというスタイルですので，モータの劣化を誘発しで，日頃の練習試合もままならなくなるおそれがあります．

ちなみに，優勝した2014,15年には首関節と肘関節を取り除くなどして500g軽量化して，ロボットが長時間安定して動くようになり，それにより練習試合を多数行うことで，優勝までたどり着きました．
その後，起き上がりの速度を上げるということで肘関節などを追加しましたが，長時間安定して動かせなくなり，結果を見る限りは良い効果を生んでいません．

以下に，ロボカップ関連の発表がありますが，2012から14年頃まで，バックラッシやモータ温度などの研究発表が並んでいます．今はみなさん感じたことが無いかもしれませんが，本体の重量を減らす前はこれらが一番の問題でした．バックラッシは，長穴減速機の効果が大きいかもしれませんが，重量が重いときには，劣化してくると人が紐で上に持ち上げながらでないと試合ができないという事態が生じていました．
今では考えられないことかもしれませんが，長い時間をかけてチームで克服してきたことですので，重量を増やすことはくれぐれも慎重にお願いします．

https://sites.google.com/a/p.chibakoudai.jp/cit-brains/home/research

なお，上位のチームのほとんどが５年以上同じメンバーで試合をしていたりしますので，モータに負荷をかける設計をしても運用でカバーしていたりしますが，まねをしないほうが良いと思います．
それらのチームも後半は動くロボットが半分以下になっていることが多いので，後半で本チームが逆転することが多いです．
（過去のビデオをご参照ください．）

ただし，軽量化のために無理な設計はしないほうが良いと思います．
同じ結果を得るためにどれだけ楽ができるかが良い設計の指標となると思います．

closed 2022-02-07 by MasatoKubotera