SUSTAINA-OP/SUSTAINA-OP #93: リポジトリ整理

MasatoKubotera, 2021-06-29:
使用していないファイルが多数ある．例として現状使用していない既製品のiptファイルなどがあるが，使用していないファイルについては削除しできるだけ軽いリポジトリにしたい．

また，アセンブリのファイル名についても変更したいと考えている．

yasuohayashibara, 2021-06-30:
ファイル名変更のときにできれば英語名にしていただければと思います．
日本語名だと海外のチームが採用しません．
前からの課題ですが，まだ解決していませんので．

MasatoKubotera, 2021-07-02:
ファイル名を英名にすることは，現在整備を行っている環境下中では難しいです．
問題がある部品の指摘やissueなどでは日本語で記載されてしまっているため，整備完了後の春休みや世界大会前の課題とさせていただきたいです．

新規に設計しているコックピットについては，全て英名にしています．

MasatoKubotera, 2021-09-08:
ver4_0ブランチのコミット[b402649c6d](https://github.com/citbrains/OpenPlatform_ver3_0/tree/b402649c6d9b53a47f5a53123605a81bb85c88b4)時点で本体.iamに使用されてないファイルに関しては，既製品が格納されたフォルダ，上半身下半身のフォルダからは消去されている．
過去を参照するときはコミット履歴を遡る必要がある．

MasatoKubotera, 2021-09-09:
不要なファイルの抽出は難しかったため，Pack and Goを用いて使用しているファイルだけを抽出し，抽出後と抽出前を置換することで対応した．
参照：[Pack and Go によるファイルのパッケージ化の概要](https://knowledge.autodesk.com/ja/support/inventor/learn-explore/caas/CloudHelp/cloudhelp/2021/JPN/Inventor-Help/files/GUID-018371A9-B60D-44CB-B70C-8618155CC598-htm.html)

MasatoKubotera, 2021-09-09:
英名への変更
現状で下記のファイル・フォルダについては名前を変更している．

本体.iam→Body.aim
上半身.iam→UpperBody.iam
下半身_6.iam→LowerBody.iam
下半身_6_MIR.iam→LowerBody_MIR.iam
足裏.iam→Foot_MIR.iam
足裏_MIR.iam→Foot.iam
足裏.ipt→Sole.ipt
\既製品→\Ready_Made

既製品フォルダに関しては，以前フォルダ内にさらに「新規購入」というフォルダが存在したが，使用既製品のみに整理した際に廃止している．

MasatoKubotera, 2021-11-09:
>旧名称との対応表を作成したい方針である

についてはタスク量的に行えないため，このissueについてはcloseする．

closed 2021-11-09 by MasatoKubotera