SUSTAINA-OP/SUSTAINA-OP #96: 脚軸のオイレス 部品の導入

MasatoKubotera, 2021-07-02:
現在，脚のピッチ軸などでは摩擦を減らすためにリンクと関節間のブッシュの両端にシリコングリスが塗られているがこれらのブッシュをオイレス部品に変更ができるか調査する．

MasatoKubotera, 2021-07-08:
関連性があるissue：[グリスが必要な場所の調査 #81](https://github.com/citbrains/OpenPlatform_ver3_0/issues/81)

NaokiIwasawa, 2021-09-03:
オイレスの導入を一斉に行うにはリソースが不足しているので[付け根ユニット ピッチ軸の再設計 #32](https://github.com/citbrains/OpenPlatform_ver3_0/issues/32)にて先ずは導入してみようと思う。その上で適切かの判断を行いたい。また、付け根のピッチにおいて先んじて導入を検討したのはベアリングの故障が多いためで、滑り軸受けによって問題が解決できるか検討するためである。

但し、グリスが必要な部分の調査 #81 はメンテナンスの面からも必要である。

NaokiIwasawa, 2021-09-15:
# オイレス導入

上記ではすべての軸においてオイレスに変更するのは見送ったが結果的に全ての軸でオイレスを採用した。
理由は3つあり

1. 一部だけはオイレスの評価が十分に出来ない
2. メンテナンスや設計の面で軸の基本構造統一をしたい
3. 部品点数を減らさないと製作費が膨らむ

以上より脚部の軸を全て再設計をしたので詳細を記したissueは以下に示す（※未記入のissueは後に追加する）

1. [脚の付け根ピッチ](https://github.com/citbrains/OpenPlatform_ver3_0/issues/32#issuecomment-919665377)
2. [脚の付け根ロール](https://github.com/citbrains/OpenPlatform_ver3_0/issues/12#issuecomment-919638411)
3. [揺動スライダクランク機構](https://github.com/citbrains/OpenPlatform_ver3_0/issues/11#issuecomment-919666725)
4. 膝 ※
5. [足首ピッチ](https://github.com/citbrains/OpenPlatform_ver3_0/issues/10#issuecomment-919745003)
6. 足首ロール ※

MasatoKubotera, 2021-09-25:
現状のオイレスブッシュでは，切削側面と擦れ早期摩耗が予想される．
[LED-0635](https://jp.misumi-ec.com/vona2/detail/221000090992/?PNSearch=LED-0635&HissuCode=LED-0635&searchFlow=suggest2products&Keyword=LED-0635&list=SuggestPreview)に変更した方が良いかもしれない．

NaokiIwasawa, 2021-09-26:
提案ありがとうございます。
[LED-0635](https://jp.misumi-ec.com/vona2/detail/221000090992/?PNSearch=LED-0635&HissuCode=LED-0635&searchFlow=suggest2products&Keyword=LED-0635&list=SuggestPreview)と[80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0603)の大きな違いは外径に両面カットがある点ですね。
また他の相違点はフランジの厚さで80F-0603の1mmがLED-0635では1.5mmである点。これにより部材との接触断面が2mmから1.5mmになります。（L-t-a=1.5　※aについてLED-0635のみ0.5）
[image][image]

ご提案の通り切削面は両面カット部分に当たり回転せず内径部分で回転する機構となりますので、摺動部分は既製品のスペーサ軸になり摩耗の程度は少なくなると考えられます。

また、0.5mm分軸の隙間を調整する必要があるのでUUW-0312-10BH又はCC-0612-25を変更する必要があります。

NaokiIwasawa, 2021-09-26:
外側のワッシャーを0.5mmに変更すると少し問題が起こるかもしれない
場所は付け根ロールの軸で両側から6mmのネジで締結をすると画像で示した部分が12mmとなり干渉の恐れがある
[image]

MasatoKubotera, 2021-09-26:
@NaokiIwasawa 
減速ホーン部分については，変更できないものと考えております．
理由としては，上記で指摘している「最小部が2mm」ということではなく，オイレスブッシュの側面が回転ではなく，**稼働**するためです．そのためこの部分については，円形である必要があります．

MasatoKubotera, 2021-09-26:
そのため，この部位については交換頻度の高い消耗品として考えた方が良いかと思います．

yasuohayashibara, 2021-09-27:
すでに変更せずで決着したような気もしますが，LED-0635は『許容最高PV値』が低めですので，ご注意下さい．

MasatoKubotera, 2021-09-27:
>部材との接触断面が2mmから1.5mmになる

>LED-0635は『許容最高PV値』が低め

という指摘から，現状の設計で進めたいと思います．

>切削側面と擦れ早期摩耗
には変わりがないので検証時の確認項目としたいと考えます．

closed 2021-10-03 by MasatoKubotera