SUSTAINA-OP/SUSTAINA-OP #97: 3Dプリンタの管理

MasatoKubotera, 2021-07-02:


MasatoKubotera, 2021-07-08:
Brains保有の3Dプリンタは2機あるが，管理しきれていない状態である．
要因として，3Dプリンタの知識不足，決まった担当者が存在しないということが挙げられる．
現状のままでは，1号機のように「いつのまにか壊れた」という悪循環が生まれてしまう．

そのため，担当者を割り当て，3Dプリンタについて知識をつける必要がある．

1号機の故障を踏まえ，Googleフォームの[3Dプリンタ使用報告](https://docs.google.com/forms/d/1QYvrg0mR8shA_44dR8zXYnZMJwbtbsXNPhDuIDpj6E0/edit?usp=sharing)，新人研修の一環として[3Dプリンタ/スライサーソフト勉強会](https://docs.google.com/presentation/d/1WxZWb7rKMKwqMuiG7ODh9XqPMiT-jLJ6awMu3v8MAW0/edit?usp=sharing)を行っている．

NaokiIwasawa, 2022-01-05:
知識を付けるために個人的にチームと同型の3Dプリンターを導入した。
そのため今後はチームのものを管理できる人員になろうと思う。また経験したことは
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/26
で記録していこうと思う

NaokiIwasawa, 2022-02-09:
## 付属のスクレーパーについて
自宅の機体でもスクレーパーを利用したが，ビルドプレートに傷をつけるため利用非推奨です．
そのため，チームで保管しているスクレーパーも利用頻度が高く適している切削物の取り外し用にしようと思う．

MasatoKubotera, 2022-08-08:
長期間稼働していないのでCloseします．

closed 2022-08-08 by MasatoKubotera