SUSTAINA-OP/SUSTAINA-OP #99: サーボモータの修理申請

MasatoKubotera, 2021-07-03:


hiroki-0001, 2021-07-07:
こちらは僕が1, 2年生に教えながら修理申請をやっていくつもりです。

MasatoKubotera, 2021-07-07:
[サーボモータのコネクタ検証 #98](https://github.com/citbrains/OpenPlatform_ver3_0/issues/98)こちらも同時に進めていただければと思います．

MasatoKubotera, 2021-07-07:
@hiroki-0001 
備品管理システムの仕様について
1. 現在中の故障のものはフォームで**あらかた**故障登録をしていますが，故障内容については記載されていません．そのため情報を残すためには，故障しているサーボについてフォームで故障日時などの登録をする必要があります．
2. 修理に出した後には，**故障登録後に修理登録を行う**必要があります．
3. 修理から帰ってきたときは，**在庫登録をすることで予備のサーボとして再登録**されます．

(最終フォーム入力が)故障登録のサーボだけを抽出機能は現在のシステムにはありません．
現状は，［B3M］型番による抽出で型番を指定することで管理番号ごとに最終フォーム記録が表示されますので，そちらから修理に必要な書類を作成するという流れになるかと思います．

hiroki-0001, 2021-07-08:
https://docs.google.com/document/d/1MsO7cTQQ8Z5c2m2XbP8l9NtJM6mksZJ8x2GYoWaFl9U/edit?usp=sharing

以前書いた修理申請の手順に関するマニュアルです。どこにいったか忘れないうちに置いておきます
(こちらは前回の修理時のものなので今回の修理ではまた違った手順になっている可能性は高いです)

Kenzo-Fujisaki, 2021-07-10:
#### 今週のメモ
- 型番はケーブル付近に書いてある
- 日付、故障状況、故障内容を反映させる
入力はドライブの備品管理パーツシステムの目マークを押す
その際工大メールでやる
状態報告、管理番号打ち込む、分類は故障報告、内容に無いときはその他に書き込む

hiroki-0001, 2021-07-11:
本来なら、こちらにサーボモータ(B3M)の備品管理番号と故障内容が書かれたスプレッドシートを送信しようと思ったのですが、リストがいまいち正確ではなかったので、火曜日か水曜日に実物を確認してから送信します。
なので、スプレッドシートの提出が遅れます。。。。

hiroki-0001, 2021-07-23:
https://docs.google.com/spreadsheets/d/1E9HemDR6vYD97DA3DW_0jLP3VUTUtdGlB9eqTrwbnoY/edit?usp=sharing

上のリンクから、故障対象の管理番号・故障時期・故障状況・故障内容をまとめておきました。
入力作業は今度集まって行いましょう

Kenzo-Fujisaki, 2021-07-30:
入力の方一応終わりました

Kenzo-Fujisaki, 2021-07-30:
修正の方も完了しました

hiroki-0001, 2021-08-01:
@MiyawakiKaito 

先日話した、『**修理依頼書**』の作成をお願いします。
『修理依頼書』の記入例は[こちらのドキュメント](https://docs.google.com/document/d/1MsO7cTQQ8Z5c2m2XbP8l9NtJM6mksZJ8x2GYoWaFl9U/edit?usp=sharing)に記載してあるのでそちらを参照してください。

『修理依頼書』のPDFは[近藤科学(B3Mを制作している会社)のサイト](https://kondo-robot.com/faq/customer-afterservice-info)からダウンロードできるのでそちらからしてください。

また今回修理するサーボモータの数が多く、故障内容を記入する欄に収まりきらないので、**故障内容は別紙に記入してください。**
一応例として[スプレッドシート](https://docs.google.com/spreadsheets/d/1yf6bnUZWCo5b0q9yixh3ekKIGSiGcYGHqso0-_5VvJU/edit?usp=sharing)を作成しておいたので、活用してください。

hiroki-0001, 2021-08-01:
@MiyawakiKaito 

『修理依頼書』の編集はPDFを編集するソフトなら何でも大丈夫ですが、一応昔自分が使用していたPDF編集サイトのリンクも張っておきます。https://smallpdf.com/jp/edit-pdf

hiroki-0001, 2021-08-06:
@MiyawakiKaito 『修理依頼書』ですが、作成できたらこのissueに乗っけるようにしてください。言うのを遅れてしまいすいません。

MiyawakiKaito, 2021-08-06:
いえ、提出するのを失念していたのは僕なので・・・申し訳ありません。
https://github.com/citbrains/OpenPlatform_ver3_0/files/6917387/repair_send20150817.1.pdf
https://docs.google.com/spreadsheets/d/1yf6bnUZWCo5b0q9yixh3ekKIGSiGcYGHqso0-_5VvJU/edit?usp=sharing
追加する事項や直すべき箇所があれば教えていただけると助かります。

hiroki-0001, 2021-08-10:
資料作成ありがとうございました！

hiroki-0001, 2021-08-10:
資料自体の作成は大体終わったので、8/17の入構日にモータの梱包・発送をしておきます。

hiroki-0001, 2021-08-24:
資料の整理をしました。[こちら](https://drive.google.com/drive/folders/1xCj39KxWb81CF5eKN5-8PcDzvxOM_5it?usp=sharing)に置いておきます。
slackでの指摘を受け、少し資料の内容も変えておきました。
本日中に発送しておきます。

hiroki-0001, 2021-08-30:
slackのほうでの議論が多かったので、[会話ログ](https://docs.google.com/document/d/1x3iL3ZtM_u1ffzcw9vxvQgABT2nTzejoCIvfDrNm8jw/edit?usp=sharing)を残しておきます。今後修理申請の際に参考になればと思います。

hiroki-0001, 2021-08-30:
書類の整備が終わりました。書類は[こちらに置いてあります。](https://drive.google.com/drive/folders/1xCj39KxWb81CF5eKN5-8PcDzvxOM_5it?usp=sharing)
チェックを受けて問題なさそうであれば、発送いたします。

hiroki-0001, 2021-08-30:
また書類には記入しましたが、故障登録されていた、『70-1993』は故障していなかったため、今回の修理申請には出しません

hiroki-0001, 2021-08-31:
チェックを受けたので、発送したいと思います

hiroki-0001, 2021-09-08:
連絡遅れましたが、発送しました。連絡が届き次第、こちらに報告はしますが詳細は[こちらのドキュメント](https://docs.google.com/document/d/1OC_CzZlHnqysJ16qTEJYPaVCg-WHtrKTncKrh0wkYuQ/edit?usp=sharing)に記載します。

hiroki-0001, 2021-09-17:
相手側から返信が来ました。返信内容については[210830_連絡記録](https://docs.google.com/document/d/1OC_CzZlHnqysJ16qTEJYPaVCg-WHtrKTncKrh0wkYuQ/edit?usp=sharing)に記載しておきました。また頂いた見積書に関しては、[こちらに保存](https://drive.google.com/file/d/1xCswpSyJwloqxXwj7tunAOfcEortWrE4/view?usp=sharing)してあるので確認してください。

yasuohayashibara, 2021-09-18:
ありがとうございます．
この後の処理を進めていきたいと思います．

一つ共有していただきたいこととして，27個の内9個が**修理不要**ということで返却されるようです．
（ギアの洗浄に関しては，近藤科学にお願いした方が良いかもしれませんが，異常なしも含まれます．）

懸念していたことがやや起きていますが，これが続くと**近藤科学との信頼が崩れます**．
今回修理不要として返却されるサーボモータがどうしてそのような判断となったかを明らかにして，次回はこのようなことができるだけ起きないようにしていただければと思います．

あと，フリーホーンの取付方向が逆というのがありますが，組み立てに際して，どうしたらこのようなことが起きないかを考えてみると良いと思います．

先日AMAZONの記事を読みましたが，AMAZONでは失敗したことに際して，それを責めるよりは次にどのようにして起きないかを考える文化があるようです．
エキスパートで無い方が担当することもありますので，

・サーボに問題があることを正確に判断する方法
・メカニズムを間違いなく組み立てる方法

を考えてみなさんで共有した方が良いと思います．

yasuohayashibara, 2021-09-19:
@hiroki-0001 
修理申請のフォーマットが変わりました．
修理後にすぐに壊れないことを説得することが必要になりました．
毎年出しているので，おそらく修理することに対する疑問が出ているおそれがあります．
今回は間に合いませんが，サーボモータの履歴を出して，故障までの間隔を示すことが必要になりそうです．

MasatoKubotera, 2021-09-19:
[ベアリング付アルミフリーホーン（B3M/KRS-6000シリーズ用）](https://kondo-robot.com/product/02137)の商品ページに装着例(下図)があります．

[image]

ホーン内部のベアリングフランジがサーボ内側に向くのが正しい向きとなります．

---

故障判定ですが，世界大会の時に検証として使用したパウダーブレーキを用いて経験ではなく数値として判定する．ということを以降の判定で取り入れるべきかもしれません．

yasuohayashibara, 2021-09-23:
@hiroki-0001 
修理の申請が通りました．
近藤科学に修理を依頼して下さい．

『今回は特別』という文言がありました．
前回も特別扱いされている状態ですので，いつ特別扱いされなくなるかが心配です．

hiroki-0001, 2021-09-23:
了解しました。修理依頼の連絡を贈りたいと思います。
今回の修理申請もccに林原先生を追加してもよろしいでしょうか？

yasuohayashibara, 2021-09-29:
修理済み部品が到着して，検収しました．
津田沼にありますので，新習志野に移動していただいて構いません．
管理をしっかりとお願いします．

https://photos.app.goo.gl/5RJ6utrk5V3LoorN8

hiroki-0001, 2021-10-05:
修理済みのサーボモータを11号館まで運びました。
前回の修理申請では、修理後の写真も必要だった気がするので、一応[ドライブ](https://drive.google.com/file/d/1gwkJQl7s0OD0v0tKCbeI5KbqNToodurX/view?usp=sharing)に修理後の写真は載せておきました。
備品管理登録の更新がまだできていないので、そちらが終了次第、このissueは閉じたいと思います。

hiroki-0001, 2021-10-05:
[サーボモータのコネクタ検証 #98](https://github.com/citbrains/OpenPlatform_ver3_0/issues/98)に関連しそうな情報を[こちらのドキュメント](https://docs.google.com/document/d/1-7z9AIR0u6wZby_AQ8tSodtGzRsak4b-sf8geb0e5OI/edit?usp=sharing)に記載しておきましたので、閲覧してください。
金銭関係の内容なので、一応分けました。

yasuohayashibara, 2021-10-06:
レセプタクルの交換ですが，どの程度効果があったかをできればレポートしていただければ助かります．
個人的には電子回路は手を入れれば入れるほど劣化していくような気がしますので，完全に故障でもしていない限りは交換はしないほうが良いかなと思います．
緩くなっていたのがしっかりと嵌るようになったなど，交換した効果がありましたらお願いします．

hiroki-0001, 2021-10-09:
コネクタを交換したサーボの備品管理番号は『***40-1912**』です。
一応忘れないようにここに記録しておきます。

closed 2021-10-09 by hiroki-0001