SUSTAINA-OP/SUSTAINA-OP #100: ねじの軽量化

MasatoKubotera, 2021-07-04:
現在，干渉する部分においては低頭六角穴付きボルトを使用しているが，すべての箇所で低頭ネジを用いることによって軽量化が見込めるのではないだろうか．また，適切な長さのネジを使用することによって軽量化が見込めると考える．

yasuohayashibara, 2021-07-04:
低頭は入手性が悪く高価ですので，必要な部分だけにしたほうが良いと思います．
長さに関しては，要検討ですね．
短すぎると緩みやすくなるそうですので，その兼ね合いが難しいです．
本数を減らすというのはあるかなと考えています．

MasatoKubotera, 2021-07-07:
こちらの製作所での購入を検討しています．
浅井製作所：http://nejikouba.com/
>低頭ネジや超低頭ネジ、規格品のタッピンねじ小ねじ等製造販売

- <a href="http://nejikouba.com/syourot.htm">小ロット品等の製作について</a>

  >手持ちの工具、材料を使用し製作可能な頭部形状で長さを半端にする程度の製作で一式５０００円です。 特殊な頭部形状、製作しづらい形状や材料の場合に若干ずつ料金が上がります。 なお難加工品以外は通常１０００本位まで一式料金は変わりません。
  
  <a href="http://nejikouba.com/seisaku.htm">小ロット品製作見積もり例</a>を見ると，ほとんどの事例が5000円/1000本であり，1本あたり5円という価格になります．（高くても8000円/1000本であり，1本あたり8円）MiSUMiで現在購入しているGankenKunで一番多く使われているM3×6mm(<a href="https://jp.misumi-ec.com/vona2/detail/221000551286/?HissuCode=CSH-SUS-M3-6">CSH-SUS-M3-6</a>)で10000円/1000本であり，1本あたり10円である．その他の六角穴付きボルトでも浅井製作所の価格を下回るものはない．
  
  [低頭平頭六角穴付小ネジ　３×７　ステンレス](http://nejikouba.com/img/rokkaku.htm)
  [image]
  
  [低頭平六角穴付小ネジ Ｍ３×８ ステンレス](http://nejikouba.com/img/susroku.htm)
  [image]

- さらなる軽量化 
  ステンレス製よりも高価であるが，
  >省スペース化と軽量化
  さらに高い耐食性を兼ね備えた低頭チタンネジ【ＴＩＬＯ】
  鉄やステンレスの６０％という比重のチタン材を使用し、さらに頭部を低く抑える事で軽量化と 省スペース化を実現しました。もちろん耐食性は抜群です。ロボットやラジコン製作、腐食が心配な部分等にご利用下さい。
  
  [低頭チタンネジ【ＴＩＬＯ】](http://nejikouba.com/titan2.htm)  がある．

  [【ＴＩＬＯ－ＢＨ】](http://nejikouba.com/titan2.htm)
  [image]

MasatoKubotera, 2021-07-08:
ネジの本数を減らすことも視野に入れる．

MasatoKubotera, 2021-08-15:
新規で購入するネジに関しては，浅井製作所のねじという方針になり，購入に関するissue [ネジの購入 #129](https://github.com/citbrains/OpenPlatform_ver3_0/issues/129)で引き継ぎます．

ネジの本数による軽量化については，[適切なネジを選定する #49](https://github.com/citbrains/OpenPlatform_ver3_0/issues/49)で扱いたいと思います．

役目を終えたためこのissueを閉じます．

MasatoKubotera, 2021-08-15:
役目を終えたためこのissueを閉じます．

closed 2021-08-15 by MasatoKubotera