SUSTAINA-OP/SUSTAINA-OP #101: 切削パラメータ/方法の見直し

MasatoKubotera, 2021-07-05:
輪郭の切削を行う際に，削り残しをなくすためにZ軸方向の切削を-0.1[mm]まで行っている．

部材厚よりも深く削っているため，部材と捨て板間に張られた両面テープまで一緒に切削している．
これにより，輪郭切削を行った側面がざらつき精度が出ているか疑わしい．

この精度によって，はめあいにも影響が出ていると考える．

このissueでは，切削方法とパラメータの見直しについて議論したい．

yasuohayashibara, 2021-07-06:
悩ましいところですが，精度を出すことに自信のある先輩は0.0mmで加工していました．
（ただし，津田沼2号館3Fの工作室のマシニングセンタでの加工です）

林原も挑戦したことがありますが，しばしば削り残しが発生して，作り直しになったので諦めました．
剛性の高い12号館4Fのマシニングセンタならできるかもしれません．

MasatoKubotera, 2021-07-07:
ORIGINALMINDのKitMill RZ300 / 420であれば，精度出しに時間をかけることができれば，12号館4階のマシニングセンタに比べ 容易にZ軸方向の精度を出すことができるかと思います．

自身が所有しているCNCで精度出しを行った際の参考サイト
http://yunit.techblog.jp/archives/73886876.html

また，現在捨て板としてダイソーのPP板を用いていますが，167の切削時にノギス計測でわかる程度の場所により厚みに違いがあったため，精度が必要なものに関しては[ORIGINALMIND 捨て板](https://www.originalmind.co.jp/goods/03548#169813)を利用すべきかもしれません．

Z軸方向の切削を-0.1[mm]まで行っても支障がないものとしてはコックピットに使用されているPOMなどが挙げられ，精度出しに時間を要する必要性がない部品です．

hiroki-0001, 2021-07-23:
現在、12号館4階のCNCを用いた切削は板材を両面テープを用いて固定することで切削を行っている。
この手法は手軽に行えるが、クランプを用いて板材を固定する手法のほうが精度が向上するため、クランプを用いた切削を行えるようにしたほうがいいかも

MasatoKubotera, 2021-08-15:
Z方向の精度だけでなく，切削速度についても見直していきたいと考える．
現状φ2 2枚刃の超硬エンドミルを使用しているが，羽数や径を大きくすることでより短時間で切削することができる．

MasatoKubotera, 2021-11-14:
XALシリーズ超硬スクエアエンドミル 4枚刃 XAL-PEM4L2在庫がある．
切削条件を変えることで切削時間が短縮されそう．

NaokiIwasawa, 2022-01-21:
# 検証実施予定
ver4.1の量産に向け4枚刃の運用が時間短縮のカギになりそうなので取り組む

## 期間
2022年1月24日～31日

## 担当
@NaokiIwasawa 

## 方針
現在の2枚刃よりも早く切削出来るようにするのが目的
やみくもに変更するのでなくエンドミルの仕様書などから適当なパラメータを算出しながら行う
検証の過程でエンドミルは折る可能性がある

## 評価
自分だけでは評価が出来ないため @MasatoKubotera に評価してもらう

yasuohayashibara, 2022-01-22:
以下に分かりやすく説明されていますので，ご参考まで．
https://toolremake.com/endmill-blade/

機械加工の得意なOBは3枚刃を推奨していました．

MasatoKubotera, 2022-01-23:
4枚刃ですが切り屑排出性が悪く ，両面テープ(-0.1mm)まで切削するGFRPには両面テープが刃に付着し詰まってしまうため，使用することは避けた方が良いかと思います．
すべての部材において4枚刃が使え切削時間を短縮できるわけではありませんのでご注意ください．

また，4枚刃ですが切り屑排出性が悪さについてはパスの作成方法によりある程度は解決することができます．例：溝加工で工具退避の有効化や切込みピッチを小さくするなど

12号館4階で切削した切り屑を見た感じ，両面テープで詰まらなければ，排出が詰まるほどの長い切り屑などが出るということもあまりないので，T5の溝加工(2D輪郭など)もそこまで心配なく切削できるかと思います．

今回検証する4枚刃エンドミル[XAL-PEM4L2](https://jp.misumi-ec.com/vona2/detail/110600204820/?HissuCode=XAL-PEM4L2)

ミスミのHPがメンテナンス中で確認できませんでしたが，2枚刃と4枚刃では波形公差？が異なるようですので確認した方が良いかもしれません．
嵌め合いのところに大きくかかわるかと思います．

https://mori.nc-net.or.jp/qa9450823.html

NaokiIwasawa, 2022-01-28:
# 検証実施
2022年1月28日に主な検証を実施したので報告する

## 結論
切削送り速度を通常の300 mm/minから倍の600 mm/minまで速度を上げても明らかな断面の荒れ・ビビり音等は無く問題なかった．ただし，よく見るとφ3程度の穴の円精度が出ていないなどがあったので精度がいる穴などは今までの300mm/minなどで行った方が良さそうである．

## 条件

- 12号館4階のマシニングセンタ
- 主軸回転速度の最大が10,000rpm
- 検証で使う4枚刃のカッタ[XAL-PEM4L2](https://jp.misumi-ec.com/vona2/detail/110600204820/?HissuCode=XAL-PEM4L2)
- 今までの一刃当たりの送り0.015mm
- 進入送り速度：60mm/min
- 退出送り速度：60mm/min
- ランプ送り速度：60mm/min
- 切り込み送り送り速度60mm/min
- 切り込みピッチ：0.53mm

基本は通常のパラメータと同じだが，**刃の枚数と切削送り速度だけ**を変えている．
速度は以下のように上げていった．

1. 1倍：300mm/min
2. 1.2倍：360mm/min
3. 1.4倍：420mm/min
4. 1.5倍：450mm/min
5. 1.6倍：480mm/min
6. 1.7倍：510mm/min
7. 1.8倍：540mm/min
8. 1.9倍：570mm/min
9. 2.0倍：600mm/min

## 切削の結果
各画像と動画のリンクが以下の**details**に折りたたんである
<details>
    <summary>Details</summary>

1. 1倍：300mm/min
[動画](https://drive.google.com/file/d/17xMNnZG68vifdkj8ErSqc-fwp7Pz9nXH/view?usp=sharing)
[image]
[image]
2. 1.2倍：360mm/min

[動画](https://drive.google.com/file/d/12wyL8llTRpgTycKv_YcAly81vtDGQvow/view?usp=sharing)
[image]
エンドミルの様子取り忘れ

3. 1.4倍：420mm/min
[動画](https://drive.google.com/file/d/1uqNA4iBv87ihH7SmUAL7tU-2gZGzfD7m/view?usp=sharing)
[image]
[image]

4. 1.5倍：450mm/min
[動画](https://drive.google.com/file/d/1cVpcgmmLk-vuMBt78K4qvOG4WvQSUsxp/view?usp=sharing)
[image]
[image]

5. 1.6倍：480mm/min
[動画](https://drive.google.com/file/d/1cVpcgmmLk-vuMBt78K4qvOG4WvQSUsxp/view?usp=sharing)
[image]
[image]

6. 1.7倍：510mm/min
[動画](https://drive.google.com/file/d/12e6lpE_DE9Z71d8AgBhhP9GeLxUmTlmu/view?usp=sharing)
[image]
[image]

7. 1.8倍：540mm/min
[動画](https://drive.google.com/file/d/1sSIf6pdGERk0AF_YddHkhXJbWAoXSEAZ/view?usp=sharing)
[image]
エンドミルの様子取り忘れ

8. 1.9倍：570mm/min
[動画](https://user-images.githubusercontent.com/71487827/151634806-ba4a6fde-7745-46b7-81c0-3d190a170c35.JPG)
切削物の様子取り忘れ
[image]

9. 2.0倍：600mm/min
[動画](https://drive.google.com/file/d/1uK-il5zJ5w5gu-zfEE9FvRUcUaEdljt0/view?usp=sharing)
[image]
[image]

10. 2.0倍：600mm/min
- 進入送り速度：30mm/min（通常の半分）
- ポケットの工具退避を有効

[動画](https://drive.google.com/file/d/1IIGY5qMMO3Wa_U4vJFJAidsI3PrPzh0L/view?usp=sharing)
[image]

</details>

## 今後
ビビり音と切削側面で判断するに大きな問題はなかったが，きりこがエンドミルに連なって付着するのは問題と @MasatoKubotera から意見をもらった．なぜならエンドミルの刃が詰まってしまうことを意味するからだ．
そのため，10.で新たに進入速度等を変えてみたが結果付着物はついてしまい解決できなかった．ただし，動画よく見ると付着するのは最初の穴を切削する際なのでそこだけは送り速度を通常のままにするなど工夫が必要と考える．
また，穴に関して速度を上げていくと円精度が悪くなっていると指摘されたので精度が必要である部分に関しては今までの速度の方が良さそうである．

MasatoKubotera, 2022-02-24:
#273 で引き継ぎます．
何か変更点などがある場合は，Reopenやコメントで対応してください．

closed 2022-02-24 by MasatoKubotera