SUSTAINA-OP/SUSTAINA-OP #103: 付け根ユニットの再設計

MasatoKubotera, 2021-07-07:
https://github.com/citbrains/citbrains_humanoid/issues/1140
ここのissueで提言さてているように減速ホーンを内部に収納する形式に変更したい．

[image]
この形式を取り入れることで，付け根ロール_メインフレーム_1_1を軽量化することができる．

yasuohayashibara, 2021-07-07:
重心位置が下がるので運動特性は変わりますが，ver.1と同様に膝に２個つけても良いかもしれません．
ほぼ上下対象になるので，作りやすくなるかもしれません．

[image]

MasatoKubotera, 2021-07-08:
再設計を行うことで，https://github.com/citbrains/citbrains_humanoid/issues/1428 を実現することができる．

MasatoKubotera, 2021-09-15:
Ver3.0で脚 付け根ユニットにあったリンクを動かすためのピッチ軸サーボをVer4.0では膝ユニットに移行した．
[image]「Ver4.0の左脚」

[image]「Ver4.0とVer3.0の比較：正面」
[image]「Ver4.0とVer3.0の比較：斜め前」

移行した理由・メリットとして下記の内容が挙げられる．

- 付け根ユニットの奥行が小さくなったことによるLeg_Joint_roll_horizontal_01（旧名：付け根ロール_メインフレーム_1_1）の軽量化，[脚のロール軸方向のガタの解明と対策 #1140](https://github.com/citbrains/citbrains_humanoid/issues/1140)で提言されている「たわみ」の低減．
[image]「Leg_Joint_roll_horizontal_01」
[image]「付け根ロール_メインフレーム_1_1」

- 膝ユニットを介してリンクの上下共通化
※画像編集中

- 付け根ユニットロール軸前後共通化
[image]「ver3.0 付け根ユニット前方」
[image]「ver3.0 付け根ユニット後方」
[image]「ver4.0 付け根ユニット前方」
[image]「ver4.0 付け根ユニット後方」

軸に関しては下記のissueで取り扱っていた．
- [脚軸のオイレス 部品の導入 #96](https://github.com/citbrains/OpenPlatform_ver3_0/issues/96)
- [脚ロール軸 ベアリング回りの再設計 #12](https://github.com/citbrains/OpenPlatform_ver3_0/issues/12)
-[付け根ユニット ピッチ軸の再設計 #32]( https://github.com/citbrains/OpenPlatform_ver3_0/issues/32)

yasuohayashibara, 2021-09-16:
少々気になったので指摘
実際に作って動かしてからでも構いませんが，以下の黄色の部分のフィレットを大きくしたほうが良いかもしれません．
赤の方向の力が加わったときに，下図のような変形が発生するおそれがあります．
構造的に変形が少ない良い機構になっていると思いますが，梁が薄いのでいまいち不安です．
（これも解析してみれば良いことですが．．．）

[image]

ちょっと図を借りてきてしまいましが，このような感じです．
[image]

MasatoKubotera, 2021-09-17:
処置として指摘いただいたフィレットについてR1からパーツに干渉しない最大限のR4に修正しました．

[image]「修正前」
[image]「修正後」
[image]「修正後の脚付け根ユニット」

closed 2021-10-03 by MasatoKubotera