SUSTAINA-OP/SUSTAINA-OP #104: カメラUSBケーブルの変更

MasatoKubotera, 2021-07-07:
現状では，カメラとJetsonを繋ぐケーブルには [Mini USBケーブル 左向き両端L型 USB L字 USB充電ケーブル](https://www.amazon.co.jp/dp/B071D7ZH2B) が使われているが，この長さのケーブルを使う必要性がない．伸縮できる仕様になっているが使用することがないため，最低限の長さでよい．

カメラ側のケーブルが上部に向かっているが，転倒時 相手ロボットに接触する恐れがあるため，向きを変更したい．

[image]

TakuyaNagamine, 2021-07-27:
# カメラとJetsonを繋ぐために必要な最低限のUSBケーブルの長さ
## コックピットと胴体_メインフレーム_1_13の間を通してつなげる場合
・紐を使い,必要な長さを測った.
・首の可動域はUSBケーブルの長さが最長となるようにして計測した.
### 結果
下の写真のようになり,約30cmとなった.
[image]
### 考察
・今回計測した長さは30cmであったが,実際に買うときは切れてしまわないように40cm以上のものを買う必要がある.
・候補となりそうなコードを探したが,[長さが25cmのもの](https://amzn.to/3i7aOLX)や[USB A端子がL字に曲がっていないもの](https://amzn.to/3rEgCj6)しか発見できなかった.
・カメラ側のケーブルの向きを変更するにはL字に曲がっていないものを採用する(負荷がかかる可能性あり),もしくは基盤の向きを上下逆さまにする(認識に問題あり?)必要があるかもしれない.

TakuyaNagamine, 2021-08-11:
認識班によると「基盤の向きを上下逆さまにすることは可能ではあるが確証がない」とのことだったので,大学に入れるようになったときに検証してみる.

MasatoKubotera, 2021-12-31:
こちらについて，新規カメラモジュールが採択が確定されるまで保留にしたいと思います．

closed 2022-02-14 by MasatoKubotera