SUSTAINA-OP/SUSTAINA-OP #107: ハンドルスポンジのCAD反映

MasatoKubotera, 2021-07-10:


MasatoKubotera, 2021-07-13:
現状の機体でも，ハンドルに取り付けているスポンジとコックピットが外す際に干渉し，コックピットを取り外すためにハンドルを解体しなければいけない状況になってる．
新規コックピット開発でこのような状況にならないためにも，ハンドルに取り付けているスポンジをCADに反映させるべきである．

MasatoKubotera, 2021-07-23:
167機を参考にしてください．
質量設定も．既定の素材で行わず測定して反映してください．
部品名は．必ずその商品の品番を入れてください．

KentaTsukino, 2021-08-25:
[image]
Inventorのデータです。↓

https://drive.google.com/file/d/1Cjk4Mzng23Vy_i1Sm4pJg7iuXNiiRKK9/view?usp=sharing

MasatoKubotera, 2021-09-02:
上記で共有されているSGB-100.iptをver4_0に反映しました．

[image]

MasatoKubotera, 2021-09-02:
ハンドルのスポンジ材料は[SGB-100](https://www.amazon.co.jp/Hikari-SGB100-%E3%82%B9%E3%83%9D%E3%83%B3%E3%82%B8%E3%82%AB%E3%83%90%E3%83%BC-15%C3%97120-SGB-100/dp/B0091G5AW2)です．

MasatoKubotera, 2021-09-02:
>詳細なデータ（重さや材質）はまだ入れられてない状態です。

入構規制解除後に再度修正することにします．そのため，このissueを保留とする．

MasatoKubotera, 2021-10-03:
ハンドルのスポンジについては，カメラでの認識を妨げるため廃止とする．
そのため，このissueの必要性がなくなった．

関連issue [ハンドルの再設計 #19](https://github.com/citbrains/OpenPlatform_ver3_0/issues/19)

closed 2021-10-03 by MasatoKubotera