SUSTAINA-OP/SUSTAINA-OP #110: 整備後機体 再組立て

MasatoKubotera, 2021-07-15:
動く機体数が少ないため，整備が終了した機体を再度組み立てし，動く状態に復元したいと考える．

この再組立ての問題点として下記が挙げられる．
- 組み立てに時間がかかる
- ねじ材がついているねじが再利用できないことから新品のねじが必要
- 167機のハードウェアに書かれていない部分が存在し，経験/知識が必要
- 組み立て直したとしても，サーボモータの修理，再設計機体の組み立て前に解体する必要がある．

上記の問題点を含め，再度組み立て直すことが妥当であるか判断し行う必要がある．

NaokiIwasawa, 2021-07-23:
# 組み立てを行う方針に決定しました

## 部品の再利用に関して

ベアリングを除き基本部品はすべて残してあるので再利用する

## 日程
**26、28、29**
3日間しか確保できないが、その中で1年生に組み立ての一部を体験してもらう方針

また組み立ての際に特筆することがあればissueに記載する。

NaokiIwasawa, 2021-07-26:
# 三日間の予定（仮）

1日目：頭、腕
2日目：胴体から足のユニット
3日目：足の残り

出来るとこまで進め、余裕があれば先に進める。

MasatoKubotera, 2021-08-15:
それぞれの機体ごとにissueが立てられました．
[163の再組み立て #122](https://github.com/citbrains/OpenPlatform_ver3_0/issues/122)
[165機体再組立て #123](https://github.com/citbrains/OpenPlatform_ver3_0/issues/123)

再組立てで浮上した問題については，それぞれのissueで記載しています．
このissueが不要となりましたのでcloseします．

closed 2021-08-15 by MasatoKubotera