SUSTAINA-OP/SUSTAINA-OP #111: 163機体コックピット故障

Keita1Mori, 2021-07-15:
163機体のコックピットの解体と組み立てをしていた際にOSコンが取れてしまいました。原因として経年劣化だと思われます。
[image]
[image]

MasatoKubotera, 2021-07-15:
再度組み立てを行う際は，修繕の必要がある．

RyodaiArai, 2021-07-26:
今使われているOSコンが秋月電子で取り扱っていないため，以下のOSコンに変更する．
https://akizukidenshi.com/catalog/g/gP-14452/

RyodaiArai, 2021-07-29:
https://files.slack.com/files-pri/TNC67U8FP-F02942FKZLP/img_20210729_154000.jpg

RyodaiArai, 2021-07-29:
OSコンの修繕完了．
[image]

RyodaiArai, 2021-07-29:
終わったのでCloseします

closed 2021-07-29 by RyodaiArai