SUSTAINA-OP/SUSTAINA-OP #112: 電子回路niyoru 電流計測

MasatoKubotera, 2021-07-18:


MasatoKubotera, 2021-07-18:
マイコンボード(Arudino Pro Micro)，[電流センサIC ACS712ELCTR-30A-T](https://akizukidenshi.com/catalog/g/gI-12292/)などを用いてサーボモータに搭載されているICではなく外付けで計測が可能なシステムを作製する．

現状は，安定化電源に付属している電流計で簡易的な計測を行っているが，このシステムを用いてJetson の最大消費電流，機体の最大消費電流などを計測したいと考えている．

MasatoKubotera, 2021-07-22:
[Arduino Pro micro 互換](https://www.aitendo.com/product/12196)を使用した，電流測定回路を作成．

回路は，[電流センサIC ACS712ELCTR-30A-T](https://akizukidenshi.com/catalog/g/gI-12292/)の[データシート](https://akizukidenshi.com/download/ds/allegro/acs712.pdf)に記載されているApplication 1.のサンプル回路である．

使用する部品は，下記のとおりである．
- [Arduino Pro micro 互換](https://www.aitendo.com/product/12196)×1
- [電流センサIC ACS712ELCTR-30A-T](https://akizukidenshi.com/catalog/g/gI-12292/)×1
- セラミックコンデンサ 0.1uF ×1
- セラミックコンデンサ 1nF ×1

計測した電流によって出力電圧が変化するため，その出力電圧をArudinoを使用し測定，電流値に変換．

変換するプログラムは，[ACS712を使ってみた](https://qiita.com/k_zoo/items/34ffc2b33b842dfb51c9)を参考に作成する．

YukiShigematsu, 2021-08-08:
## 電流値測定回路設計方針
### 必要な機能
・機体動作時にも使用可能な設計
・モバイルバッテリーと電池、両方で駆動可能
・パソコンを用いなくても使用可能な設計（LCDモニター設置など）
・異常電流の警告（スピーカーの設置）

RyodaiArai, 2021-08-08:
@MasatoKubotera 
## 電流値測定回路設計方針
### 必要な機能
・機体動作時にも使用可能な設計
・モバイルバッテリーと電池、両方で駆動可能
・パソコンを用いなくても使用可能な設計(LCDモニター設置など)
・(異常電流の警告(スピーカーの設置))

### 必要となる回路
・電流計測ＩＣ周辺回路 
・モニター周辺回路
・マイコン周辺回路
・電源回路 
・(スピーカー周辺回路)

MasatoKubotera, 2021-08-08:
@RyodaiArai 
>モバイルバッテリーと電池、両方で駆動可能

が気になりました．
Aruino Pro Micro はUSBケーブルを用いてパソコンから給電することができます．
また，測定する電源から駆動電源を三端子レギュレータなどを用いて供給しても良いと思います．
モバイルバッテリや電池から給電した場合，重さが増大し
>機体動作時にも使用可能な設計

を行ったとしても，正確な電流測定が行えない恐れがあります．（それよりは，三端子レギュレータなどを用いた方が良い気がします．そんなに電力を食うわけでもないはずです．）

---
>パソコンを用いなくても使用可能な設計(LCDモニター設置など)

非常にいいと思いますが上記で記載したように設計には気を付けてください．
arduinoでLCDを使う場合は，[Ｉ２Ｃ接続小型キャラクタＬＣＤモジュール　１６×２行　３．３Ｖ／５Ｖ](https://akizukidenshi.com/catalog/g/gP-08779/)が安いのでお勧めです．（サンプルプログラムも秋月で公開されている）

---
>(異常電流の警告(スピーカーの設置))
>(スピーカー周辺回路)

についてですが，[圧電スピーカー](https://akizukidenshi.com/catalog/g/gP-04118/)などを用いれば，Arduinoと接続するだけで簡単に音を発音できます．それに特化した[Aruinoの関数](http://www.musashinodenpa.com/arduino/ref/index.php?f=0&pos=2484)も用意されています．そのため，
>異常電流の警告

程度であれば，特にこだわってアンプなどの設計する必要もないように思います．

TakuyaNagamine, 2021-09-04:
# 回路の試作
## 追加した機能
・マイクロSDカードリーダー
・LCD表示器
・電流計測IC
・USB通信
・スピーカー
## 使用したマイコン
ATMEGA328P-PU
## 追記
9月の7日までに部品を決定する必要があったため,シンボルなどは一部作成せずに回路を製作しました.
シンボルは今後作成する予定です.
## 写真
[image]

TakuyaNagamine, 2021-09-12:
# 回路製作
## 作業内容
回路の整理
## 写真
[image]

TakuyaNagamine, 2021-12-18:
# 回路製作
## 製作状況
ALLPCBにて基盤を発注し,現在動作確認中.
## 変更点
・モーターへ流れる電流の計測と全体に流れる電流の計測を行うために,電流センサーICを[ACS724LLCTR-10AB-T](https://akizukidenshi.com/catalog/g/gI-14880/)×4,[ACS712ELCTR-30A-T](https://akizukidenshi.com/catalog/g/gI-12292/)×1に変更した.
・配置する電流センサーICの数が4から5に変更されたことにより,ATMEGA328P-PUのPort Pinの数が足りなくなったので,ADコンバーター[HX711](https://akizukidenshi.com/catalog/g/gI-12473)を導入した.
## 写真
[image]

MasatoKubotera, 2021-12-23:
@TakuyaNagamine @YukiShigematsu 
昨日，長峰君と動作確認しました．

まず，Micro USB type - B（データ通信可能なケーブル）を基板とPCに接続すると，「USBデバイスが認識されません」のエラーウィンドウが出る．

USB・シリアル変換ICが認識しなかったため，購入元である秋月電子の[ＵＳＢ・シリアル変換ＩＣ　ＣＨ３４０Ｎ](https://akizukidenshi.com/catalog/g/gI-16305/)にある[ドライバーソフト(Windows、インストーラー、2021.5.18更新)](https://akizukidenshi.com/catalog/g/gI-16305/)をインストール，再起動し再度接続しなおしたが同エラーウィンドウが出た．

回路，はんだミスの恐れがありますで再度確認する必要があります．

また，CH340NでArudinoを駆動している参考資料が見当たらないのであったら共有してください．

YukiShigematsu, 2022-01-21:
USB接続できない問題について、CH340NのVCC部分がGNDにつながっていたため、動作していなかったものと考えられます。
また、V3がGNDでなくVCCに接続されていたため、これもch340Nの動作に不具合を与えていると思われます。

YukiShigematsu, 2022-01-21:
基板に駆動電源供給用のコネクタが未設置であったため、HX711から外部に信号を送信するためのXHコネクタを電源供給用コネクタとして使用する。
配線の変更はXHコネクタとレギュレータを配線で接続することで行う。

YukiShigematsu, 2022-01-21:
基板上のIC [ACS724ELCTR-10AB]4つとLCDの動作確認を行った。
現状、この基板でＵＳＢを用いたマイコンへのプログラム書き込みは行えず、自身もこのマイコン単体にプログラムを書き込むためのarduino UNOを所有していないため、接続部にA-ster 32U4 をジャンパ線で接続し動作確認を行った。
ACS724ELCTR-10ABの動作確認はDCモータに負荷を与えることによる電流値の変化を確認することで行った。
ACS724ELCTR-10ABとLCDはどちらも正常に機能していた。

MasatoKubotera, 2022-01-21:
>接続部にA-ster 32U4 をジャンパ線で接続し動作確認を行った。

基板上には無い，**Astar 32U4モジュール**を基板上に実装された32U4マイコンにジャンパー線で接続したという解釈で大丈夫でしょうか．

実験/検証の様子を画像で貼付してもらえると嬉しいです．
プログラムなどもドライブなどでも構わないので共有しておいてください．無くすと0からの作成になってしまうので．．．

https://github.com/citbrains/OpenPlatform_ver3_0/issues/112#issuecomment-997133198
に示している回路図を確認しましたが，ATMEGA328P-PUの電源周りも疑わしい部分もあります．
CH340Nだけが原因でない可能性があります．

YukiShigematsu, 2022-01-25:
検証時の写真です
[image]
・LCDにはADCからの値を直接表示している。
・電流計測ICからの値の変動はモータに負荷を与えそれをLCDに表示させることで確認した。

実験用のプログラムをgoogleドライブにアップしました
https://drive.google.com/file/d/1OAkcrNcxEF56Wq3As0HXudCxbf8XOkK_/view?usp=sharing

YukiShigematsu, 2022-01-25:
「ATMEGA328P-PUの電源周りも疑わしい部分」はSTMEGA3238P-PUにコンデンサが直列で接続されていることだと思います。
この部分はコンデンサは接続せずはんだでつないでいます。

YukiShigematsu, 2022-02-05:
今回使用している電流計測ICACS724ELCTR-10ABは電流量に応じて出力電圧を変化させるが、電流の流れる向きに応じて、2.5V付近を中心として+方向とー方向の両方向に電圧が変化するものとみられる。
アルディーノのADCの分解能は1024であるため、計測できる電流量の変化は0.02Aずつになると思われる。

MasatoKubotera, 2022-02-07:
よく使われるICなので計測プログラムについては，公開されているものを使用した方が早いです．
https://www.instructables.com/ACS724-Current-Sensor-Measurements-With-Arduino/

MasatoKubotera, 2022-02-15:
Arduino 公式にサンプルプログラムがありました．
https://www.arduino.cc/reference/en/libraries/acs712/

TakuyaNagamine, 2022-04-16:
現在、緊急で行う必要がないため一旦クローズします

TakuyaNagamine, 2022-11-07:
回路設計の練習として、再度設計を行う

MasatoKubotera, 2023-03-07:
https://github.com/citbrains/SUSTAINA-OP/issues/349 のために電源計測が必要となり以下のモジュールを検証用途に購入しました．

- [INA238 高精度I2Cディジタル電流・電圧・電力計モジュール(最大20A)](https://strawberry-linux.com/catalog/items?code=12238)

@KANBE8810 に計測用ソースコードの作成を依頼しました．

KANBE8810, 2023-03-07:
Arduino Microを用いてINA238で電流・電圧・温度を計測できるコードを作成しました．
検証としてDMN29を安定化電源12Vで動かした際の値を計測して，おおよそ正しい値が取れていることを確認しています．

以下ソースコードです．
``` 
#include <Wire.h>

#define INA238_ADDR        0x40
#define INA238_CONFIG      0x00
#define INA238_ADC_CONFIG  0x01
#define INA238_SHUNT_CAL   0x02
#define INA238_VBUS        0x05
#define INA238_DIETEMP     0x06
#define INA238_CURRENT     0x07

int result = 0, result2 = 0, result3 = 0;
double v = 0.0, tem = 0.0, a=0.0;

void INA238_write(byte reg, byte val)
{
  Wire.beginTransmission(INA238_ADDR);
  Wire.write(reg);
  Wire.write(val);
  Wire.write(0x00);
  Wire.endTransmission();  

  delay(10);
}

short INA238_read(byte reg)
{
  unsigned char ret1, ret2;
  int res = 0;
  // request the registor
  Wire.beginTransmission(INA238_ADDR);
  Wire.write(reg);
  Wire.write(0x00);
  Wire.write(0x00);
  Wire.endTransmission();  

  delay(100);

  // read
  Wire.requestFrom(INA238_ADDR, 2);
  ret1 = Wire.read();
  ret2 = Wire.read();

  //Serial.println(ret1);
  //Serial.println(ret2);

  if(reg == INA238_DIETEMP){
    res = ret1 << 4; 
    ret2 = ret2 >> 4; 
    res |= ret2;
  }
  else {
    res = ret1 << 8; 
    res |= ret2;
  }
  
  return res;
}

void setup() {
  Wire.begin();
  Serial.begin(9600);
  INA238_write(INA238_CONFIG, 0x80);
  INA238_write(INA238_ADC_CONFIG, 0xF0);
}

void loop() {
  result = INA238_read(INA238_VBUS);
  v = 0.003125 * result;

  Serial.print("bus v : ");
  Serial.println(v);

  result3 = INA238_read(INA238_CURRENT);

  double current_lsb = sqrt(3/0.002)*2 / pow(2,15);
  a = result3 * current_lsb;

  Serial.print("A : ");
  Serial.println(a);
  
  result2 = INA238_read(INA238_DIETEMP);
  tem = 0.125 * result2;

  Serial.print("tem : ");
  Serial.println(tem);

  delay(1000);
} 
```

KANBE8810, 2023-03-09:
上記のコードを一部変更して，Serial.readで0を取得した際に，INA238で取得したセンサ値(電圧，電流，温度)をSerial.writeでバイナリデータで送信するようしました．
なお，送信するデータは12byteです．
また，ボーレートが1000000でも動作することを確認してあります．

MasatoKubotera, 2023-03-13:
Arduino Microに上記のソースコードを書き込み下図のように172に装着しました．
[image]

バッテリとMainBoard間にINA238モジュールを装着しました．

MasatoKubotera, 2023-03-13:
計測した値を使った評価については，#349 に引継ぎます．

closed 2023-07-25 by MasatoKubotera