SUSTAINA-OP/SUSTAINA-OP #116: リポバッテリーの過放電防止回路の設計

sekitorimisaki, 2021-07-23:


sekitorimisaki, 2021-07-29:
[image]
ブレッドボードで組んでみましたが、うまくいきませんでした。

YukiShigematsu, 2021-08-05:
## 過放電防止回路の設計
https://drive.google.com/drive/folders/100LxzD-blRXK54aMpYRlk5bJvftpMHFW?usp=sharing
回路に使用する部品を選定し上記のドライブにリスト化した。（随時更新予定）
現段階で作成する回路として、電源回路、マイコン周り、分圧抵抗、（スピーカー）が挙げられる

sekitorimisaki, 2021-08-06:
できるところまでマイコン周りの回路図を作成した。
[image]

MasatoKubotera, 2021-08-06:
@sekitorimisaki @YukiShigematsu 
コメントを残します．

見る限り，セラミックコンデンサ，抵抗において既存するシンボルではなく自作もしくはダウンロードしたものを使っていると思います．

セラミックコンデンサ，抵抗においてはできる限りEAGLEインストール時から入っている（と思われる）rclライブラリ内のR-US_○○○○○やC-US○○○○○を使ってほしいです．

理由としては，サイズと値が同じであれば指定した型番でなくともある程度代用が可能であること，ライブラリ数の増加が挙げられます．また，回路図としたときに型番表示されると値が不明確になってしまうデメリットがあります．

[image]
[image]

sekitorimisaki, 2021-08-07:
[image]
修正しました。

sekitorimisaki, 2021-08-10:
[image]
足りなかった部分を増やしました。

sekitorimisaki, 2021-08-17:
[image]
電源回路をつなげました。

MasatoKubotera, 2021-08-17:
@sekitorimisaki 

[image]

**オレンジ色で囲った部分**：
回路図の基本ルールとして，電源電圧(+5VやVINなど)は上向きに，GNDは下向きに書きます．
参考：[aruduino UNOの回路図](https://www.allaboutcircuits.com/uploads/articles/Arduino_schematic.png)

**青色で囲った部分**：
交点がないのが謎です．

**緑色で囲った部分**：
紛らわしいことになっているので，水晶?と接続してない場合は位置を変えた方が良いと思います．

sekitorimisaki, 2021-08-24:
<img https://user-images.githubusercontent.com/82258905/130536602-f1c7c93b-183c-448e-8bbc-b862fdb8dc5b.jpg" width="320px">
修正しました

sekitorimisaki, 2021-08-24:
[image]
修正しました

sekitorimisaki, 2021-08-24:
[image]
電圧表示と分圧抵抗とスピーカーを追加しました。

sekitorimisaki, 2021-12-19:
# 製作状況
・基板を発注し、ブートローダーを書き込むのに必要な部品はつけ終わりました。
・ブートローダーの書き込みがうまくいっていません。

# 回路図
[image]

MasatoKubotera, 2022-02-07:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/224 で引き継いで行うためこちらはcloseします．

closed 2022-02-07 by MasatoKubotera