SUSTAINA-OP/SUSTAINA-OP #117: ヒューズの搭載

MasatoKubotera, 2021-07-23:
ケーブルの断線や基板上のねじなどの導体が落ちる可能性があり，短絡した場合の措置として導入する．
電源(LIPOバッテリー入力部，安定化電源入力部，DCDCコンバータ出力部)にヒューズを接続する．

[機体の消費電流計測 #55](https://github.com/citbrains/OpenPlatform_ver3_0/issues/55)で計測した値，搭載している部品等の理論値を考慮しヒューズの遮断電流値については選定する．

ヒューズは，溶断時間の短い自動車用のブレードヒューズを用いる．候補を下記に記す．
- [FUSE AUTO 20A 58VDC BLADE MINI](https://www.digikey.jp/product-detail/ja/littelfuse-inc/0891020-NXS/F4991-ND/2046669)
- [FUSE BLOCK BLADE 500V 30A PCB](https://www.digikey.jp/product-detail/ja/keystone-electronics/3557-2/36-3557-2-ND/2137305)

yasuohayashibara, 2021-07-23:
ヒューズは諸刃の剣になるおそれがあります．

普段は良いと思いますが，大会でここぞというときに切れて試合続行不可能になるおそれがあります．
例えば，練習ではロボット同士が絡み合うと壊れることを懸念して離しますが，大会では審判の指示があるまで離せませんので，そのときに普段流れないような大電流が流れて，ヒューズが切れるおそれがあります．

これを防ぐ方法としては，試合のときだけ遮断電流の大きなヒューズに変更するという手はありますが，運用が面倒になりヒューマンエラーが発生しやすくなります．

あと，CPU周辺に関しては，導体によるショートは電源だけでなくI/Oポートの故障も誘発しますので，電子回路をケースで囲ったほうが良いかもしれません．

YukiShigematsu, 2021-07-26:
大会時などはヒューズの機能を無効化するために、ヒューズとスイッチを並列に繋ぐのはどうでしょうか。

スイッチを入れた場合:ヒューズが切れていても電流が流れる。
スイッチを切った場合:ヒューズが切れていたら電流が流れない。

MasatoKubotera, 2021-07-26:
ヒューズに並列で繋ぐスイッチを入れることの問題点として以下が挙げられます．

- ヒューズの個数分 スイッチ，ロードスイッチ用のFETが必要になり回路スペースの拡大する．
(最低限スイッチ１個とヒューズの個数分のFET)

そのため，スイッチを搭載することは難しいと考えます．
解決策として，試合の際はニューズの電流値を大幅に上げる，ヒューズ部分を導通させることが挙げられます．

MasatoKubotera, 2021-07-26:
電子回路のケースについては，部材と接触しやすい底面だけでなく，上下で覆うようなカバーの設計を行っていこうと考えています．

MasatoKubotera, 2021-07-26:
ヒューズの選定については，KIJIMA(キジマ)製の[ヒューズブレーカー 平型](https://www.monotaro.com/p/0569/2733/?utm_id=g_pla&utm_medium=cpc&utm_source=Adwords&utm_campaign=246-833-4061_6466659573&utm_content=118200484700&utm_term=_496718128754__pla-1200563622924&gclid=Cj0KCQjwl_SHBhCQARIsAFIFRVWg3ul3ulq_JIOMV9KNABxvdAqqp-7Fw313oD6MIwiOzSD4Py-7rmsaAq7cEALw_wcB)も候補として考える．

このヒューズブレーカヒューズの利点として，
>万が一、対応電流以上のアンペアが流れてもピンが飛び出して回路をカットし安全です。再度ピンを押すと再使用可能です。

がある．

MasatoKubotera, 2021-08-17:
夏季の試作基板では，
全体のヒューズ：20A（概算）
DCDCコンバータ出力：2A（概算）
を搭載し検証を行う．

使用するヒューズ，ヒューズホルダーは下記に記す．
20A：[0891020.NXS](https://www.mouser.jp/ProductDetail/Littelfuse/0891020NXS?qs=%2Fha2pyFadugOdQ1JPwngNLWtK1s7FD5dtUeelFP6pxWaN9wxMkLdgg==)
2A：[0891002.NXS](https://www.mouser.jp/ProductDetail/Littelfuse/0891002NXS?qs=5b8eIUDTsllqv1a3n5kgDQ%3D%3D)
ヒューズホルダー：[3557-2](https://www.mouser.jp/ProductDetail/Keystone-Electronics/3557-2?qs=cxy41lVAGV%2FfbyASjpXK1g%3D%3D)

closed 2021-08-17 by MasatoKubotera