SUSTAINA-OP/SUSTAINA-OP #118: CNC(ORIGINALMAIND KitMill RD)の運用

MasatoKubotera, 2021-07-23:
夏休み中に行うこととして，下記の要項を行う必要がある．

- 組み立て
- 精度出し
- 収納BOXの作成
- 切削パラメータのテンプレート作成

MasatoKubotera, 2022-02-10:
ダウンロード資料については[Googleドライブフォルダ](https://drive.google.com/drive/folders/1fi2T0FbRbXrZ1i3O-6IrHZ3xGZZ6HT-z?usp=sharing)に格納します．
※チーム外秘

MasatoKubotera, 2022-02-10:
オンラインマニュアルについて

下記のアドレスから閲覧することができます．
閲覧には，会員登録とパスコード（シリアルコード）が必要になります．
https://www.originalmind.co.jp/manual/rd/

シリアルコードについては通常X軸カバー裏面に取り付けられているようですが，分解しないと確認できないため機体前方に貼付しなおしました．

[Googleドライブフォルダ](https://drive.google.com/drive/folders/1fi2T0FbRbXrZ1i3O-6IrHZ3xGZZ6HT-z?usp=sharing)にも画像ファイルで保存しています．

Keita1Mori, 2022-02-11:
# 2/11 進捗報告
## 担当者
@MasatoKubotera @Keita1Mori 
## 動作確認

- 精度
  - X軸：✕
  - Y軸：✕
  - Z軸：△
 - 詳細
   - X軸、Y軸を動かした際にリニアガイドをすれるような音がした。

 - 原点復帰
    - X軸：△
    - Y軸：△
    - Z軸：✕
 - 詳細
   - 原点センサーに異常あり。X軸、Y軸は原点センサーが反応はするが止まるのに時間がかかる。一方Z軸は通電はしているものの反応はしていない。

## ソフトウェア設定
上記の先輩の報告にもあるように資料をみてください。
※CNC4.02を起動する場合、管理者で開かないとエラーを吐くので以下の手順で管理者の設定を行ってください。

1. CNC4.02のアプリの画面を右クリックしプロパティを選択
2. 互換性を選択
3. 設定の欄内にある管理者としてこのプログラムを実行するを選択
4. 適用をクリックして閉じる

以上がセットアップの説明になります。

Keita1Mori, 2022-02-11:
# 追記②
軸精度がわかる写真

[image]
X軸のリニアガイドの写真

---
[image]
Z軸のリニアガイドの写真

Keita1Mori, 2022-02-11:
# 精度だしの参考資料
[CNCの制度だし](http://yunit.techblog.jp/archives/73886876.html)

MasatoKubotera, 2022-02-12:
CNC駆動用のPCについて

下記のPCを７月（世界大会終了）まで貸し出します．

HZ550/D
intel celeron cpu 3855U @ 1.60GHz 1.60GHz
RAM 4.00GB

低スペックなPCなため，大きいファイルを扱う場合は固まることがありますのでご注意ください．

Keita1Mori, 2022-02-12:
# 調査報告(2/12)
Z軸の原点センサーが反応しないとの報告でしたが、今日調査をしました。

## 調査内容
Z軸の原点センサが反応しないとのことだったので、検証するためにX軸の原点センサのケーブルとZ軸の原点センサのケーブルを交換しました。
## 結果
Z軸の原点センサをX軸のケーブルに接続して原点移動をさせたところ、センサが反応して止まりました。一方Z軸の原点センサをZ軸ケーブルで動かしたところ動くようになりました。
## 考察
Z軸とX軸は運用している時に右往左往するのでケーブルに負荷がかかったいて接触不良を起こしているのだと思われる。よって正式に運用する前にケーブルを全部確認した方がいいかと思います。

MasatoKubotera, 2022-02-14:
>Z軸とX軸は運用している時に右往左往するのでケーブルに負荷がかかったいて接触不良を起こしているのだと思われる。よって正式に運用する前にケーブルを全部確認した方がいいかと思います。

補足します．
Y軸のステッピングモーターについては稼働しても定位置なため，ケーブルの根本に負荷がかかることはありませんが，X・Y軸においては稼働するときにケーブルが適切に固定されていないと可動位置によっては根元に負荷がかかります．

現状，接触不良状態ではありますが使えないことは無いです．
今後運用し低区時には根元に負荷を与えないように固定する必要があります．

Keita1Mori, 2022-02-15:
# 分解報告
リニアガイドを詳しく見るためにZ軸の解体を行いました。解体動画は以下のリンクGoogle Driveにあげました。
https://drive.google.com/drive/folders/17hVVNX6IF_aRPBeQ7mo6K3ZXOAAln7aD
解体工程に関してはマニュアルの組み立ての項目を見ながら進めました。

Keita1Mori, 2022-02-15:
# 追加報告
上記の報告のように分解してしまったパーツを混ざらないように箱に入れて保管しています。保管場所としてメカ班の棚上から3段目のところにまとめて置いておきました。

[image]
[image]

以下物品は @MasatoKubotera なので久保寺先輩の机にまとめて置いておきました。
[image]

Keita1Mori, 2022-02-21:
# 報告
## 後日の作業工程
現状分解されているので戻す必要がある。
最初にネジ穴の部分を洗浄する必要がある。
## 担当者
現状自分だけ把握してる状態なので、情報共有として岩澤と作業する必要がある。
## 作業日
21日切削終了後作業開始予定

Keita1Mori, 2022-02-21:
# 締め切り
23日までにグリスの指標を決める。

Keita1Mori, 2022-02-23:
# 報告
現状色々調べて進めてはいるもののグリスについて何を使用していいか自分ではわからないため、明日（2/24）の午後
@MasatoKubotera と @NaokiIwasawa と一緒に話しあって決めたいと思う。

Keita1Mori, 2022-02-23:
# 追記
知れべた結果一覧
- misumi
  - [THK](https://jp.misumi-ec.com/vona2/result/?Keyword=THK%20%E3%82%B0%E3%83%AA%E3%82%B9&isReSearch=1&relatedKeyword=1)
- Amazon
  - [NSK](https://www.amazon.co.jp/%E6%97%A5%E6%9C%AC%E7%B2%BE%E5%B7%A5-NSK-%E3%82%B0%E3%83%AA%E3%83%BC%E3%82%B9-GRS-80%E3%82%B0%E3%83%A9%E3%83%A0/dp/B088BK37TT)
## 参考資料
[リニアガイドについて](https://jp.misumi-ec.com/tech-info/categories/technical_data/td04/a0073.html)

MasatoKubotera, 2022-08-08:
長期間運用されていないのでCloseします．

RikuYokoo, 2022-10-28:
# 現状の状態(10月28日時点)
12号館のCNCが不調で使えなくなる可能性があることから，オリジナルマインド性のCNCの現状の調査を行った．
動画，11号館に置いてある状態を確認したところ，CNCのz軸が分解されリニアガイドとリードスクリューが掃除されていることは確認した．
(リニアガイドはhttps://github.com/citbrains/SUSTAINA-OP/issues/118#issuecomment-1045923125 からリードスクリューは動画から確認できた．)
上記コメントにもあるがグリスの選定が必要である．

RikuYokoo, 2022-10-28:
## グリスの選定
https://github.com/citbrains/SUSTAINA-OP/issues/118#issuecomment-1049130741 を参考にグリスの選定を行った．
リニアガイド用のグリスがありhttps://www.monotaro.com/p/3924/9201/ を候補としてあげる．

yasuohayashibara, 2022-10-29:
オリジナルマインドにグリスを購入できるかを問い合わせました．
購入できない場合は推奨するグリスの型番を教えてくださいとの問い合わせを送りました．

yasuohayashibara, 2022-10-31:
本体に付属しているグリスは以下の型番とのことです．

エーゼット リチウムグリース40g DS764

NaokiIwasawa, 2022-11-24:
# グリスの給脂に向けた確認
@MasatoKubotera @RikuYokoo 
### 実施日
2022年11月24日

### 立ち合い
@RikuYokoo

上記であがっている**エーゼット リチウムグリース40g DS764**が届いたのでz軸のリニアガイドに給脂する方法を調べた。
[image]
[image]

## 給脂方法
[こちらの資料](https://tech.thk.com/upload/tec_data/RECOMMENDATION_For_correct_LUBRICATION_LM_guide_jp.pdf)によるとリニアガイドの側面にある小さな穴から給脂をするそうだが、その際に必要な「グリスニップル」「グリスガン」が手元になかった。そのため、穴サイズに合ったノズルを用意する必要がある。
[image]

リニアガイドに刻印されていた文字
CPC MR15M シリーズ　N+1.5 D306B L324^(2)　NV1 L426
[image]

穴の径：1.8 mm
[image]

[メーカのリニアガイドについてのPDF](http://www.yachiyo-corp.co.jp/wp-content/themes/yachiyo/file/cpc_Linear-Guide_jpn4.pdf)

## グリスニップルの必要性
工場の人にグリスニップルが必要か相談したところ、穴に給脂できれば問題ないのでなくてもいいとのことだった。
そのため、グリスガンの先端が十分に細ければ代用可能とのこと。
また、相談した際にグリスについて通常の2号（今回購入したもの）以外にもテフロンが入っているものがあるが、こちらの方が耐久性があり高速駆動する場所以外なら優れていると教えてもらった。また、スプレータイプもあるのでそちらの方がグリスアップしやすいとのことだった。

## グリスガンの代替品
100均でピストンスポイトが売っているそうなのでこちらが使えそう
外径が[こちら](https://jibundeyarou.com/cdinjctr/)の記事によると1.3mm程度と利用可能そうである

NaokiIwasawa, 2023-03-06:
# 仮組調査報告　※現在編集中
長らく報告が途絶えていたが仮組調査が完了したので報告する

## 実施期間
2023/01/17 ~ 2023/02/26
作業日合計：7日

## 担当者
@NaokiIwasawa @KentaTsukino @HarukiOgawa1  

## 作業方針
分解された状態から一度仮組をし機械が動作するか調査する
調査の過程で破損している部品があれば報告し必要ならば交換する
また，赤ネジ剤が使われた痕跡があったが今回の**組み立てでネジ剤は使用しない．**

## 問題と処置
調査が完了するまでに以下の問題を発見したのでしたのでその内容と実施した処置法を記載する

### 1. リードナットAの破損
### 2. セットピースの紛失
### 3. 丸ベルトの紛失
### 4. リニアガイド・リードスクリューのグリスアップ
### 5. 各部品の洗浄・ネジ交換
### 6. スピンドルカバーの取り付け小ねじ不足
### 7. 制御基板の設定
### 8. 制御基板の配線固定
### 9.  制御ソフトUSBCNCV4の起動方法
### 10. Y軸原点センサドグの調整

## 作業ポイント
組み立てをするにあたりいくつかポイントがあったので記載する

### 1. リニアガイドの脱落防止処置
### 2. z軸サポートユニットの取り付け
### 3. セットピース・セットスクリューの取り付け
### 4. リニアガイドのバックラッシ調整
### 5. Z軸ベースとX軸テーブルの固定
### 6. Z軸カバーユニットの取り付け
### 7. スピンドルユニットの取り付け
### 8. 丸ベルトの取り付け
### 9. Y軸テーブルと両サイドフレームの取り付け
### 10. 各軸・センサの動作確認

## 作業の記録
作業中の録画一覧をスプレッドシートに示す
https://docs.google.com/spreadsheets/d/1aH4EhZJqvbXRvLL1dqZueoWA_yiwHfFijrT3gsdIE2I/edit#gid=0
※チーム内管理

closed 2023-07-25 by MasatoKubotera