SUSTAINA-OP/SUSTAINA-OP #122: 163の再組み立て

NaokiIwasawa, 2021-07-26:
[163,165の共通報告](https://github.com/citbrains/OpenPlatform_ver3_0/issues/110)はこちらで行い
ここには163の詳細を記載する

NaokiIwasawa, 2021-07-26:
# 組み立て1日目

## メンバー

 責任者： @NaokiIwasawa 
作業者： @aiba0921 , @KentaTsukino 

## 作業目標
頭, 肩, 腕

KentaTsukino, 2021-07-26:
https://drive.google.com/file/d/13egoNFzIeY-SarHHxOnbi8qUVSUKDrfE/view?usp=sharing
左側のが今までのもので、右側のが予備のものです。この左右でバージョン違いが確認出来ました。
もともとが63mmで予備が59mmです。
結果的に予備パーツが足りなのでもともとのを使います。
予備パーツは2個ありますなので、6個再製作しないと統一が出来ません。

aiba0921, 2021-07-26:
カメラ_メインフレーム1_KBCR-S02MU
[image]
inventorで寸法を計測したところ11.45ｍｍだったのに対して実際に163に使用されていたパーツを計測した際には9.7ｍｍだったので、inventorのバージョン違いによる差異が発生していると思われます。
[image]
加えて、予備パーツに関しては赤で囲った部分の追加工がありませんでした。

KentaTsukino, 2021-07-26:
https://drive.google.com/file/d/1vmykX8yOYP5UdmD5MNnGGRDidTiyjml4/view?usp=sharing
カメラ外装
ねじ穴ががガバガバでねじが止まらなかったので、長峰の作ったものに交換しました。
長峰が作ったものは4つの穴がインサートナット化しています。

NaokiIwasawa, 2021-07-26:
左腕の組み立てを進めると1本M2.6x15のプラス鍋ネジがありました。
再組立ての際は新品の六角穴付きボルトに交換します。

aiba0921, 2021-07-26:
[image]
左ひじの出力ホーンの赤枠内のネジ穴がガバ穴になっていました。組み立ての際は一旦放置しサーボの書き込みの時に調整したいと思います。

aiba0921, 2021-07-26:
過去のissueでも報告されていたのですが、左腕のサーボホーンと部品を固定するネジが加工されており精度が非常に低いため今回は使用しませんでした。（下記はその時の画像のリンクです。）
https://user-images.githubusercontent.com/53966346/126086566-c526f2ba-95be-40a3-8b44-7106291c6945.jpg

NaokiIwasawa, 2021-07-27:
## 右腕の問題点
組み立てる際に問題点が2点ありました。

1. 上腕_サブフレーム_1_1がサーボホーンと干渉
→やすりで干渉部分を削り対応した
2. ひじのサーボ出力ホーンのネジ山が潰れて奥までネジが入らない
→サーボ交換をしました。但しサーボの書き込みはまだしていません。
＊右ひじID：9
[image]
潰れたネジ山の様子

NaokiIwasawa, 2021-07-27:
## 頭_サーボホーンフレームの交換

元々ついていたものはバージョン違いで高さが5mmであったが現行は8mmのため交換した

## 肩のスラストベアリング

左右ともベアリング内に隙間があり球が浮いていた。そのためシムを入れグリスアップを行いスムーズかつ隙間が空かないように調整した。

NaokiIwasawa, 2021-07-27:
## 1日目の進捗

目標としていた頭、肩、腕の組み立ては終わりました
全体として163はバージョン違いが目立ちました。しかし切削部材と時間が限られているため再利用して問題のなさそうなものはそのまま組み立てに使いました。
POMのネジ山などは潰れかけているものが多く電動ドライバーで締めるのは危険と判断し手で確認しながら締結しました。

NaokiIwasawa, 2021-07-28:
# 組み立て2日目

## メンバー

 責任者： @NaokiIwasawa 
作業者： @aiba0921 , @KentaTsukino 

## 作業目標
胴体, 付け根ユニット, 足首ユニット

NaokiIwasawa, 2021-07-28:
# 胴体上

## 新品ネジの利用

首の上サーボの固定：M2.6x15 x4　以前は十字鍋ネジだった
首の下サーボの固定：M2.6x20 x4　以前は十字鍋ネジだった

胴体_メインフレーム_1_13
もとはM2.6でタップ加工されていたため他機体同様のM3は合わない。しかし一部でM2.6では緩くなっている。
→M3として追加工し直して対応した→新品ネジM3x6mm x8本使用

aiba0921, 2021-07-28:
胴体_メインフレーム_1_1:1の赤枠で囲んだ部分がガバ穴になっていました。
[image]

aiba0921, 2021-07-28:
右肩サーボの固定に新しくM2.6の6mmの六角ネジを4個使用しました。

KentaTsukino, 2021-07-28:
左肩サーボの固定にボの固定に新しくM2.6の6mmの六角ねじを6個使いました。

KentaTsukino, 2021-07-28:
肩＿サブフレームの固定に＋ネジが使用されていたので、六角ネジに変更しました。個数は4個です。

KentaTsukino, 2021-07-28:
頭＿サーボホーンフレームを胴体の首サーボモーターに付けるときのM2.6×12mmが無かったので新しく4本追加しました。

KentaTsukino, 2021-07-28:
カメラ_メインフレーム:2と頭_メインフレーム:2を止めるネジがM2×4mmになっていたのでM2×6mmに変更しました。

KentaTsukino, 2021-07-28:
RearBumper_Lower_MIR:1のM3のナットが2個足らなかったので追加しました。

aiba0921, 2021-07-28:
2日目の進捗
[image]
上半身がほぼ完成しました。全体的に不適切なネジが使われている箇所がいくつかあり、ネジの交換を行いました。

aiba0921, 2021-07-29:
組み立て 3日目
---
メンバー
---
責任者： @NaokiIwasawa
作業者： @aiba0921 , @KentaTsukino

作業目標
---
機体完成

KentaTsukino, 2021-07-29:
左足のスラストベアリングで使われているSLCGが削れていて簡単に緩んでしまうので新しいのに4個交換しました。

aiba0921, 2021-07-29:
[image]
4.1.2.13.16で長穴減速_ホーン_2_1に低頭六角ねじ M3×5mmのネジで固定すると記載されているが実際に写真のようにこていするために使用したのは 低頭六角ねじM3×８mmだった。

aiba0921, 2021-07-29:
[image]
B3Mと足首_サブフレーム_1_1を締結していたM2.6の15mmの十字ネジを同じ規格の六角ネジに変更しました。

aiba0921, 2021-07-29:
足首メインフレーム_1_1の固定にM2.6の15mmの十字ネジが使われていたので代わりに同規格の六角ネジを4本使用しました。

aiba0921, 2021-07-30:
3日目進捗
---
右足の膝完成。左足の膝途中。
十字ネジから六角ネジへの交換が目立った。

aiba0921, 2021-09-23:
[image][image]
右足のフレームとリンク_筋交いを締結しているネジが接触して折りたためなくなっていた。

KentaTsukino, 2021-09-24:
[image]
モータの管理番号を確認したとこ、左足付け根ロールの番号が実際使っているものとスプレッドシート上で違っていました。

誤　→　70-1971　　　正　→　70-1929 
今使っているものは70-1929です。

MasatoKubotera, 2021-12-12:
１６３機については展示用などとし，今後の運用は行わない方針となりました．
役目を終了したため，closeします．

closed 2021-12-12 by MasatoKubotera