SUSTAINA-OP/SUSTAINA-OP #123: 165機体再組立て

Keita1Mori, 2021-07-26:
# 165機 再組立てメンバー

## 責任者
-  @Keita1Mori 

## 作業者
- @MiyawakiKaito 
- @TakuyaNagamine 
- @DaaFront 
 
163,165の共通報告は [こちら](https://github.com/citbrains/OpenPlatform_ver3_0/issues/110)に報告します。
ここでは165機体の再組立ての詳細を記載する。

Keita1Mori, 2021-07-26:
# 今日の作業目標
頭、肩と腕

DaaFront, 2021-07-26:
# 7/26の進捗

## 頭部
[image]

## 左腕
[image]

## 右腕
[image]

以上三ヵ所の組み立てが完了しました。

MiyawakiKaito, 2021-07-26:
サーボモータが二つ故障していました。一つ目のサーボモータは出力ホーンの軸が曲がっており、二つ目のサーボモータは回転が硬かったです。
[image]
[image]

TakuyaNagamine, 2021-07-26:
## 部品の状態・取り替え報告
・頭_サーボホーンフレームが歪んでいたので取り替えた.
[image]
・カメラ_外装が破損していたので取り替えた.
[image]
・頭_メインフレームを取り付ける際にM2の6mmねじがM2の8mmねじに変わっていたので取り替えた.
[image]
・頭_サブフレームを取り付ける際にM3の6mmねじがM3の8mmねじに変わっていたので取り替えた.
[image]
・腕先:1が破損していたので取り替えた.
[image]
・右肘ピッチのサーボモーターと右肩ロールのサーボモーターをサブフレームで取り付ける際にねじが曲がっているように見えた.
[image]

yasuohayashibara, 2021-07-27:
> 二つ目のサーボモータは回転が硬かったです。

ブレーキモード（B3MではHOLDを呼んでいるようです）になっていると，硬くなります．
フリーモードにして検証してみてください．

参考
https://kondo-robot.com/w/wp-content/uploads/B3M_SoftwareManual1.2.0.3.pdf

Keita1Mori, 2021-07-28:
# 7/28今日の作業目標
## 胴体と両足付け根
###作業者
- @Keita1Mori 
- @TakuyaNagamine 
- @MiyawakiKaito

MiyawakiKaito, 2021-07-28:
# ７月２８日
・低頭ネジの不足(３個)
[image]
[image]
[image]

TakuyaNagamine, 2021-07-28:
## 部品の状態・取り替え報告
・肩_メインフレーム_1_1を胴体_メインフレーム_1_12に取り付ける際に5mmの六角低頭ねじが3本不足していたので追加した.
[image]
・胴体_メインフレーム_1_13:1を使用する際に,歪みがあったが予備のパーツがないので利用した.
[image]
・左側の胴体_サブフレーム_1_9をサーボモータに取り付ける際にねじ穴が傷ついて使用できなかったので予備と交換した.
[image]
・胴体_サブフレーム_1_3:1を胴体_メインフレーム_1_12:1に取り付ける際に,胴体_メインフレーム_1_12:1のねじ穴が貫通しているもの(写真1)と,していないもの(写真2)があることが分かった.
・写真1
[image]
・写真2
[image]
・胴体_メインフレーム_1_12:1と胴体_サブフレーム_1_4:1を取り付ける際に,M3の低頭ねじが1つ足りなかったので追加した.
[image]
・肩のベアリングを取り付ける際にM2.6の6mmを取り付けるべきところにM2.6の8mmが取り付けられていた.
[image]

DaaFront, 2021-07-29:
# 7/29(三日目)

## メンバー
@Keita1Mori 
@MiyawakiKaito 
@DaaFront 
@TakuyaNagamine 

## 作業目標
機体完成
足の再組み立てを通して構造を理解する

TakuyaNagamine, 2021-07-29:
## 部品の状態・取り替え報告(7/29)
・左足のスラストベアリングを作成する際に,使用したシムの数は0.1mm,4枚であった.
[image]
・右足のスラストベアリングを作成する際に,使用したシムの数は0.1mm,1枚と0.05mm,1枚(岩沢先輩の削りあり)であった.
・右足のスラストベアリングを作成する際に,付け根ロール_1:1に明らかな歪みが見て取れたので交換した.
[image]
・右足のスラストベアリングを作成する際に,加工済みのSLCG5_5-6を取り付けていたところ **根元から折れた(2個)** ので交換した.
[image]
[image]
・左右の膝を作成する際に,付け根ロール_サブフレーム_6_5が歪んでいたので取り替えた.
[image]
[image]
・左足の膝を作成する際に,M2.6の10mm六角ねじが8本壊れていたので取り替えた.
[image]
・右足の長穴減速機構を作成する際に,長穴減速_ホーン_2が歪んでいたので取り替えた.
[image]
・左足を作成する際に,リンク_メインフレーム_6_3:1が歪んでいたので2枚取り替えた.
[image]
・左足を作成する際に,リンク_メインフレーム_6_1:1が歪んでいたので2枚取り替えた.
[image]
・左足を作成する際に,リンク_6_4:1が歪んでいたので2枚取り替えた.
[image]
・左足を作成する際に,リンク_6_2:2が歪んでいたので **1枚**  取り替えた.
[image]
・長穴減速_ホーン_2とARU-305_5を繋ぐねじが「低頭六角ねじ M3×8mm」が「低頭六角ねじ M3×5mm」になっていたので
取り替えた.
[image]

MiyawakiKaito, 2021-09-20:
## ９月２０日

部品の状態・取り換え報告
[image]
[image]

MibuchiYuta, 2021-09-21:
# 9/16 
## 問題点

右足首ピッチ軸のフリーホーンが歪んでいたため交換しました

[image]
[image]

yasuohayashibara, 2021-09-23:
こちら例のフリーホーンを逆に取り付けているというケースではありませんでしょうか？
詳しいことは野口君のサーボモータの修理に関するissueを御覧ください．

KiyoshiroKawanabe, 2021-09-25:
# 9/25 ケーブル問題報告  
おそらく電源基板→足付け根ロールで使用されていた ，長さ320㎜のケーブルの圧着が外れていました．
[image]

MiyawakiKaito, 2021-09-25:
## ９月２４・２５日

## 部品の状態・取り換え報告
## 報告①
リンク_メインフレーム_6_1_1に取り付けるベアリング(MF85FF)を壊してしまったため取り換えた。
[image]

## 報告②
膝_同軸_スペーサ_1_5のポケット加工の切削精度が悪かったので4つほど新しくした。
[image][image]

## 報告③
リンク_ふくらはぎ_サブフレーム_1_4とリンク_メインフレーム_6_2を締結するM2.6×10のネジが古くなっていたので2本交換した。
[image]

## 報告④
足首_サブフレーム_1_1と足首_メインフレーム_1_1_MIRを締結するM2.6×15のネジがなかったため、新たに4本追加した
[image]

## 報告⑤
足首_メインフレーム_1_1とサーボモータ(70-1941)を締結するM2.6×15のネジがなかったため、新たに4本追加した

<img src =
"https://user-images.githubusercontent.com/83056429/134763682-804c10f7-9c8b-4d23-985c-005a84b6d47e.jpeg)" width = "320px">
[image]

KiyoshiroKawanabe, 2021-09-28:
# 9/27 組み立て報告  
①長穴減速機構で使用されているブッシュがなかったので，加工済みのものを追加しました．  
②画像のはめ合が緩ったので交換しました．  
[image]

③左足の組み立て終わりました．

MasatoKubotera, 2021-12-12:
再組立てから時間が経過し，状態が変化してしまっているためcloseします．

closed 2021-12-12 by MasatoKubotera