SUSTAINA-OP/SUSTAINA-OP #124: 突入電流防止回路の作成

MasatoKubotera, 2021-07-26:
[GankenKunの電源回路の調査
#1332](https://github.com/citbrains/citbrains_humanoid/issues/1332)で指摘されている，突入電流を防止する回路を検証・実装する．

パワーサーミスタ，PchFETによるロードスイッチ回路で突入電流の対策を行い，DCDCコンバータの搭載廃止を目指す．

ノイズの要因でもあるDCDCコンバータを廃止することで頻発する電源故障問題を解決する．

MasatoKubotera, 2021-07-28:
サーミスタには，NTCとPTCがあり温度による特性が全く違うようだ．
参考：[NTC/PTCサーミスタの基礎](https://www.digikey.jp/ja/blog/basics-of-ntc-and-ptc-thermistors)

下記のように電源とコンデンサ間にNTCサーミスタを設置することで突入電流が防止できるようだ．

[アプリケーションノート 突入電流防止NTCサーミスタの使い方](https://product.tdk.com/ja/techlibrary/applicationnote/howto_ntc-limiter.html)より引用
>DC-DCコンバータにおける突入電流防止
DC-DCコンバータなどの直流電源回路においても、NTCサーミスタがパワーサーミスタとして用いられ、電源投入時に入力コンデンサや出力コンデンサを充電するときに流れる突入電流を効果的に制限します。通電後のNTCサーミスタの抵抗値はきわめて低くなるので、固定抵抗を使用した場合よりも電力損失が少なくてすみます。

[image]

しかし，
[突入電流制限用PTCサーミスタ](https://www.tdk-electronics.tdk.com/ja/385484/tech-library/articles/applications-cases/applications-cases/%E3%81%A9%E3%82%93%E3%81%AA%E5%A0%B4%E5%90%88%E3%81%AB%E3%82%82%E5%AE%89%E5%85%A8%E3%81%AB%E4%BF%9D%E8%AD%B7/911464)には，
>突入電流から回路を保護するICL：Inrush Current Limiter (突入電流リミッタ）としてNTCサーミスタが電源装置で使われることもありますが、より最適なソリューションはPTCサーミスタの利用です。特に過酷な温度や出力状態において、PTCサーミスタは、より信頼性の高い保護機能を提供することができます。

と記載されており，PTCサーミスタのするべきか検討する．

MasatoKubotera, 2021-08-11:
サーミスタにおいては現状の消費電流が不明確なため，今後の課題としたいと思います．

[【ロードスイッチとは】回路と原理について分かりやすく解説！](https://detail-infomation.com/load-switch/)，[【ロードスイッチの突入電流】『計算』や『問題点』などを図を用いて分かりやすく解説！](https://detail-infomation.com/load-switch-current/)に記載されている方法でロードスイッチのGS間にコンデンサを投入することでゲートソース間電圧を下げ突入電流を防止する設計を採用したいと思います．

MasatoKubotera, 2021-08-19:
上記の方法で回路に組み込みました．
検証については全体の回路の検証のissueを立ち上げて行います．
作成に関することは終了したため，このissueは閉じます．

yasuohayashibara, 2022-06-18:
@MasatoKubotera 
突入電流の問題ですが，こちらで対処できそうな気がしますので，現在使用しているFETの型番を教えてください．

MasatoKubotera, 2022-06-18:
@yasuohayashibara 
現在使用しているFETは，[SQM40P10-40L_GE3](https://www.mouser.jp/ProductDetail/Vishay-Siliconix/SQM40P10-40L_GE3?qs=jHkklCh7amg3sgsHIUrcaw%3D%3D)です．

その他の使用部品に関しては，下記のスプレッドシートに記載しています．
https://docs.google.com/spreadsheets/d/1DKiIKeoARyJN5wfXeKecXwp7EOx2dI6HBQyiw86vy40/edit#gid=1749995181

---
FETの在庫状況です．

[SQM40P10-40L_GE3](https://www.mouser.jp/ProductDetail/Vishay-Siliconix/SQM40P10-40L_GE3?qs=jHkklCh7amg3sgsHIUrcaw%3D%3D)	MOSFET P-Channel 100V AEC-Q101 Qualified　×48
[SUM110P06-08L-E3](https://www.mouser.jp/ProductDetail/781-SUM110P06-08L-E3)	MOSFET 60V 110A 272W 8.0mohm @ 10V　×30
[SUM110P06-07L-E3](https://www.digikey.jp/ja/products/detail/vishay-siliconix/SUM110P06-07L-E3/1656732)　MOSFET P-CH 60V 110A TO263　　×38

yasuohayashibara, 2022-06-18:
突入電流ですが，以下の部分にスイッチに直列に【10kΩの抵抗】を入れると解決する可能性があります．

[image]

抵抗を入れることでRC回路が構成されて，ゲートに加わる電圧がゆっくりと下がるようになります．
時定数は　10kΩ x 0.1uF = 1.0ms

最終的な電圧は-6Vですが，ゲート・ソース間しきい電圧は-2.5Vですのでスイッチングには十分です．

その間の電圧の変動は以下のようになりますので，0.1s程度かけて徐々に電流を流すことができるようになると思います．

[image]

その間，FETの抵抗値が高いため比較的多くの電力が消費されますが，最大許容損失が150Wなので問題無いと思います．

yasuohayashibara, 2022-06-18:
参考にしたサイト
※MOSFETで突入電流を低減できることを確信したという意味で

https://toshiba.semicon-storage.com/jp/semiconductor/knowledge/faq/mosfet/when-using-a-mosfet-as-a-load-switch-how-do-i-reduce-the-inrush-.html

https://toshiba.semicon-storage.com/content/dam/toshiba-ss-v2/master/en/semiconductor/knowledge/faq/mosfet/MOSFET-Gate-Drive-Circuit-Application-Notes_JA_36266.pdf

MasatoKubotera, 2022-06-18:
サーボモータ全19個を接続して，上記で提示されている抵抗（10k）を入れたときと現状（抵抗なし）でサーボの供給電圧を確認しました．

抵抗なし
https://youtube.com/video/56ubJD8ikOs
[image]

直列抵抗10k
https://youtube.com/video/2yNZLCjT4Lo
[image]

立ち上がりについてほぼ変わらないという結果です．
10k抵抗12V-10Aのプリセットでしたが，CCランプが一瞬点灯していたので10A流れていることを確認しました．

yasuohayashibara, 2022-06-18:
お疲れ様です．
突入電流ですので，立ち上がりが重要となりますが，上下の波形を見ると抵抗を入れたほうが緩やかになっています．
もっと緩やかにするためには，抵抗値を大きくすると良いですが，おそらく10kΩで十分だと思います．

あと，10Aの制限のある電源だとそもそも差が分かりづらいかもしれません．
バッテリで動かしているときに問題が発生するおそれが高いかと思いますが，抵抗無しの実験をするのはおすすめしませんので，いくつか抵抗を切り替えて比較すると良いかもしれません．

MasatoKubotera, 2022-06-20:
手持ちの抵抗で閾値電圧ギリギリまでロードスイッチに直列につなぐ抵抗値を大きくしてみました．
それぞれの抵抗値をサーボモータのロードスイッチにつないだ時の電圧波形を下記に示します．
青色がサーボモータ供給電圧，黄色が入力電圧（電圧計のXHコネクタに接続）になります．

20k抵抗以下だとかなりの電圧降下が確認できます．

33k抵抗では電圧降下がほぼないため，コンピュータボード（キャリアボードやJetson）の入力電圧生成に使っているDCDCを廃止することも出来そうです．

[image]「33k」
[image]「20k」
[image]「10k」

MasatoKubotera, 2022-06-20:
33kΩ・バッテリー駆動最低電圧10.8Vの時に，スイッチ投下するとゲートに7.82Vかかりゲート閾値2.5 VギリギリでON状態にはなりサーボに電力は加わりますがB3MのLEDが点灯説起動しないことがわかりました．

MasatoKubotera, 2022-06-20:
閾値電圧2.5Vを超えるためには，ロードスイッチに直列な抵抗の最大値は30k程度かと思います．
[image]

yasuohayashibara, 2022-06-21:
-2.5Vは電流が流れ始める電圧で，30kだと

-12V * 10kΩ/(10kΩ+30kΩ) = -3V

となり，最大電流は8A程度になります．
-3.5V以上にしたほうが安全です．

[image]

MasatoKubotera, 2022-06-27:
スイッチに直列で接続する抵抗について，20k抵抗を挿入したいと思います．
使用している抵抗は[カーボン抵抗（炭素皮膜抵抗）　１／４Ｗ２０ｋΩ](https://akizukidenshi.com/catalog/g/gR-03940/)です．

この抵抗については，下図のようにスイッチに直接取り付けています．
[image]

このスイッチはMainBoard No.5~の動作検証より実装しました．

※この抵抗を装着しなかったことによりFETが故障しているMainBoard No.1~No.4については応急的に10k抵抗を接続しています．

---

20k抵抗の時のゲート電圧は下記のようになります．

バッテリ最大時 　１３．０５ｖ
-13.05×10/(10 + 20)=-4.35

安定化電源駆動時　１２ｖ
-12×10/(10 + 20)=-4v

バッテリ最低時　１０．８Ｖ
-10.8×10/(10 + 20)=-3.6v

MasatoKubotera, 2022-07-02:
下記のissueで，ロードスイッチに直列で繋いだ抵抗値が20kの時，電流が流れにくくなるためサーボの電圧が降下する現象が発生しました．

https://github.com/citbrains/SUSTAINA-OP/issues/338#issue-1290428187

そのため，抵抗値を10kに順次置換します．

MasatoKubotera, 2022-08-08:
RoboCup2022までに対策したのは，ロードスイッチ（CPU電源，サーボ電源）部分の2か所のみです．
これに加えて，[低損失PowerPathコントローラIC LTC4412ES6](https://akizukidenshi.com/catalog/g/gI-07305/)を使って，2電源を切り替えしているFETのGateに同様にRC回路を投入すれば突入電流にバッテリー，スイッチング電源挿入時の突入電流についても対策できそうです．

低損失PowerPathコントローラIC LTC4412ES6を使って2電源を切り替えていますが，RoboCup2022を通してバッテリー2本のみでバッテリーからバッテリーに交換出来たらさらに利便性が高いかなと感じました．

それができそうなチップ
https://akizukidenshi.com/catalog/g/gI-07306/
https://akizukidenshi.com/download/ds/linear/j4417f.pdf

この，LTC4417のデータシート．突入電流の対策についてもかなり詳しく書かれているので大変参考になりそうです．

closed 2023-07-25 by MasatoKubotera