SUSTAINA-OP/SUSTAINA-OP #125: 電源基板のノイズ調査

YukiShigematsu, 2021-07-26:
現在使用している基板の電圧・ノイズの調査を行う。
これは、コンデンサの劣化等に伴うノイズの発生を早期に発見することを目的とする。
また、この調査は定期的に行うことが望ましい。

YukiShigematsu, 2021-07-26:
# 163機体
163機体の回路調査を行う際に電源回路（DCDCを含む）のノイズの調査を行った。
CPU側から出力される電圧に目立ったノイズは確認されなかった。

NaokiIwasawa, 2021-07-27:
回路のことは詳しくないのですが、ノイズ確認を行ったときの判断基準やデータを載せた方がいいと思います。
というのも、定期的に調査すると調査する人が変わる可能性があり、その時に判断が統一されない恐れがあります。

MasatoKubotera, 2021-07-27:
この調査では，どのバージョンの基板を使用したのか，どのように電源を供給し計測位置はどこなのかが重要かと思います．
また，その時の安定化電源のプリセットも記載してくれると嬉しいです．

ノイズがあるかどうかについてですが，電源投入時が一番乱れるポイントでもありますのでその時の波形も確認してください．

YukiShigematsu, 2021-07-29:
# 165機体
163機体の回路調査を行う際に電源回路（DCDCを含む）のノイズの調査を行った。
バッテリ交換時のACアダプタ
[image]
安定化電源
[image]
バッテリ交換時のACアダプタと安定化電源 同時
[image]
LIPOバッテリを除く、電源の状態で目立ったノイズは確認されなかった。

sekitorimisaki, 2021-07-31:
# 162機体
162機体の回路調査を行う際に電源回路（DCDCを含む）のノイズ調査を行った
バッテリ交換時のACアダプタ
[image]
安定化電源
[image]
バッテリ交換時のACアダプタと安定化電源同時
[image]

MasatoKubotera, 2021-08-02:
# 161機

JETSON TX2/J120を取り外したサーボも含めた回路構成で電源スイッチ・サーボスイッチON/OFF時のDCDCコンバータの入力電圧と出力電圧をオシロスコープで観察した．

黄色の波：入力
青色の波：出力

下図は，電源スイッチ投入（サーボ電源OFF）の波形である．
[image]

動画では，サーボ電源を入れると電圧降下が起きていることが確認できる．

https://drive.google.com/file/d/1elRql3TVT4Bs975ZsI_VtBjb2pRCmBz5/view?usp=sharing

161機の回路は，形状からver. 1.3であると考えられる．
下図は，メイン基板の回路図である．

[image]

yasuohayashibara, 2021-08-16:
お疲れさまです．
すでに見ているかもしれませんが，電源関連の問題は以下のissueで議論していますので，参考にしてください．

https://github.com/citbrains/citbrains_humanoid/issues/1329
https://github.com/citbrains/citbrains_humanoid/issues/1332

あと，見直して思い出しましたが，以下のコマンドでCPUに入力される電圧を監視することができます．
```
cat /sys/bus/i2c/devices/0-0040/iio_device/in_voltage0_input
```

問題の無い回路は負荷をかけても12V以下になることはありませんでしたが，CPUがリセットする回路は負荷をかけると12Vを下回ることがありました．

MasatoKubotera, 2021-08-17:
コマンドの詳細については，[Power Management for Jetson TX2 Series Devices](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/power_management_tx2_32.html)に記載がありました．

コマンドについての引用
>To read VDD_IN voltage, current, and power, enter the commands:
$ cat /sys/bus/i2c/drivers/ina3221x/0-0041/iio:device1/in_current0_input
$ cat /sys/bus/i2c/drivers/ina3221x/0-0041/iio:device1/in_voltage0_input
$ cat /sys/bus/i2c/drivers/ina3221x/0-0041/iio:device1/in_power0_input

資料によると電圧だけでなく電流もコマンドを叩くだけで計測できるようです．
162機で実際に叩いてみました．
[image]
電源投入直後の座らせた状態で計測をしました．
電圧：12072[mV]
電流：126[mA]
とかなり細かく読み取れた．

精度に関しては，上記のリンク先に
>Note:In terms of accuracy, assume a 5% guard band for INA measurements greater than 200 mW. Below that, accuracy can deviate by as much as 15%.

と記載があった．

このコマンドを用いて電流計測も行えそうです．