SUSTAINA-OP/SUSTAINA-OP #126: GankenKun全機体回路調査

MasatoKubotera, 2021-07-26:
GankenKun161~167機体においてDCDCコンバータは全機体共通であるが，そのほかの回路については統一されておらず，分解しなければどのような構造になっているかわからない状態である．

よって，このissueでは分解前にどのような回路配置になっているか，どのバージョンの基板をどのような配線でつながれているかを調査する．

YukiShigematsu, 2021-07-27:
# 163機体の回路調査

### 初期の各基板の配置
初期の各基板の配置は以下の写真のようになっていた。
また、バッテリーボックス上部に設置されているスイッチ基板はJETSONにUSB_Type_Bで接続されていた。
[image]

### 電圧表示基板
電圧表示基板の線の接続部は写真のようであり、保護は養生テープで行われていたが、断線等のリスクを減らすために、接続部の強化（接続方法の変更）と収縮チューブなどでの保護が必要と思われる。
[image]
[image]

### JETSON下部にある基板
外観ではわかりずらかったが、JETSONの下部に以下のような基板が配置されていた。（モータ書き込みのための基板？）
この基板は３Ｄプリンタ部品で固定されていた。また、JETSONへの接続方法はUSB_Type_A、反対側の接続先はXHコネクタ基板にXHコネクタで接続されていた。
[image]
[image]
[image]

[image]
[image]

### JETSON
JETSONはCPUを冷やすためのFANが埃を多く吸い込んで汚れていた。定期的な清掃が必要と考えられる。
また、FANの線が片方、断線の前兆がみられた。
[image]
[image]

### DCDC
写真の青線で囲まれているコンデンサのはんだが取れていた。（現在は半田済）
コンデンサが取れていた原因として、コックピット内に固定されずに入れられていたことが考えられる。
[image]

### メイン基板
他のISSUEで挙げられていたが、コンデンサが1つ取れていた。
メイン基板のコネクタのそれぞれの接続先は以下の写真の通りである。
[image]
[image]

### 組み立て後
・今回、組み立てるにあたってDCDCの位置をしたの段に持ってきた。
また、それに伴ってDCDCとJETSONをつなぐ線を延長した。
DCDCの位置を変更した理由は、DCDCが初期位置では固定されておらず、そのままの状態で機体を動かした場合にJETSONとDCDCをつなぐ線が断線する可能性があった為。
[image]
[image]
・基板をコックピット部品に固定する際、ねじ部分にスペーサを4枚挟んだ。
スペーサを挟んだ理由は、基板をネジで固定する際に基板の裏面に使われているグルーガンが原因で基板が歪んでしまい、ネジで確実な固定が行えなかったからである。
[image]

### コックピット全体から見えた問題点
①グルーガンと養生テープの多用
基板の部品やケーブルの固定にグルーガンが多く使用されていたため、接合部分などを確認することができなかった。
メンテナンス時や不具合発生時に迅速な対応ができるために、グルーガンの使用はなくすことが望ましいと考えられる。
②定期的なメンテナンスの実施
長期間内部基板の清掃が行われていなかったためか、基板上に多くの埃が蓄積していた。埃は、FANの冷却性能の低下や基板のショートの原因になる可能性がある。そのため、定期的なメンテナンスを実施する必要があると思われる。
③各基板の固定
現在、固定されている基板はDCDCとXHコネクタ基板が固定されていない。
これは、断線の原因や、基板の故障の原因になりうる。そのため、３Dプリンタパーツなどでこてーパーツを作り固定する必要がある。
④スルーホール実装の部品の使用
基板に使用されている部品のうち、コネクタ以外はすべて表面実装の部品が使用されており、OSコンのような大き目の表面実装部品は一部が取れてしまっていた。これを防ぐために、サイズの比較的大きい部品はスルーホール実装のものにし取れにくくする必要がある。

MasatoKubotera, 2021-07-27:
@YukiShigematsu 調査ありがとうございます．コメントを残します．

>バッテリーボックス上部に設置されているスイッチ基板はJETSONにUSB_Type_Bで接続されていた。

USB_Type_Bだと別の形状になってしまうので正しくは，micro USB Type-Bです．

---
JETSON下部の基板
外観ではわかりずらかったが、JETSONの下部に以下のような基板が配置されていた。（モータ書き込みのための基板？）
この基板は３Ｄプリンタ部品で固定されていた。また、JETSONへの接続方法はUSB_Type_A、反対側の接続先はXHコネクタ基板にXHコネクタで接続されていた。

この基板は，[RS485USB/シリアル変換アダプター](https://kondo-robot.com/product/02133)です．この機体では既製品を使用していますが，そのほかのほとんどの機体は自作して搭載されています．自作にするメリットとして，必要サイズが小さくなる，コストが1/6以下になるということが挙げられます．

---

電圧表示基板についてですが，現状は既製品である秋月の[超小型２線式ＬＥＤデジタル電圧計（パネルメータ）３桁表示　ＤＣ３～１５Ｖ（赤色）オートレンジ](https://akizukidenshi.com/catalog/g/gM-08468/)を使用していますが，この電圧計の欠点として，ケーブルがスルーホールに直はんだされています．GankenKunの故障でもよくここの断線は出てきており新規コックピット設計でも改善が要求されます．

現在の案としては，電圧計と[10段バッテリー残量LED表示器 [AY10C4A]](https://www.aitendo.com/product/18637)や[バッテリー残量表示器 [XW228DKFR4]](https://www.aitendo.com/product/16615)などの電圧以外で残量を可視化できるようにできれば，運用が簡単になるかと思っています．そのため，既製品の電圧計を時間があれば一から製作できれば良いなと思います．
自作することでコネクタ接続が可能になり断線が起こりにくくなるかと思います．（自作する余裕がなければ，現在の既製品を使用し，3Dプリントカバーでケーブルを固定し補強するなどの対策を取ります．）

---

DCDCの配置位置についてですが，DCDCコンバータはノイズの発生源となります．
J120のUSBの上部に配置されている機体もあり箇所はバラバラですが，信号線の近くに配置することは避けるのが一般的です．コックピットの設計上仕方ないことですが，USBやRS485付近には配置しないことが望ましいです．

---

>①グルーガンと養生テープの多用
基板の部品やケーブルの固定にグルーガンが多く使用されていたため、接合部分などを確認することができなかった。
メンテナンス時や不具合発生時に迅速な対応を行うために、グルーガンの使用はなくすことが望ましい。

まったくその通りだと思います．できれば収縮チューブについても廃止することが出来ればより不具合の発生時に迅速に対応できます．収縮チューブの品質・ケーブルとの相性にもよりますが一度収縮チューブをつけると外すのが大変でその部分を切ってしまうのが手っ取り早く，だんだんケーブルが短くなってきてしまいます．コネクタによっては，収縮チューブを必要としない[XT60H](https://store.shopping.yahoo.co.jp/suzakulab/amass-xt60h-m.html)などが存在します．

>②定期的なメンテナンスの実施
長期間内部基板の清掃が行われていなかったためか、基板上に多くの埃が蓄積していた。埃は、FANの冷却性能の低下や基板のショートの原因になる可能性がある。そのため、定期的なメンテナンスを実施する必要。

定期メンテの必要性を提案しています．[定期メンテの提案 #106](https://github.com/citbrains/OpenPlatform_ver3_0/issues/106)メカと同時期に行っていければと思います．埃の蓄積については，[エアダスター](https://www.yodobashi.com/product/100000001001431929/?gad1=&gad2=g&gad3=&gad4=452607784558&gad5=8163999086604200431&gad6=&gclid=Cj0KCQjw3f6HBhDHARIsAD_i3D8NjMalH4k7YWYzCfrAeQvfCJU7CN0JVKZJdFVig9im7zB0oXSkUDQaAt7BEALw_wcB&xfr=pla)を導入できれば楽に行えるかと思います．

>④スルーホール実装の部品の使用
基板に使用されている部品のうち、コネクタ以外はすべて表面実装の部品が使用されており、OSコンのような大き目の表面実装部品は一部が取れてしまっていた。これを防ぐために、サイズの比較的大きい部品はスルーホール実装のものにし取れにくくする必要がある。

サイズが比較的大きい表面実装が取れてしまう要因としては，ケーブルを詰め込みすぎていることが要因だと考えています．基板上にケーブルのあまりを押し込んでいる機体や，別基板が乗っているものなども見受けられこれらの要因をなくすことができればスルーホール部品を使用しなくてもよいかと個人的に思っています．

再設計中の基板では，コネクタ以外の部品はすべて表面実装のものを使用しており，表面実装にする理由としては，基板配線時に一層しか使用せずもう一層に干渉しない，サイズがスルーホールよりも小さいことが挙げられます．

YukiShigematsu, 2021-07-29:
# 165機体の回路調査

### 初期の各基板の配置
初期の各基板の配置は以下の写真のようになっていた。
また、バッテリーボックス上部に設置されているスイッチ基板はJETSONにmicro USB Type-Bで接続されていた。
[image]

### メイン基板
メイン基板のコネクタ配置は以下のようになっていた。
また、OSコンはスルーホール型が使用されていた。
[image]
[image]

### JETSON
JETSONはCPUを冷やすためのFANが埃を多く吸い込んで汚れていた。定期的な清掃が必要。
また、FANの線には収縮チューブが使用されていた。
[image]
[image]
[image]

### DCDC
DCDC基板はJETSONの下部に配置されており、DCDC自体も３Ｄプリンタ部品に固定されていた。
また、DCDCを固定する３Dプリンタ部品が一部破損していた。
[image]
[image]

### DCDC/JETSON/FAN 接続基板
DCDCの出力は写真のような基板を用いてJETSONとFANにつながっていた。
[image]

### IC基板
USBからの通信を変換してモータに伝達する基板と思われる。
[image]
[image]

### コックピット全体から見えた問題点
165のコックピットは基板の配置が比較的整っており、DCDC基板も固定されていた。
しかし、以下の改善点があると思われる。
①定期的なメンテナンスの実施
長期間内部基板の清掃が行われていなかったためか、基板上に多くの埃が蓄積していた。埃は、FANの冷却性能の低下や基板のショートの原因になる可能性がある。そのため、定期的なメンテナンスを実施する必要あり。
②基板のカバー作成
DCDCは固定されていたが、IC基板とDCDC/JETSON/FAN 接続基板は固定されておらず、絶縁のために養生テープが使用されていた。
養生テープを使用しないために、基板のカバーを作るか、固定を行う必要がある。
③ネジ落下防止の措置
JETSONの上部はCPU冷却のために大きく空間を設けていると思われるが、機体のネジなどが落下した際に、CPUFANに当たりFANが破損する可能性がある。
他機体ではこの対策として養生テープを張っているものもあるが、通気性確保のためにネットなどを配置するなど、排熱性を考慮した設計の改良が必要と考える。
④WIFIの固定方法の変更
現在、WIFIアンテナは結束バンドで固定されているが、結束性が弱い。
より確実な固定に設計を変更する必要がある。
また、現在アンテナが固定されている位置ではWIFIのアンテナ角度が制限されている。
WIFIアンテナ角度を制限しない設計への改良も考える必要がある。
⑤グルーガンの多用
電圧表示版を除き、基板への直接的なグルーガンの使用は見られなかったが、コネクタへは使用されていた。
メンテナンス時や不具合発生時に迅速な対応を行うために、グルーガンの使用はなくすことが望ましい。

RyodaiArai, 2021-07-31:
163機体，動作確認しました．

MasatoKubotera, 2021-07-31:
165機体については有線接続が出来なかったため，別issueを立ち上げました．
[165機 J120LED点滅/有線接続不可 #1538](https://github.com/citbrains/citbrains_humanoid/issues/1538)

sekitorimisaki, 2021-07-31:
# 162機体の回路調査

## 初期の各基板の配置

初期の各基板の配置は以下の写真のようになっていた。
また、バッテリーボックス上部に設置されているスイッチ基板はJETSONにmicro USB Type-Bで接続されていた。

[image]

# メイン基板

メイン基板のコネクタ配置は以下のようになっていた。
[image]

# JETSON
JETSONはCPUを冷やすためのFANが埃を多く吸い込んで汚れていた。定期的な清掃が必要。また、DCDCを固定するためにテープがはってあったせいで粘着部分がついていた。
[image]
[image]

# DCDC
JETSONの上に直接テープで固定されていた。
[image]

# IC基板
USBからの通信を変換してモータに伝達する基板と思われる。
グルーガンでカバーと固定されていた。
またコードの部分もグルーガンがついていた。
[image]  
[image]

# コックピット全体から見えた問題点

①定期的なメンテナンスの実施
長期間内部基板の清掃が行われていなかったためか、基板上に多くの埃が蓄積していた。埃は、FANの冷却性能の低下や基板のショートの原因になる可能性がある。そのため、定期的なメンテナンスを実施する必要あり。
②基板のカバー作成
DCDC基板、IC基板は固定されていなかった。また、DCDC基板は絶縁、固定のためにテープが使用されていた。
養生テープを使用しないために、基板のカバーを作るか、固定を行う必要がある。
③グルーガンの多用
基板やコードでグルーガンを使用しているところが多々見受けられたため改善が必要。

YukiShigematsu, 2021-12-03:
今後使用する予定の基板の回路の調査が終了したためissueを閉じます。
現在使用している基板のコネクタ配置の情報(171機 基板を除く)はgoogleドライブにあげています。
https://drive.google.com/file/d/1zYFyVboJP5pbvP2CFXoH20jXOjTT2Gfb/view?usp=sharing

closed 2021-12-03 by YukiShigematsu