SUSTAINA-OP/SUSTAINA-OP #127: スラストベアリングの六角スペーサーの既製品の加工について

TakuyaNagamine, 2021-07-29:
[165機体再組立て #123](https://github.com/citbrains/OpenPlatform_ver3_0/issues/123)に書いてある通り,右足のスラストベアリングを取り付ける際に, **既製品の6mmの六角スペーサーを3mmに加工したSLCG5_5-6** が **二回根元から折れた** .このことから既製品を無理に加工すると強度が弱くなる可能性が考えられる.
既製品に3mmの六角スペーサーは存在しないので,6mmの六角スペーサーにも対応する構造にする必要がある.

[image]

MasatoKubotera, 2021-07-29:
既製品の加工調査については，[既製品の追加工調査/改善 #28](https://github.com/citbrains/OpenPlatform_ver3_0/issues/28)でissueが立っているように無くしていきたい問題でもあります．

品番については正しくは，**SLCG5.5-6**かと思います．（少し違うだけでも検索にヒットしないので気をつけよう．Inventorのファイル名が正しい品番とは限りません．）

>既製品を無理に加工すると強度が弱くなる可能性が考えられる.

追加工について自分が把握してる限りでは，下記の加工方法で行っています．

- 161~166機：電光ペンチで切断．やすりがけはほぼ行っていない．
- 167機：リューターやルーターなどと呼ばれる工具で切断し，やすりがけ．

どちらも根元に力がかかる方法なので @TakuyaNagamine の指摘が要因として考えられると思います．
また片足立ち状態の時は，体重を子のスペーサ4本で支えていることになり，せん断力も働いていないのでかなり根元には負荷がかかっているものと思います．

>既製品に3mmの六角スペーサーは存在しないので,6mmの六角スペーサーにも対応する構造にする必要がある.

こちらについては，自分の方で再度探したところ[ステンレススペーサー（六角・短間段重ね用）/MSU](https://jp.misumi-ec.com/vona2/detail/221006202577/?KWSearch=%e3%82%b9%e3%83%86%e3%83%b3%e3%83%ac%e3%82%b9%e3%82%b9%e3%83%9a%e3%83%bc%e3%82%b5%e3%83%bc%ef%bc%88%e5%85%ad%e8%a7%92%e3%83%bb%e7%9f%ad%e9%96%93%e6%ae%b5%e9%87%8d%e3%81%ad%e7%94%a8%ef%bc%89%2fMSU&searchFlow=results2products&list=PageSearchResult)のラインナップで，SLCG5.5-6の雄ねじ部分が3mmである類似したスペーサ[MSU-306-03](https://jp.misumi-ec.com/vona2/detail/221006202577/?searchFlow=combo2products&KWSearch=MSU-300-03&HissuCode=MSU-306-03&PNSearch=MSU-306-03)がありました．スチール(SUM)製の[MSF-306-03E](https://jp.misumi-ec.com/vona2/detail/221006202397/?PNSearch=MSF-306-03E&HissuCode=MSF-306-03E&searchFlow=suggest2products&Keyword=MSF-306-03E&list=SuggestPreview)もありましたので強度を見て選定することになるかと思います．

MasatoKubotera, 2021-08-17:
[スラストベアリングの改良 #45](https://github.com/citbrains/OpenPlatform_ver3_0/issues/45)で告知をしたためこのissueは閉じたいと思います．

closed 2021-08-17 by MasatoKubotera