SUSTAINA-OP/SUSTAINA-OP #129: ネジの購入

NaokiIwasawa, 2021-08-04:
ネジ購入の方針を議論し注文をする。

NaokiIwasawa, 2021-08-04:
本issueを開いた時点で関連のあるissue
https://github.com/citbrains/OpenPlatform_ver3_0/issues/14
https://github.com/citbrains/OpenPlatform_ver3_0/issues/49
https://github.com/citbrains/OpenPlatform_ver3_0/issues/100

NaokiIwasawa, 2021-08-07:
# ネジ購入の方針決め

- 現在ある分は無視して全部再購入
- 16機体分～のネジ本数
- 全て[浅井ネジ](http://www.nejikouba.com/)の低頭で特注
- 浅井ネジの場合1,000本までは料金が変わらないので注文は最低1,000～とし、基本料金は5,000円である
- 使うネジは多めに買い直しがないように
-  ネジはけちる部分ではない
- ネジ頭の面幅・厚さ・頭部外径を統一
- 10種類程度に収める

NaokiIwasawa, 2021-08-10:
# 問い合わせ

## 見積もり依頼

方針に基づいた問い合わせ内容は以下の通り
・低頭平頭六角穴付小ネジ 2.6 x 6 ステンレス　1,500本
・低頭平頭六角穴付小ネジ 2.6 x 8 ステンレス　1,000本
・低頭平頭六角穴付小ネジ 2.6 x 10ステンレス　1,000本
・低頭平頭六角穴付小ネジ 2.6 x 15ステンレス　1,500本
・低頭平頭六角穴付小ネジ 2.6 x 20ステンレス　1,000本
・低頭平頭六角穴付小ネジ 3 x 6 　ステンレス　3,500本
・低頭平頭六角穴付小ネジ 3 x 8 　ステンレス　1,500本
・低頭平頭六角穴付小ネジ 3 x 10　ステンレス　1,000本
詳細は
M2.6：低頭キャップスクリュー形状 六角穴平径2 頭部外径φ4.8 高さ1.4
M3　：低頭キャップスクリュー形状 六角穴平径2 頭部外径φ5.5 高さ1.4
ステンレスの種類は何でしょうか？

---

## 浅井製作所からの返答
内容をまとめると

- M2.6は10mmまでしか製作できない
- 平径が軸部より大きく六角穴を深くできない
 → 首飛びの危険性がある
- 型を新規で作れば1.2mm以下の深さも可能だが別料金
- M3については頭部高さ1.2mmであればφ5.1で在庫対応可能
- ステンレスはSUSMX7

また、M3のみの見積もり結果は従来よりも安く入手できそうだった。

### 詳細
[ドキュメント](https://docs.google.com/document/d/1urzQCieh7g8BS05BBPje0UOD4B9opjkza1sUb4Yk-4w/edit?usp=sharing)を参照

NaokiIwasawa, 2021-08-11:
# 再問い合わせ

## 結果

- 鉄製であればM2.6であっても15, 20mmが製作でき、通常メッキ処理を施す。追加で焼き入れも可能（強度がステンレス以上になる）
→メッキはクロアエン（腐食対策）を選択
- M2.6の平径を1.5にすれば、問題なく作成できる
- M3の在庫品は頭の外径φ5.5である必要があるので利用できない
→六角穴平径2 頭部外径φ5.5 高さ1.4で依頼
- M3とM2.6合わせ全て揃うまでは**約７週間程度**はかかりそう（実働）

M2.6 x15, 20以外は、ステンレス(SUSXM7)製で製作

## 見積もり結果
[こちら](https://docs.google.com/spreadsheets/d/15jMa_qVc9eZrNs1h0gy4_dA9A1JH585WRgkWUkJKI4Q/edit?usp=sharing)を参照

NaokiIwasawa, 2021-09-03:
# 発注

8月22日に**発注完了**した。注文は下記に示す
・低頭平頭六角穴付小ネジ 2.6 x 6 　ステンレス　1,500本
・低頭平頭六角穴付小ネジ 2.6 x 8 　ステンレス　1,000本
・低頭平頭六角穴付小ネジ 2.6 x 10　ステンレス　1,000本
・低頭平頭六角穴付小ネジ 2.6 x 15　鉄製・クロアエン・焼入　1,500本
・低頭平頭六角穴付小ネジ 2.6 x 20　鉄製・クロアエン・焼入　1,000本
・低頭平頭六角穴付小ネジ 3 x 6 　ステンレス　3,500本
・低頭平頭六角穴付小ネジ 3 x 8 　ステンレス　1,500本
・低頭平頭六角穴付小ネジ 3 x 10　ステンレス　1,000本
詳細
M2.6：低頭キャップスクリュー形状 六角穴平径1.5 頭部外径φ4.8 高さ1.4
M3　：低頭キャップスクリュー形状 六角穴平径2 頭部外径φ5.5 高さ1.4

NaokiIwasawa, 2021-12-16:
# 製作過程での問題と実際に運用しての報告
製作過程において一部形状と材質を製作会社との相談の上変更したが、報告が遅れてしまい申し訳ないです。

## 変更内容
以下において適切な平径にて製作できず、1.5のレンチで回すとほぼ空転してしまうために形状変更の申し出を受けた
・低頭平頭六角穴付小ネジ 2.6 x 20　鉄製・クロアエン・焼入　1,000本
（・低頭平頭六角穴付小ネジ 2.6 x 15　鉄製・クロアエン・焼入　1,500本　※再製作の際に統一して変更した）

### 変更点
頭高さ1.4→1.1
穴深さ0.8→1.0

## 運用しての問題
上記の通り平径は対処してもらったが、チームが利用する電動ドライバで締め付けを行ったところM2.6（平径1.5）についてはトルク4（約0.6N･m）で完全になめってしまった。またそれ以下のトルクでもものによってなめりが発生してしまった。
特に材質の異なる15mm20mmは簡単になめってしまうため実装は出来ない状態である。 

## 届いたネジの写真
[image][image]
右：ステンレス製　左：鉄・クロアエン・焼入

NaokiIwasawa, 2021-12-16:
メールのやり取り詳細などは[こちら](https://docs.google.com/document/d/1urzQCieh7g8BS05BBPje0UOD4B9opjkza1sUb4Yk-4w/edit?usp=sharing)

MasatoKubotera, 2022-01-01:
第一陣での購入は終了したためCloseします．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/221 にて，第二陣の購入計画を行っています．

closed 2022-01-01 by MasatoKubotera