SUSTAINA-OP/SUSTAINA-OP #130: 頭_サーボホーンフレームにポケット加工を追加する

NaokiIwasawa, 2021-08-06:
現在、頭_サーボホーンフレームとサーボ出力ホーンの締結にのみ、M2.6 x12が4本使われている。しかし、この部分だけにネジを用意するのは非効率なので、再設計をしてネジの使用種類を減らしたい。
また、設計の注意点として

- 首の衝撃が集中する部分なので強度が必要

が上げられる。そのため設計ではポケット加工部分を新たに設け、その部分にサーボ出力ホーンを被せるようにする。
メリットは2つあり

- せん断応力についてはポケット加工でサーボ出力ホーンに3mm埋め込んだ方が強度が上がる
- ネジの種類を減らせ、購入費用・管理の手間を省ける

である。
再設計後は比較的よく使用するM2.6の10mmを利用
[image]

MiyawakiKaito, 2021-08-23:
サーボフレームホーンのポケット加工をする場合は図のような穴を中心とした円型のポケット加工をすればいいのでしょうか。
[image]

NaokiIwasawa, 2021-08-23:
上記で示された4つ穴でなく以下の画像の計測で示した部分です。
サーボの出力ホーンがすっぽり3mm埋まるようにポケット加工をしてください。
[image]

MiyawakiKaito, 2021-08-23:
[image]

MiyawakiKaito, 2021-08-23:
ポケット加工をしました。サーボホーンフレームにサーボ出力ホーンを3mm分はめ込めるように加工したので、ネジをM2.6 x12mmからM2.6の10mmに変更することが可能になります。

[image][image]

MasatoKubotera, 2021-09-02:
すべての工程が終了したためcloseします．

MasatoKubotera, 2021-09-26:
POM T8部材廃止に伴い，POM T5に変更する．
この時に，現状のはめあい方法ではサーボモータを締結するネジと干渉するために，2.5mmポケット加工を行いはめあう形に変更となっている．
これにより，転倒時の衝撃に耐えれるかは検証してみないとわからないが，それについてはカメラ外装からもアプローチして　衝撃の緩和をしたいと考えている．

[image]

MasatoKubotera, 2021-09-26:
上記の内容はhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/66#issuecomment-922168253 でも指摘していた．

MasatoKubotera, 2021-10-03:
修正が終了したためcloseする．

closed 2021-10-03 by MasatoKubotera