SUSTAINA-OP/SUSTAINA-OP #131: カメラ_メインフレーム・頭_メインフレーム ネジ径の変更

MasatoKubotera, 2021-08-06:
カメラ_メインフレームと頭_メインフレームを締結するネジに現状M2x6が使われているが，M3x6に変更する．
変更する理由としては，M2ねじを廃止するためである．

[image]

TakuyaNagamine, 2021-08-12:
@MasatoKubotera 
・**頭_メインフレーム**にあるねじ穴(図1)を7つから2つほどに減らしたいと考えているのですが,これは必要不可欠なものなのでしょうか?
・**頭_メインフレーム**と**頭_サブフレーム**(図2),**頭_メインフレーム**と**頭_サーボホーンフレーム**(図3)を繋げている部分もM3に変更しますか?
・図1
[image]
・図2
[image]
・図3
[image]

MasatoKubotera, 2021-08-13:
@TakuyaNagamine 
>頭_メインフレームにあるねじ穴(図1)を7つから2つほどに減らしたいと考えているのですが,これは必要不可欠なものなのでしょうか?

以前も同じようなisse[カメラ_メインフレーム・頭_メインフレームのネジ穴修正 #68](https://github.com/citbrains/OpenPlatform_ver3_0/issues/68)があり，そこでは
>頭_メインフレーム.ipt，カメラ_メインフレーム.iptにおいて現状使われていない穴が多数空いているが，これはカメラの角度を修正するものであり，大会などにより変更する可能性もあるため現状維持としたい．

となりました．しかし，角度を変更することは今までになく変更することはほとんどないため，塞いでもよいかと思われます．そのため，最低片面2か所あれば締結できると思うので減らしても大丈夫です．

---

>頭_メインフレームと頭_サブフレーム(図2)頭_メインフレームと頭_サーボホーンフレーム(図3)を繋げている部分もM3に変更しますか?

ここは現状M3ネジが用いられていたかと思うので現状のままで大丈夫です．

TakuyaNagamine, 2021-09-02:
## 報告
・カメラ周りの部品の再設計に伴い,カメラ_メインフレームと頭_メインフレームも再設計した.[(試作品データのリンク)](https://drive.google.com/drive/folders/1-ELrFyQRYSPTypYH1MUbgS2P9SowftKe)
## 写真
・カメラ_メインフレーム
[image]
・頭_メインフレーム
[image]

MasatoKubotera, 2021-09-05:
[カメラ_外装の再設計 #67 コメント](https://github.com/citbrains/OpenPlatform_ver3_0/issues/67#issuecomment-913198670) 

上記のissueで同時に統合を行いました．

@TakuyaNagamine 
問題が無ければcloseしてください．

TakuyaNagamine, 2021-09-06:
統合が終了したのでcloseします.

TakuyaNagamine, 2021-09-06:
統合が終了したのでcloseします.

MasatoKubotera, 2021-09-11:
@TakuyaNagamine 
再度確認したところ，Camera_vertical_frame_01（旧名：カメラ_メインフレーム）とCamera_horizontal_frame_01（旧名：カメラ_メインフレーム1_KBCR-S02MU）のはめあいが適切ではありませんでした．はめあいは片側0.01mmずつの間隔でお願いします．

また，Camera_horizontal_frame_01（旧名：カメラ_メインフレーム1_KBCR-S02MUのメスはめあい側が長さ15.1に対し，Camera_vertical_frame_01（旧名：カメラ_メインフレーム）が10.9であるためこれについても上記のように隙間は片側0.01mmでお願いします．

[image]

MasatoKubotera, 2021-09-11:
追加して，Head_vertical_frame_01，Camera_vertical_frame_01の上部角のフィレットを大きくしていただければと思います．（できる限り軽量化したいため）
無理やりφ５にしてしまいスケッチエラーが出ています．（手元環境下のみで実施し，アップしていない）

TakuyaNagamine, 2021-09-11:
@MasatoKubotera 
上部角とは下の写真の赤線部分のことでしょうか?
[image]
[image]

MasatoKubotera, 2021-09-11:
@TakuyaNagamine 
はい．そうです．φ2フィレット部分になります．

TakuyaNagamine, 2021-09-12:
@MasatoKubotera 
カメラ_メインフレームの再設計を再度行おうと思ったのですが,派生パーツに頭_メインフレームを選択したところ,ボディしか選択できませんでした.
原因が想像つかないのですが,どのようなことが考えられるでしょうか?
[image]

MasatoKubotera, 2021-09-12:
@TakuyaNagamine 
派生元のiptファイルのスケッチが参照できないときは，表示設定がされていなかったり，共有設定がされていない場合です．
その場合は，派生元ファイルの派生させたいスケッチを右クリックし，「スケッチを共用」をクリックしてください．
そのあとに，派生させることでスケッチを派生させることが出来るようになります．
[image]

頭_メインフレームについて，最新版では名前が変更になっています．最新のリポジトリ（フォルダ）をGitHubからcloneして作業を行ってください．

TakuyaNagamine, 2021-09-12:
@MasatoKubotera 
・Camera_vertical_frame_01 と Camera_horizontal_frame_01のはめ合いを修正しました.
・Head_vertical_frame_01 と Camera_vertical_frame_01 の上部角のフィレット部分をφ５にしました.
・[試作品データのリンク](https://drive.google.com/drive/folders/1-ELrFyQRYSPTypYH1MUbgS2P9SowftKe)(パーツ名はCamera_vertical_frame_01 と Head_vertical_frame_01)
[image]
[image]

MasatoKubotera, 2021-09-13:
@TakuyaNagamine 
迅速な修正ありがとうございます．はめあいの修正も大丈夫でした．

ver4_0ブランチに反映しましたので確認お願いします．
[image]

確認次第，closeしてください．

TakuyaNagamine, 2021-09-13:
確認できたのでcloseしたいと思います

MasatoKubotera, 2021-09-26:
再設計したときに下穴径が間違っていたので共有する．
また，ねじをCADモデルに反映する際に，ねじ頭がフィレットに干渉する問題も確認できた．

該当箇所についてはすでに修正済みである．
[image]
[image]

MasatoKubotera, 2021-10-03:
修正が終了したためcloseする．

closed 2021-10-03 by MasatoKubotera