SUSTAINA-OP/SUSTAINA-OP #135: GankenKun MainBoard ver2.0

MasatoKubotera, 2021-08-19:
MainBoard ver2.0についての設計，検証に関することを記載します．

[image]

MasatoKubotera, 2021-08-20:
## 新規回路基板の概要

- **USB Serial-RS485**：サーボモータ用の通信変換．既存の回路に加え，既製品に搭載されており，旧バージョンにはなかった終端抵抗を追加．
- **USB-HUB**：JetsonからのMini USB-Type B入力でRS485とUSB-Type A ×2Portに増設．将来的にオーディオ回路やバッテリの残量監視等が行えるようにするために新規搭載．
- **昇降圧(DCDC)**：中華既製品から移行．メイン基板に搭載することでケーブル配線を削減．半固定抵抗による出力電圧調節も不要に．
- **ヒューズ**：短絡など防止のために試験的に新規搭載．検証を行い使用するか判断する．
- **平滑コンデンサ**：旧バージョンよりも容量を拡大．より電圧降下が起きにくくなると考えられるが要検証．
- **ロードスイッチ**：突入電流防止，ボックススイッチの定格遵守のために新規搭載．定格は以前が10Aであるが140Aまで上昇．
- **2電源切り替え回路**：旧バージョンでは，安定化電源からLIPOバッテリに電源を落とさずに変更するために15VのDC電源を用いていたが，直接変更が可能に．LIPOバッテリよりも安定化電源を優先的に供給するように自動で切り替わる．

- **XHコネクタ統一**：旧バージョンでは，XH・VH・ZHR・PHが用いられていたがこれをすべてXHコネクタに統一し部品種を削減．
- **サイドからのサーボコネクタへのアプローチ**：旧バージョンでは，サーボケーブルが抜き差ししにくいという事例があったため，サイドからアプローチできるように変更．
- **配線ケーブルの削減**：基板を旧バージョンの最大4枚から1枚にまとめ，配線ケーブルを削減．
- **自作USBの廃止**：コネクタとシールドケーブルを用いた自作から既製品を用いられるようにコネクタをZHRコネクタからMini USB-Type Bに変更．

MasatoKubotera, 2021-08-20:
## データ管理

OpenPlatform_ver3_0は環境が整っていないため，現在は下記のリポジトリで管理しており，回路図（sch，brd），パーツリスト，ライブラリなどについてまとめている．
https://github.com/MasatoKubotera/GankenKun_Circuit_ver2_0

- [sch](https://github.com/MasatoKubotera/GankenKun_Circuit_ver2_0/blob/master/MainBoard/MainBoard.sch)
- [brd](https://github.com/MasatoKubotera/GankenKun_Circuit_ver2_0/blob/master/MainBoard/MainBoard.brd)
- [パーツリスト](https://github.com/MasatoKubotera/GankenKun_Circuit_ver2_0/blob/master/MainBoard/parts_list.md)
- [ライブラリ](https://github.com/MasatoKubotera/GankenKun_Circuit_ver2_0/tree/master/MainBoard/libraries)
- [ガーバーデータ](https://github.com/MasatoKubotera/GankenKun_Circuit_ver2_0/blob/master/MainBoard/MainBoard_2021-08-08.zip)

MasatoKubotera, 2021-08-20:
## sch
[image]

## brd
[image]

## preview
[image]
[image]

MasatoKubotera, 2021-08-20:
## 現状発覚している問題点

1. サーボコネクタとコンデンサ（C20，C21）が干渉．
2. サーボコネクタの配列が逆
3. サーボコネクタのピン役割を記したシルク印刷があっていない．
4. LEDを3つ搭載しているがその役割についてシルク印刷で記載がない．

MasatoKubotera, 2021-08-21:
## 部品の実装

[image]

表面実装部品は，PCBステンシルとホットプレートを使った簡易リフローで実装している．
実装方法については，[リフローマニュアル-Google ドキュメント](https://docs.google.com/document/d/1T1JqayP8ig-2tJKjySOk-GSw9ZU2BsN0X5cMatKDjuc/edit?usp=sharing)に記載している．

リフローの使用物品は下記の通りである．
- PCB（プリント基板）
- PCBステンシル
[image]

- [ホットプレート](https://www.amazon.co.jp/%E3%82%A2%E3%82%A4%E3%83%AA%E3%82%B9%E3%82%AA%E3%83%BC%E3%83%A4%E3%83%9E-%E3%83%9B%E3%83%83%E3%83%88%E3%83%97%E3%83%AC%E3%83%BC%E3%83%88-%E3%83%97%E3%83%AC%E3%83%BC%E3%83%88-%E3%83%96%E3%83%A9%E3%83%83%E3%82%AF-APA-136-B/dp/B07NCZCQP4/ref=asc_df_B07NCZCQP4/?tag=jpgo-22&linkCode=df0&hvadid=309236714101&hvpos=&hvnetw=g&hvrand=16777322081017079961&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1009320&hvtargid=pla-664506264852&psc=1&th=1&psc=1)
- [レーザー温度計](https://www.amazon.co.jp/gp/product/B07B4D9VWM/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1)
- [クリームはんだ](https://www.amazon.co.jp/gp/product/B078YJJVXC/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1)
- ヘラ
- マスキングテープ
- [ピンセット](https://www.amazon.co.jp/gp/product/B002F9MQN8/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1)
- つまようじ

リフロー時の様子は[動画](https://drive.google.com/file/d/1RoAUvNIFSkHbvKXsFi0oy2mlaQNbjnA7/view?usp=sharing)に残した．

MasatoKubotera, 2021-08-21:
## 基板発注

基板とPCBステンシルは，[ELECROW](https://www.elecrow.com/pcb-manufacturing.html?l=jp)で発注している．
発注の時のオプション設定は下図の通りである．
[image]

- **層数**：二層を選択．
- **寸法**：100x100mmまでに収めることで安く制作することが出来る．
- **製造枚数**：5の倍数枚数分は注できる．ELECROWでは多くの場合が発注枚数より1~3枚多く届く．
- **異種面付けの種類**：基本的に1種類．基板に同じレイアウトを並べるとき，変更する．
- **板厚**：1.6mmを選択．薄くすることで軽量化することが出来るが，その分曲げに弱くなり，パターン断線を起こすことにつながる．
- **レジスト色**：紫，マットブラックにするとさらに価格が上がる．
- **基板の表面処理**：特に指定せずデフォルトで最安のHASLで構わない．
- **端面スルーホール**：該当する方を選択する．
- - **銅箔厚**：電源部は大電流が流れるが部品点数がおおく，パターン幅の拡大に限界があるためオプションで1oz(35um)から2oz(70um)に変更している．
- **PCBステンシル**：Stencil15cmX15cm no frameの最安ので構わない．
生産時間：急がない限りは特急料金がかかるため指定しない．
- **国家**：JAPANを選択．
- **運送費用**：最安のSAGAWAを選択．7-10daysと記載されているが，発送通知の翌日，翌々日には届くことが多い．

ALLPCBでは，毎月1枚送料を含め無償で発注することが出来るので，動作を確認するときなどはこちらを利用するとよいが，PCBステンシルや銅箔厚のオプションは有料である．

MasatoKubotera, 2021-08-22:
## 発注前の確認

Eagleでガーバーデータを出力して基板の外注を行うが，出力したガーバーデータを可視化するためのソフトウェアとして，Gerbvを用いる．

使い方については，[Gerber viewerマニュアル](https://docs.google.com/document/d/1qlVQaaSNPMazrfmFupfyf-I9H9whuPHp6Oj2IfulcJw/edit?usp=sharing)に記載した．

[image]

MasatoKubotera, 2021-09-06:
## 使用するケーブルについて

- J120⇔MainBorad
[ViViSun USB 2.0 ミニケーブル USB(A)オス-USB(miniB)オス 同時L型 左右90°方向変換ケーブル 金メッキ付き 高速480Mbpsのデータ転送同期リード (右Ｌ)](https://www.amazon.co.jp/gp/product/B06ZY6D16R/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)：￥765×1
※USB HUBを用いる場合はこのケーブルで接続することを想定して設計している．

[image]

- J120⇔StartStopSwitch(Micro USB Type B対応バージョン)
[両側L型HOSTケーブル20cm MicroUSB 左L型(オス)-MicroUSB 左L型(オス) スマートフォン周辺機器接続ケーブル/USBMCH-LLL20【変換名人製】](https://www.amazon.co.jp/gp/product/B01G1ECBLO/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1)：￥558×1
[image]

- MainBoard⇔スイッチ，電圧計など
[耐熱電子ワイヤー　赤色　２０ｍ　導体外径０．５４ｍｍ（ＡＷＧ２４相当）](https://akizukidenshi.com/catalog/g/gP-11090/)
[耐熱電子ワイヤー　黒色　２０ｍ　導体外径０．５４ｍｍ（ＡＷＧ２４相当）](https://akizukidenshi.com/catalog/g/gP-11091/)
このケーブルを[ＸＨコネクタ　ハウジング用コンタクト](https://akizukidenshi.com/catalog/g/gC-12265/)で圧着し，使用する．
※スイッチは，安定化電源，LIPOバッテリの大電流が流れないロードスイッチ回路を用いている．

MasatoKubotera, 2021-12-01:
RCAP2021で2枚の基板が燃えた．燃えた個所はいずれも電源部のロードスイッチ用FETである．原因としてはデモ用に接続していたオーディオ回路部分がショートしたことが考えられる．

今後の対策として，すべての回路の絶縁と，外部機器それぞれにヒューズ/ポリスイッチの導入を行う．

また，基板サイドのXHコネクタも１か所使用不可となった．これはただ単にはんだミスの恐れがある．

安定化電源接続後，LIPOに切り替えしたときに，安定化電源側の接続確認用LEDが転倒しているのも気になる．

USB側にも電源・通信確認用のLEDの搭載を行うべき．

LIPOバッテリの接続の難易度も高く，安定化電源供給部も基板にコネクタを設置しているが，パネル取付型に変更したい．

RS485変換器を使うときに，通常はjetsonを通して通信しているが，メカの修理の際は，jetsonのUSBポートを取りはずしているため，新たにUSBコネクタメスを基板に設置し，USBの入力側切り替え回路を実装したいと考えている．

YukiShigematsu, 2021-12-11:
## RCAPから見えた改善点

- [ ] 基板発火の原因特定
- [ ] CPUボードスイッチが効かなくなった原因特定
- [ ] サーボ接続部コネクタ（XH 4P）への負荷軽減
- [ ] Li-po battery着脱が行いにくいため、これの改善

closed 2022-02-03 by MasatoKubotera