SUSTAINA-OP/SUSTAINA-OP #145: 切削パス作成

MasatoKubotera, 2021-09-02:


MasatoKubotera, 2021-09-02:
## 雛型について

パス作成を行う際に雛型となるtemplate.iamを作成しました．
OpenPlatform_ver3_0/cam 階層に保管しています．
[image]

雛型では，POM，A2017，GFRPのエンドミルφ2の2D輪郭，2Dポケットが用意されている．また，セットアップ(部材)については200x300x5を用いている．

T5部材ように設定されているので，部材厚を変更するためには，セットアップのストックタブから「高さ(Z):」を部材厚に変更する．
ボトム(-Z)からのオフセットが-1となっているのは，Disposable_Board(捨て板)を基準に構築しているからである．
[image]

エンドミルを変える場合は，工具ライブラリから選択することで変更することが出来る．

平面拘束が行いやすいように作業平面を設置している．これにより，視点を下方からの固定でもほぼすべての拘束が可能となる．作業平面と拘束するときには，作業平面の面ではなく，枠をクリックすることで選択することが出来る．
Disposable_Board(捨て板)をストックの基準として使っていて表示省略することが出来ない．そのため，捨て板上にある部品については下方からの拘束が難しいので枠外に移動させてから拘束する必要がある．

## 工具データについて
工具データに関しては，DT-30N.hsmlibにまとめた．
使用するときには，工具ライブラリからインポートすることで使用することができる．
切削の速度などを変えるときはこの部分を変更・更新することで，ほかのCAMにも容易に反映することが出来る．

[image]

MasatoKubotera, 2021-09-09:
## 切削パスの共有について

167機体製作時にGoogleドライブでパスの共有を行っていたが，環境の違いにより統合が困難であったため，今回は原則GitHubで共有を行う．

共有方法としては，全体でCAM開始前にブランチ**cam**を作成する．

### 環境構築

以下の手順で自分の環境構築する．
```
git clone https://github.com/citbrains/OpenPlatform_ver3_0
git branch -r
git checkout cam
```
### ファイル名について

パス作成後，\cam階層に「部材名_厚み_通し番号.iam」に形式に合ったファイル名で保存する．
部材名は下記の通りである．
「A2017」：通称アルミ
「GFRP」：通称ガラエポ
「POM」：通称ポム

厚みは「T1~T8」のいずれかである．

通し番号は「01」から順に振っていくこととする．

### リモートへの反映方法

Push（リモートへ反映）するフォルダの最上位層をエクスプローラで開き，アドレスバーに「cmd」を入力し，エンターキーを押す．
[image]

コマンドプロンプトが起動するので，自分のファイル名に合うように〇〇〇部分を変更しPushする．
commit の””内に「.(ドッド)」を含む拡張子を入れるとエラーが発生する．commitのメッセージについては特に指定は設けないが何が変更されたかわかるようにしておくこと．
git commit -m "add 〇〇〇"したときにメッセージが出るが，下記のように**1 file changed**であることを確認する．

> 1 file changed, 0 insertions(+), 0 deletions(-) 
>create mode 100644 cam/A2017_T5_001.iam  

git add するときに-Aオプションでリポジトリの変更があったファイルをすべてcommit対象にできるが，意図しないファイルが対象になることもあるので本工程では使用を禁ずる．

```
git pull
git add cam\〇〇〇.iam
git commit -m "add 〇〇〇"
git push origin cam
```

[image]

TakuyaNagamine, 2021-09-13:
@MasatoKubotera  @NaokiIwasawa 
camブランチにA2017_T2_01とA2017_T2_02をpushしました.
ご確認ください.

NaokiIwasawa, 2021-09-14:
@TakuyaNagamine 
A2017_T2_01とA2017_T2_02に関して間違えていた点は以下の通り

- セットアップでストックの厚さが5mmでした。T2なので2mmにする必要があります

その他自分が見る限り問題ないと思います。ある程度輪郭が分けられている点が良いです！

MasatoKubotera, 2021-09-15:
@TakuyaNagamine @HirokiHasegawa777 

GFRP T2 で切削予定だった**Slider_crank_horn**が部材変更となりA2017 T2になります．
対応お願いします．

yasuohayashibara, 2021-09-15:
> GFRP T2 で切削予定だった**Slider_crank_horn**が部材変更となりA2017 T2になります．
> 対応お願いします．

これは股ロール軸用の部品ですよね．そうだとして，
おそらくこちらの方が良い（少なくとも歩行にとっては）と思いますが，もしかしたら樹脂部分が衝撃吸収の役割をしていたかもしれません．
他の部品の摩耗が早くなったり，サーボモータのギアが壊れやすくなるかもしれませんので，しばらく注意深く観察したほうが良いと思います．
変更するといろいろと新しい現象に出会うことがありますので，どちらが適切かはしばらく動かして判断することが必要になります．

MasatoKubotera, 2021-09-15:
@yasuohayashibara 
下記のissueに変更になった理由を含め記載していますのでご確認ください．
[長穴減速機構の再設計 #11](https://github.com/citbrains/OpenPlatform_ver3_0/issues/11)

現段階でissueの更新が間に合っておらず更新中ですが，すべての軸において構成が大幅に変更になっております．
そのことを踏まえ，実機検証ではサーボ故障や摩耗期間などを含め注意深く観察したいと思います．

MasatoKubotera, 2021-09-15:
パス作成の分担や切削部品の部品表については，[210913_切削部品/分担表](https://docs.google.com/spreadsheets/d/1Eg0qYkxJzbA_DLEdSNvW-7wdzb3ONk1b4kYIxNlf8mA/edit?usp=sharing)に記載しています．

パス作成担当者については下記の通りです．

部材 | 部材厚 | 担当者
-- | -- | --
GFRP | T2 | @HirokiHasegawa777 
GFRP | T3 | @YugoNishio 
A2017 | T2 | @TakuyaNagamine 
A2017 | T3 | @KentaTsukino 
A2017 | T5 | @MiyawakiKaito 
POM | T5 | @aiba0921 

確認担当
- @NaokiIwasawa 
- @AD58-3104

KentaTsukino, 2021-09-15:
@NaokiIwasawa @AD58-3104 
A2017_T3_01というものをpushしたので、確認をお願いします。

NaokiIwasawa, 2021-09-15:
@KentaTsukino 
A2017_T3_01の確認をしたところいくつか問題がありました

1. 2D 輪郭1(A2017)：輪郭でポケット部分を選択しています(六角ナット形状部)。輪郭だと今回の設定上-0.1mm多く削ってしまうのでポケットに変更してください。
2. 2D 輪郭1(A2017)(2)：1.同様
3. 2D 輪郭1(A2017)(3)：一つ余分に輪郭を選択しています。
4. 2D 輪郭1(A2017)(4)：1.同様
5. Thrust_frame_01のポケット加工を忘れています。

切削する輪郭をブロックごとに分けているのはいい点です！

NaokiIwasawa, 2021-09-17:
@aiba0921 
A2017_T5_01の確認をしたところ問題があります。
切削の輪郭選択で同じところを何度も切削しています。今の状態だと2D ポケット1(A2017) (5)までで同じところを最大5回切削しています。うまく伝わっていなかったみたいですみませんが、1度選択（切削）した部分は次の選択では選択しないでください。

NaokiIwasawa, 2021-09-17:
@HirokiHasegawa777 
GFRP_T2_01
大きな問題はありませんでしたが、輪郭は基本最後にした方がいいです。
→2D 輪郭(GFRP) (4)と2D 輪郭(GFRP)を入れ替える

GFRP_T2_02
問題なし。いいですね！

NaokiIwasawa, 2021-09-17:
@aiba0921 
POM_T5_01確認しました。大きな問題ないです！
（ただ、こちら側の設定で輪郭に粗取りパスが適応されているのが気になりました。）
あと、POMのT5は治具の切削もあります。

NaokiIwasawa, 2021-09-17:
全体に対して
ポケットでDisposable_Boardまで削る場合と通常通りポケット加工する場合でボトム高さに使い分けが必要かもしれません。
Disposable_Boardまで削る場合→-0.1mm
通常のポケット→0mm
変更が必要になったら対応お願いします。（現在はポケットなら0mm,輪郭なら-0.1mmのボトム高さ）

TakuyaNagamine, 2021-09-17:
@NaokiIwasawa 
A2017_T2_01.iamとA2017_T2_02.iamをupdateしました.

MasatoKubotera, 2021-09-17:
@NaokiIwasawa 
template.iamに含まれるパラメータを変更しました．
変更箇所は下記の通りです．
- 切り抜き用の2Dポケット「2Dポケット切り抜き」を追加
  - 「2Dポケット」：部材厚み**未満**のポケット加工
  - 「2Dポケット切り抜き」：部材厚み**以上**のポケット加工
- 「輪郭」「2Dポケット切り抜き」の最大切込みピッチを「0.53mm」に変更
  ※部材厚-0.1mmまで切削するときにパスが1周多くなるなるため+0.03している．
- POMの「2D輪郭」で粗取りパスが適応されていたため排除．

[image]

NaokiIwasawa, 2021-09-17:
9/18時点でのコメント(commit 5273ccf)
@TakuyaNagamine 
2D ポケット1(A2017)のトップ高さがストックボトムになっています。これではシミュレーションした際に衝突エラーが出ます。
ストックボトム→ストックトップ
[image]

NaokiIwasawa, 2021-09-18:
9/18時点でのコメント(commit 5273ccf)
@KentaTsukino 
## A2017_T3_01
ポケットの使い分けできてない　混合で選択しています
切り抜きと通常のポケットがあるので選択を分けてください

aiba0921, 2021-09-18:
@NaokiIwasawa 
POM_T5_01のエラー修正の確認お願いします。

aiba0921, 2021-09-18:
解決しました。

closed 2022-01-20 by MasatoKubotera