SUSTAINA-OP/SUSTAINA-OP #146: [重要] 既製品パーツの新規導入について

hiroki-0001, 2021-09-03:
現在、[新たなGankenKunに使用する既製品のパーツリスト](https://docs.google.com/spreadsheets/d/1sIWYemisEyVsAWJagSuFfPlpzvkaceFB6knQFEcmL0Q/edit?usp=sharing)を作成しています。
**2021/09/03時点から新たに既製品パーツを追加・変更する場合は以下のテンプレートを参考に記入をお願いします。**

## [テンプレート]

issue名：

issue URL：

- 追加(または変更)された既製品パーツの型番
    - 購入先のURL

MasatoKubotera, 2021-09-06:
issue名：ハンドルの再設計 #19

issue URL：https://github.com/citbrains/OpenPlatform_ver3_0/issues/19#issuecomment-913394631

追加(または変更)された既製品パーツの型番：
AETTF8-135-M3-N3 → AETTF8-130-M3-N3
MSU-308-03（SLCG5.5-8）→　MSU-306-03
購入先のURL：
https://jp.misumi-ec.com/vona2/detail/110300208270/?PNSearch=AETTF8-130-M3-N3&HissuCode=AETTF8-130-M3-N3

使用個数：
AETTF8-135-M3-N3 → AETTF8-130-M3-N3 ×1
MSU-308-03（SLCG5.5-8）→　MSU-306-03×4

MasatoKubotera, 2021-09-06:
@hiroki-0001 

issue名：長穴減速機構の再設計 #11

issue URL：https://github.com/citbrains/OpenPlatform_ver3_0/issues/11#issuecomment-913704610

廃止
- [PACK-WSJJ8-6-5](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult) ×2個
- [PACK-WSJJ10-8-2](https://jp.misumi-ec.com/vona2/detail/110300235320/?PNSearch=PACK-WSJJ10-8-2&HissuCode=PACK-WSJJ10-8-2&searchFlow=suggest2products&Keyword=PACK-WSJJ10-8-2)×2個
- スペーサ：[ARU-305.5S](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-305.5S)×2個

新規
- オイレスブッシュ：[80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0603)×2個
- ジュラコンワッシャ：[CC-0612-10B](https://jp.misumi-ec.com/vona2/detail/221006204759/?PNSearch=CC-0612-10B&HissuCode=CC-0612-10B)×2個
- スペーサ：[ARU-305S](https://jp.misumi-ec.com/vona2/detail/221006202465/?PNSearch=ARU-305S&HissuCode=ARU-305S)×2個

MasatoKubotera, 2021-09-08:
下半身の大改良により，大幅に既製品パーツの使用に変更がある．
モデルはhttps://github.com/citbrains/OpenPlatform_ver3_0/tree/Mod_Lower_Bodyに挙がっています．
既製品フォルダを整理しましたので，使用していない既製品のiptは削除したつもりです．

MasatoKubotera, 2021-09-15:
こちらのissueについて，変更があまりにも多いため書き込みを行っていませんでした．
新規購入物品については直接スプレッドシートを編集し行っていきます．

closed 2021-09-15 by MasatoKubotera