SUSTAINA-OP/SUSTAINA-OP #150: FreundBumper_standの再設計

MasatoKubotera, 2021-09-09:
コックピット容量の拡大に伴い，前方胸バンパーを取り付ける箇所の幅が変更となった．
それに伴い，現状のFreundBumper_stand，FreundBumper_stand_MIRでは装着できなくなったため，再設計を行う．
この個所について大幅に変更する場合には，転倒実験が必要となる．
また，[TPU Bumper のMIRパーツ廃止 #149](https://github.com/citbrains/OpenPlatform_ver3_0/issues/149)のissueでも提言されているようにMIRパーツは廃止したい考えである．

TakuyaNagamine, 2021-09-20:
# FreundBumper_standの再設計の方針
humanoidのissue [TPU素材のバンパーを検討する #999](https://github.com/citbrains/citbrains_humanoid/issues/999)に書かれているが,FrontBumper_mainとFreundBumper_Standを組み合わせる角度は15度でなければ,認識に影響を及ぼす可能性があるのでFreundBumper_StandについてはMIRパーツを作成する必要があると考える.

TakuyaNagamine, 2021-09-20:
既存のFreundBumper_Standから幅が変更されたので,FrontBumper_mainがはまらなくなってしまった(下の写真).
これによってFrontBumper_mainとFreundBumper_Standを組み合わせる角度を変更するか,FrontBumper_mainを再設計する必要が出てきたので,検証しながら行う.
[image]

TakuyaNagamine, 2021-09-26:
試作品をドライブに上げました!
[リンク](https://drive.google.com/drive/folders/1iJgbNXFHRAudKng6OLf542O0TiOmsnWP)

MasatoKubotera, 2021-10-12:
@TakuyaNagamine 
次回から確認が大変なのでGitHubでブランチを切ってアップしよう．
対面でやり方教えます<(*_ _)>

NaokiIwasawa, 2022-01-05:
上記の前バンパーについて以前装着した際に外れやすいなど調整が必要だった気がしたのですがその後どうなりましたか？
記憶違いだったらすみません。

MasatoKubotera, 2022-01-05:
FreundBumperについての扱い
Ver4.1からは胴体の仕様が大きく変わるので流用不可になり，前方バンパーについては形状が大きく変化するのであらかじめお伝えしておきます．
Ver4.0についても部品どりのために解体予定です．
@NaokiIwasawa @TakuyaNagamine

TakuyaNagamine, 2022-01-05:
@MasatoKubotera 
了解しました

NaokiIwasawa, 2022-01-05:
了解です

closed 2022-02-01 by MasatoKubotera