SUSTAINA-OP/SUSTAINA-OP #152: 足首 はめあい導入

MasatoKubotera, 2021-09-15:
下図において，青色で選択されている2つの部品が現状ねじ3本で締結されているがこの部分いついてはめあいを導入したいと考える．後方についても同様に変更したいと考える．

この部分を変更すると治具についても変更となるので注意が必要である．

[image]

NaokiIwasawa, 2021-09-18:
# はめ合いの追加
現行は3点ネジ締結だったが強度を上げるためにはめ合い導入した

## 変更に当たって
現行モデルのスケッチでははめ合い部分の押し出し向きとスケッチ内容が合わなかったのでスケッチから作り直した
[image]
また、ネジの頭径を考慮するためネジ穴の位置は上面にオフセットした。

Keita1Mori, 2021-09-21:
# 報告
変更したはめ合いの部分の治具を作成しました。
[image]

MasatoKubotera, 2021-10-03:
再設計が完了し，対応した治具が作成されたためcloseする．

closed 2021-10-03 by MasatoKubotera