SUSTAINA-OP/SUSTAINA-OP #154: 足首ロール軸の再設計

NaokiIwasawa, 2021-09-15:
軸の構造をオイレスにし他の部分と基本設計を共通化するため足首ロール軸の再設計をする。

NaokiIwasawa, 2021-09-18:
# 再設計
他の軸と基本設計を共通化し部品種類を減らしメンテナンス性を向上させるために行う。
また、現行の固定されない六角ナットはメンテナンスの際一度外すと固定しずらいので改善する。
## 変更点
[image]←再設計後
[image]←現行

## 使用部品
外側から
[UUW-0312-10BH](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0312-10BH&HissuCode=UUW-0312-10BH&searchFlow=suggest2products&Keyword=UUW-0312-10BH&list=SuggestPreview) 36円 SUSワッシャー
[LFW-0815](https://jp.misumi-ec.com/vona2/detail/221000090734/?PNSearch=LFW-0815&HissuCode=LFW-0815&searchFlow=suggest2products&Keyword=LFW-0815&list=SuggestPreview) 58円 オイレス
A2017_T2
[80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?PNSearch=80F-0603&HissuCode=80F-0603&searchFlow=suggest2products&Keyword=80F-0603&list=SuggestPreview) 32円 オイレス
[CC-0612-25](https://hirosugi.co.jp/products/POM/CC.html) 11円　ジュラコンワッシャー
[UUW-0312-10BH](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0312-10BH&HissuCode=UUW-0312-10BH&searchFlow=suggest2products&Keyword=UUW-0312-10BH&list=SuggestPreview) 36円　SUSワッシャー
Ankle_front_frame
[S3-07-3W](https://jp.misumi-ec.com/vona2/detail/221000555662/?PNSearch=S3-07-3W&HissuCode=S3-07-3W&searchFlow=suggesthistory&Keyword=S3-07-3W) カレイナット
軸のスペーサー[BRU-307S](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-307S&PNSearch=BRU-307S&KWSearch=BRU-307S&searchFlow=results2products&list=PageSearchResult)

## 変更の影響
足首がx軸方向に1.5mm拡張
[image]

MasatoKubotera, 2021-10-03:
[image]
足首ユニットの修正の際，上記に記載されている
>変更の影響
足首がx軸方向に1.5mm拡張

について，変更前の寸法 105mmに修正した．

[image]

---

具体的な修正内容として，下図において青色で選択された部品をX軸方向奥側に1.5mmだけ移動している．
[image]

closed 2021-10-20 by MasatoKubotera