SUSTAINA-OP/SUSTAINA-OP #155: USB-RS485 変換基板の作成

MasatoKubotera, 2021-09-16:


MasatoKubotera, 2021-11-05:
Ver3.0機体においてUSB-RS485 変換基板は，下図の2種類が使われていた．
[image]「USB-RS485 変換基板と分配基板が一体化なもの」
[image]「USB-RS485 変換基板のみ」

これらの基板には欠点があり，その影響もありVer3.0機体ではたびたびサーボとの通信が途絶えることがある．

欠点は下記の5点である．
1. 基板がコックピットに固定されておらず，振動により基板が動きケーブルに負荷をかける
2. 自作USBを用いており，変換基板側には圧着の難易度が高いピッチ幅の小さいZHRコネクタが用いられ，逆側はUSBコネクタに自前ではんだ付けを行っておりグルーガンで固めてはいるが接触不良がたびたび起こっている．
3. サーボ分配基板に派生するケーブルコネクタがPHであり，圧着難易度が高く，確認したほとんどが圧着不足または圧力のかけすぎでコンタクトが変形している．
4. 終端抵抗がついていない．これについては効果はわかってが，最大5個のサーボを直列でつないでいるため信号劣化は起こりえないとは言えないためつけることが推奨される．
5. 自作USBにより，シールドが破綻しDCDCコンバータのノイズを拾いやすい環境であるが基板側で対策がとられていない．

一時的に純正の変換基板を用いていたが，結局は，自作USBを用いたり，コックピットに固定されていないため中古既製品交換後も約2週間で瞬断を起こし，戦略実行中にサーボ通信が途絶えたと報告を受けている．

これらを解決するために，Ver3.0.コックピットに搭載でき，かつ既製品のUSBケーブルを用いることができる基板を設計し発注した．名称は，USB-RS485_ConversionAdapter．バージョンはVer2.0である．

[image]「回路図」
[image]「Boardイメージ表面」
[image]「Boardイメージ裏面」

使用するマイコン等は現状と変えていないが，Mini USB type Bをさすことができ，フェライトビーズでのノイズ除去も導入している．

（LEDを付け電源確認やリセッタブルヒューズを入れることができればなおよかったが頭が回らなかった）

また，機体組み立て時などのサーボモータにパラメータなどを書き込みするときや，調査などでも使えるように，安定化電源のコネクタが直接させるようにXT60コネクタも実装できるようにしている．
（コックピット内に入れる用途では使用しない）

基板データについては
https://github.com/citbrains/OpenPlatform_ver3_0/tree/ver4_0/electronics/USB-RS485_ConversionAdapter_ver2_0
で保管している．

使用する物品については下記に記載するが，後日正式にまとめようかと考えている．
Part |Value|Package
---|---|---
C1,C3|0.1u|C1608
C2|4.7u|C1608
CN1||[B4B-XH-A](https://akizukidenshi.com/catalog/g/gC-12249/)
FL1||[BLM18SD220SN1D](https://www.mouser.jp/ProductDetail/Murata-Electronics/BLM18BA220SN1D?qs=8nh1BIkDD9%2F581tZMe7HGA%3D%3D&gclid=Cj0KCQjwrJOMBhCZARIsAGEd4VGoWabRCi4JGwtHERtbgCjjhZo_piyMdI8_WSK82iSZA7UUm2a5mxkaAii9EALw_wcB)
IC1||[FT232RL-REEL](https://akizukidenshi.com/catalog/g/gI-01739/)
IC2||[LTC1480CS8PBF](https://www.mouser.jp/ProductDetail/Analog-Devices/LTC1480CS8PBF?qs=hVkxg5c3xu9xGzCVotO6JA%3D%3D)
POWER||[XT60PB-M](https://store.shopping.yahoo.co.jp/suzakulab/amass-xt60pb-m.html)
R1,R3|[680](https://www.mouser.jp/ProductDetail/YAGEO/RC0603FR-10680RL?qs=%2Fha2pyFaduj6q4pkrQEncI4Ustthx1axNiNdHI4XIU734EbLEn51kw%3D%3D)|R0603
R2|[120](https://www.mouser.jp/ProductDetail/YAGEO/RC0603FR-10120RL?qs=%2Fha2pyFaduj6q4pkrQEncI4Ustthx1axVV0EaIkDoDgiUOuSTeNvrg%3D%3D)|R0603
R4|[10k](https://www.mouser.jp/ProductDetail/YAGEO/RC0603FR-1010KL?qs=%2Fha2pyFaduj6q4pkrQEncI4Ustthx1axVV0EaIkDoDgJ9nC9cILdvg%3D%3D)|R0603
USB1||[USB2066-05-RBHM-15-STB-00-00-A](https://www.mouser.jp/ProductDetail/GCT/USB2066-05-RBHM-15-STB-00-00-A?qs=KUoIvG%2F9IlZEo87a0wQR%252BA%3D%3D)

使用するUSBケーブル
[CERRXIAN 9インチミニUSBケーブルコンボミニUSB直角＆左アングルオスUSBタイプA 2.0直角男性データシンクロチャージケーブル（2パック](https://www.amazon.co.jp/CERRXIAN-CX-052-9%E3%82%A4%E3%83%B3%E3%83%81%E3%83%9F%E3%83%8BUSB%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB%E3%82%B3%E3%83%B3%E3%83%9C%E3%83%9F%E3%83%8BUSB%E7%9B%B4%E8%A7%92%EF%BC%86%E5%B7%A6%E3%82%A2%E3%83%B3%E3%82%B0%E3%83%AB%E3%82%AA%E3%82%B9USB%E3%82%BF%E3%82%A4%E3%83%97A-2-0%E7%9B%B4%E8%A7%92%E7%94%B7%E6%80%A7%E3%83%87%E3%83%BC%E3%82%BF%E3%82%B7%E3%83%B3%E3%82%AF%E3%83%AD%E3%83%81%E3%83%A3%E3%83%BC%E3%82%B8%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB%EF%BC%882%E3%83%91%E3%83%83%E3%82%AF%EF%BC%89/dp/B07CHN5KZV/ref=sr_1_42?__mk_ja_JP=%E3%82%AB%E3%82%BF%E3%82%AB%E3%83%8A&dchild=1&keywords=USB+minib+L%E5%AD%97&qid=1634699387&sr=8-42)

MasatoKubotera, 2021-11-05:
基板サイズは20x40 T1.6
穴は角から2.75x2.75の位置にありφ3.2

MasatoKubotera, 2021-11-09:
部品を実装し，動作確認（[動画](https://drive.google.com/file/d/13qzkq54MutDqqR6fiaXA2kgGmPLr7T7Q)）ができました．
１点だけ設計ミスがあり，XHコネクタの向きがシルク印刷と逆です．

基板は10枚発注しましたが，現状在庫のある部品を実装し，とりあえず5個製作しました．
[image]

底のカバーについては，基板データと同じ階層[BottomCover.ipt](https://github.com/citbrains/OpenPlatform_ver3_0/blob/ver4_0/electronics/USB-RS485_ConversionAdapter_ver2_0/BottomCover.ipt) にアップしました．

Ver3.0機体に装着すると下図のようになります．
[image]

RCAPまでに下記の機体に実装予定です．
- 161
- 162
- 164
- 165
- 166

yasuohayashibara, 2021-11-10:
とりあえず１つのロボットに搭載して試合形式（試合でなくても良い）で動かして問題無いかを確認してから全てに取り付けたほうが良いと思います．
最優先事項だと思いますので，みなさんじゃまにならない程度に久保寺君を助けて早めに解決したほうが良いと思います．

MasatoKubotera, 2021-11-19:
試験的に機体に取り付け検証を行い，問題がなかったため，下記の機体に搭載しました．
- 161
- 162
- 164
- 165
- 166

Ver4.0コックピット仕様の167機，171機についてはMainBoardにUSB-RS485基板が搭載されていますのでこの基板については搭載していません．

MasatoKubotera, 2021-11-19:
163機体については，今後の様子を見て実装するか判断したいと思います．
基板，部品については在庫があります．

MasatoKubotera, 2021-11-19:
各機体への実装が終了しましたので，closeします．

closed 2021-11-19 by MasatoKubotera