SUSTAINA-OP/SUSTAINA-OP #156: 167機体 Jetson 不具合検査・修理

YukiShigematsu, 2021-09-16:


YukiShigematsu, 2021-09-16:
## 状況と判断
167機体のセットアップ時（2021/9/16）にJetsonが起動しないことを確認。
この時、JetsonのチップLEDが実装されていなかったため、確認はJetsonのUSBポートにUSB機器を接続し、通電したかで確認を行った。
コックピッドを解体し、DCDCコンバータからは電圧出力を確認。
そのため、Jetsonの電源のはんだ付け部分に不良があると判断し、配線のはんだを外し確認を行った。

## 配線のはんだ部分の状態
（GND部分）
スルーホールは以前の配線と見られるものでふさがっており、それがねじ取り付け穴にも干渉していた。
これにより、GNDの金属部分がVCCに干渉しそうな状態となっており危険。
→ねじを伝って漏電しないように干渉している部分のねじは取り付けずに使用されていた。
また、それらは固まっており、除去を試みたが、除去できなかった。
（VCC部分）
片面のランドがはがれてなくなっていた。

## 判断と対処
以上のことより、はんだ付けによる電源接続は確実性がなく危険と判断。
はんだ付けの代わりにJetson J120の12Vのコネクタから電圧を供給する。

## チップLEDについて
LEDがないと起動確認の際に不便なため、緑色のチップLEDを実装した。
点灯するかの確認はまだ行っておらず、Jetson起動時に確認する。

YukiShigematsu, 2021-09-18:
# 現状報告
Jetsonにコネクタから電圧をかけ、LEDの点灯を確認した。
しかし、Jetsonの起動は確認できずUSBポートの通電も確認できなかった。

YukiShigematsu, 2021-09-18:
# 現状報告
jetsonへ12vコネクタから電圧をかけることで起動が確認できた。
ただ、起動に時間がかかる（最長30分ほど）
起動に時間がかかる原因はわかっていない

yasuohayashibara, 2021-09-18:
`dmesg`と入力すると，起動時に何が起きているかが分かります．
もし可能であればコピーして貼り付けてくれると助かります．
（そもそもコピーすることが難しいかもしれませんが．．．）

https://atmarkit.itmedia.co.jp/ait/articles/1612/20/news023.html

YukiShigematsu, 2021-09-18:
現在、再度Jetsonを起動しようとしましたが、３回行っても起動しなくなりました。（それぞれ４５分程度待っています）
３時間ほど前までは起動していたため、原因が不明です。

YukiShigematsu, 2021-09-18:
次回起動した際に<dmesg>のコマンドで確認を行います。

YukiShigematsu, 2021-09-20:
# 動作確認
安定化電源で12Vを供給したところ、LEDの点灯は確認しましたが、映像の出力（起動）は確認できませんでした。

YukiShigematsu, 2021-09-20:
# 今後の方針について
現在、症状を確認するために電源を入れることと切ることを繰り返していますが、Jetsonが正常に起動しない場合は電源を切る際にコマンドを使用せず、強制的に電源を落としているため、Jetsonの劣化につながることが考えられます。
また、同じ検証環境で起動を行っているのにかかわらず、起動するときとしないときがあるため、起動しない原因として、Jetsonのシステムファイルが壊れていることが考えられます。
そのため、Jetsonを一度リカバリーし起動するかを検証したいと考えています。

YukiShigematsu, 2021-09-24:
# 現状報告
先日、163機体のJETSONでHDMIを接続した状態で起動しようとしたところ、起動が確認できなかったがUSBポートとHDMIポートをすべて未接続の状態で起動したところ正常な起動が確認できた。
そのため、167のJETSONもポートに接続しない状態で起動し、その後HDMIを接続したところ画面の出力が確認できた。
また、その後の動作もフリーズすることはなかった。

YukiShigematsu, 2021-09-24:
# dmesg コマンドを実行して表示されたログ

[    0.000000] Booting Linux on physical CPU 0x100
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.38-tegra (peter@peter-ubu-dev) (gcc version 4.8.5 (GCC) ) #2 SMP PREEMPT Sat Sep 16 13:34:13 CEST 2017
[    0.000000] Boot CPU: AArch64 Processor [411fd073]
[    0.000000] earlycon: Early serial console at MMIO32 0x3100000 (options '')
[    0.000000] bootconsole [uart0] enabled
[    0.000000] Reserved memory: initialized node ramoops_carveout, compatible id nvidia,ramoops
[    0.000000] cma: Reserved 64 MiB at 0x00000000fc000000
[    0.000000] On node 0 totalpages: 2051072
[    0.000000]   DMA zone: 8192 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 519408 pages, LIFO batch:31
[    0.000000]   Normal zone: 23933 pages used for memmap
[    0.000000]   Normal zone: 1531664 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] PERCPU: Embedded 17 pages/cpu @ffffffc1f5cf4000 s31232 r8192 d30208 u69632
[    0.000000] pcpu-alloc: s31232 r8192 d30208 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 2018947
[    0.000000] Kernel command line: root=/dev/mmcblk0p1 rw rootwait console=ttyS0,115200n8 console=tty0 OS=l4t fbcon=map:0 net.ifnames=0 memtype=0 video=tegrafb no_console_suspend=1 earlycon=uart8250,mmio32,0x03100000 nvdumper_reserved=0x2772e0000 gpt tegraid=18.1.2.0.0 tegra_keep_boot_clocks maxcpus=6 androidboot.serialno=0320618012204 bl_prof_dataptr=0x10000@0x277240000 sdhci_tegra.en_boot_part_access=1 root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Memory: 7972600K/8204288K available (10824K kernel code, 1768K rwdata, 5652K rodata, 1156K init, 679K bss, 166152K reserved, 65536K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vmalloc : 0xffffff8000000000 - 0xffffffbdbfff0000   (   246 GB)
                   vmemmap : 0xffffffbdc0000000 - 0xffffffbfc0000000   (     8 GB maximum)
                             0xffffffbdc2000000 - 0xffffffbdc9d7c400   (   125 MB actual)
                   fixed   : 0xffffffbffa7fd000 - 0xffffffbffac00000   (  4108 KB)
                   PCI I/O : 0xffffffbffae00000 - 0xffffffbffbe00000   (    16 MB)
                   modules : 0xffffffbffc000000 - 0xffffffc000000000   (    64 MB)
                   memory  : 0xffffffc000000000 - 0xffffffc1f5f10000   (  8031 MB)
                     .init : 0xffffffc001098000 - 0xffffffc0011b9000   (  1156 KB)
                     .text : 0xffffffc000080000 - 0xffffffc001098000   ( 16480 KB)
                     .data : 0xffffffc0011da000 - 0xffffffc001394000   (  1768 KB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 64.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=6.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=64, nr_cpu_ids=6
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] Architected cp15 timer(s) running at 31.25MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xe6a171046, max_idle_ns: 881590405314 ns
[    0.000002] sched_clock: 56 bits at 31MHz, resolution 32ns, wraps every 4398046511088ns
[    0.009836] Console: colour dummy device 80x25
[    0.014494] console [tty0] enabled
[    0.018050] bootconsole [uart0] disabled
[    0.022149] kmemleak: Kernel memory leak detector disabled
[    0.022160] Calibrating delay loop (skipped), value calculated using timer frequency.. 62.50 BogoMIPS (lpj=125000)
[    0.022171] pid_max: default: 32768 minimum: 301
[    0.022214] Security Framework initialized
[    0.022243] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.022251] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.022604] Initializing cgroup subsys freezer
[    0.022615] Initializing cgroup subsys debug
[    0.022753] CPU0 ipc=752
[    0.022761] CPU1 ipc=1024
[    0.022769] CPU2 ipc=1024
[    0.022777] CPU3 ipc=752
[    0.022785] CPU4 ipc=752
[    0.022794] CPU5 ipc=752
[    0.022827] ASID allocator initialised with 65536 entries
[... 901 lines cut ...]
[ 3590.862283] xhci-tegra 3530000.xhci: exiting ELPG done
[ 3591.080391] usb 1-2: new low-speed USB device number 2 using xhci-tegra
[ 3591.222502] usb 1-2: feature bit otg_vbus_off set
[ 3591.227360] usb 1-2: New USB device found, idVendor=04f3, idProduct=0103
[ 3591.234220] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 3591.243605] usb 1-2: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[ 3591.252561] usb 1-2: ep 0x82 - rounding interval to 64 microframes, ep desc says 80 microframes
[... cut ...]
[ 3747.813597] eqos 2490000.ether_qos eth0: Link is Down
[ 3751.726866] eqos 2490000.ether_qos eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3827.761000] eqos 2490000.ether_qos eth0: Link is Down
[ 3832.022469] eqos 2490000.ether_qos eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3832.962190] eqos 2490000.ether_qos eth0: Link is Down
[ 3835.770761] eqos 2490000.ether_qos eth0: Link is Up - 1Gbps/Full - flow control rx/tx
cit@cit167:~/robocup/for2050/src/pyfiles [view_servo_voltage_log]$

YukiShigematsu, 2021-09-24:
# dmesg コマンドを実行して表示されたログ
[dmesg.txt](https://github.com/citbrains/OpenPlatform_ver3_0/files/7223086/dmesg.txt)

YukiShigematsu, 2021-09-24:
## 追記
ポートに何も接続しない状態で電源を入れたところ、２回目も正常な起動が確認できた。

yasuohayashibara, 2021-09-24:
正常に起動しているdmesgと比較してみると良いと思います．
私も詳しくないですが，デバイスが起動と停止を繰り返しているような気がします．（特に通信系）

YukiShigematsu, 2021-09-24:
わかりました。
正常な個体と比べてみます。

YukiShigematsu, 2021-09-25:
## ログの比較
163機体のJetson（正常）の起動時のログと167機体の起動時のログの比較を行いました。
結果、明確な以上は見られませんでした。

49956] CFG80211-ERROR) wl_cfg80211_connect : Connectting withcc:32:e5:5a:c4:8b channel (9) ssid "TP-Link_C48C", len (12)
[   49.736307] connect failed event=0 e->status 3 e->reason 0 
[   49.767819] CFG80211-ERROR) wl_bss_connect_done : Report connect result - connection failed

このログが167のJetsonで多く見られましたが、これはJetsonにwifiアンテナを付けていないことが原因と考えられます。
（163機体のJetsonにはwifiアンテナが接続されていました。）

YukiShigematsu, 2021-09-25:
# 現状報告と今後の方針
現状、167機体のJetsonで見られる不具合は以下の通りです。
・HDMI端子を接続した状態で起動すると画面で起動が確認できない。
→ほかのJetsonでも共通する症状とみられる（原因は不明）
・LANによる接続が行えない
→LANケーブルを接続した際にLEDの点灯は確認できたため、設定の問題の可能性大

この二つの症状が同時に見られたため、画面による起動の確認が行えなくなり、故障したと誤認したと考えられる。
現状、これまででフリーズなどの症状は見られていないため、機体による運用は可能であると思われる。

closed 2021-09-27 by YukiShigematsu