SUSTAINA-OP/SUSTAINA-OP #158: 故障と判断されていたJetsonの使用可否の判断

YukiShigematsu, 2021-09-18:


YukiShigematsu, 2021-09-18:
# 現状報告
これまで故障と判断されていたJetsonが起動したため、運用が可能か検証する。

Jetsonに12v電圧をかけ、HDMIから映像の出力が確認された。
しかし、起動には30分ほどかかる。

また、リカバリモードに切り替えようとすると切り替えが行えない。
現在、スイッチを押す時間などを調整しながら切り替えを試みている。

## 現在判明している問題点
- 起動が異常に遅い
- リカバリモードへの切り替えが行えない
- USBポートが片方使用不可
- JetsonとJ120を固定するネジが4本ともない

MasatoKubotera, 2021-09-18:
@YukiShigematsu 
J120にJetson TX2共に，どこかしらに管理番号をテプラで張って管理したいと考えています．
タイミングが出来ればお願いします．

MasatoKubotera, 2021-09-18:
>USBポートが片方使用不可
USBポートが片方使用不可とのことですが，上段下段について分かれば記載してください．

>リカバリモードへの切り替えが行えない
検証環境が分かりませんが，J120のスイッチ故障も考えられます．

今後は検証環境等の写真・動画を添付するようにお願いします．

YukiShigematsu, 2021-09-18:
了解しました。
USBポートの上段、下段についてと、検証環境は後日写真を添付して投稿します

YukiShigematsu, 2021-09-20:
# 動作検証報告
## 経緯
Jetsonで　dmesg　のコマンドでの確認を行うために起動を試みた。
# 検証環境
[image]
モニタ：HDMI出力
使用ボード：Jetsonのきっとに付属していたボードを使用

## 症状
起動は成功したが、dmesgコマンドを実行したところフリーズしてしまい、画面が写らなくなった。

YukiShigematsu, 2021-09-20:
# 今後の方針について
現在、症状を確認するために電源を入れることと切ることを繰り返していますが、Jetsonが正常に起動しない場合は電源を切る際にコマンドを使用せず、強制的に電源を落としているため、Jetsonの劣化につながることが考えられます。
また、同じ検証環境で起動を行っているのにかかわらず、起動するときとしないときがあるため、起動しない原因として、Jetsonのシステムファイルが壊れていることが考えられます。
そのため、Jetsonを一度リカバリーし起動するかを検証したいと考えています。

closed 2021-12-03 by YukiShigematsu

yasuohayashibara, 2021-12-03:
@YukiShigematsu 
次のissueに引き継いだのであれば，そちらを参照してください．
もしこれ以上調査を行わないのであれば，その旨記載してください．

YukiShigematsu, 2021-12-03:
@yasuohayashibara 
すみません。報告を行っていませんでした。

上記の2つのjetsonでリカバリーモードに入ろうとしましたが、両機ともリカバリーモードに入ることができませんでした。
そのため、2つのjetsonは故障と判断しました。
現状でこの2つのjetsonの使用予定がないため一旦issueを閉じます。
また、jetsonの修復に関しては修復を目的としたissueを別に立てます。