SUSTAINA-OP/SUSTAINA-OP #159: CADモデルにねじを追加する

MasatoKubotera, 2021-09-18:
Inventor CADモデルに[ネジの購入 #129](https://github.com/citbrains/OpenPlatform_ver3_0/issues/129)で購入した浅井製作所のねじを反映する．
ネジを挿入することで，重量を製作前にある程度予測でき，干渉しているかどうか確認できる．

MasatoKubotera, 2021-09-18:
Inventorの機能として，[詳細レベル リプレゼンテーション](https://knowledge.autodesk.com/ja/support/inventor/learn-explore/caas/CloudHelp/cloudhelp/2020/JPN/Inventor-Help/files/GUID-BB22DDCB-F370-4BF0-B438-EC1D0E91D49E-htm.html)があり，この機能を用いることでパソコンスペックが低くても閲覧できるモデルを作成することが出来る．

MasatoKubotera, 2021-09-24:
切削前にすべてのネジを挿入して干渉が起こらないか確認する必要がある．

yasuohayashibara, 2021-09-24:
傍から見ていてすでに十分に議論されているような気がしますが，できるだけ同じネジで組み立てられるようにしていただければと思います．ネジの種類が多くなると組み立てミスも多くなりますので，この期に組み立てやすいメカニズムにしていただければと思います．（選定時に少し配慮するぐらいで，設計を変更する必要はありません．）

あと，長いネジを使用すると強度もそうですが，緩みにくくなりますので，ネジの種類が増えすぎない範囲で緩みやすそうなところは長いネジを使用してあげると良いと思います．

すでに決定しているところも多いと思いますので，特にこれから変更する必要はありませんが，まだ検討するところが残っているとしたら，少し気にかけていただければと思います．

MasatoKubotera, 2021-10-12:
上記の指摘を思考し，ねじをモデルに挿入した．
[image]

使用するネジは，下表に示す．

|ねじ種類/型番|本数|
|-- | --| 
M2-6 | 4
M3-6 | 226
M3-8 | 50
M3-10 | 22
M2_6-6 | 80
M2_6-8 | 4
M2_6-10 | 32
M2_6-15 | 72
M2_6-20 | 68
SPNFL-BRN-M3-6 | 2
FreeHorn_Screw | 10

コミット[148f7fd7893b13692ae1fc8c9f0d661640f56b17](https://github.com/citbrains/OpenPlatform_ver3_0/tree/148f7fd7893b13692ae1fc8c9f0d661640f56b17)時点でのねじ数量※Inventor部品表より

M2-6は，既製品のカメラ基板を締結するためのネジであり基板を変更しない限りは削減不可なものである．
このネジについては，在庫で対応するため新規購入しておらず，**低頭ではない**六角穴付きボルトを用いる．

M2_6-8 は，下図に示すが首サーボの出力ホーンへの締結の4本しか使用されていない．現状の形状では6mm，10mmのネジを挿入することが出来ないため，形状を変更し削減したいと考えている．
[image]

SPNFL-BRN-M3-6は，コックピットと胴体の両側にある飾りネジと呼ばれるローレットビスである．
[image]

サーボモータのフリーホーン軸ネジ（FreeHorn_Screw）についても挿入している．
サーボモータの出力ホーン軸のネジも使用しているが，サーボモータに必ず付属しているものとしてモデルには挿入していない．
[image]

MasatoKubotera, 2021-10-12:
上記の反映では，素材設定はしているが正確な質量にはなっていない．
そのため，電子量りipropertyに反映する必要がある．

MasatoKubotera, 2021-11-04:
作業が終了しているためcloseします．

closed 2021-11-04 by MasatoKubotera