SUSTAINA-OP/SUSTAINA-OP #162: ver2.0 MainBoardでUSB-HUBが認識しない

MasatoKubotera, 2021-09-22:
windowsでは動作化確認していたUSB-HUBがJetsonのubuntu環境では動作しなかった．

hiroki-0001, 2021-09-22:
lsusbコマンドで、デバイスの状態を確認したところ、認識すらされていなかったので、ドライバーをインストールする必要があると思われる。

hiroki-0001, 2021-09-25:
dmesgコマンドを用いて、デバイスの認識を確認した結果以下のような出力が見られた。
```
[ 2\u670813 03:51] xhci-tegra 3530000.xhci: exiting ELPG
[  +0.050705] xhci-tegra 3530000.xhci: Firmware timestamp: 2017-03-31 06:21:56 UTC, Version: 55.07 release
[  +0.013143] xhci-tegra 3530000.xhci: exiting ELPG done
[  +0.218501] usb 1-2: new low-speed USB device number 24 using xhci-tegra
[  +0.120202] usb 1-2: device descriptor read/64, error -71
[  +0.220012] usb 1-2: device descriptor read/64, error -71
[  +0.219786] usb 1-2: new low-speed USB device number 25 using xhci-tegra
[  +0.120208] usb 1-2: device descriptor read/64, error -71
[  +0.219997] usb 1-2: device descriptor read/64, error -71
[  +0.219795] usb 1-2: new low-speed USB device number 26 using xhci-tegra
[  +0.008083] usb 1-2: Device not responding to setup address.
[  +0.209004] usb 1-2: Device not responding to setup address.
[  +0.090070] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 5
[  +0.007545] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work ignore firmware MBOX_CMD_DEC_SSPI_CLOCK request
[  +0.109308] usb 1-2: device not accepting address 26, error -71
[  +0.116001] usb 1-2: new low-speed USB device number 27 using xhci-tegra
[  +0.008032] usb 1-2: Device not responding to setup address.
[  +0.209209] usb 1-2: Device not responding to setup address.
[  +0.206750] usb 1-2: device not accepting address 27, error -71
[  +0.006117] usb usb1-port2: unable to enumerate USB device
[  +0.217897] xhci-tegra 3530000.xhci: entering ELPG
[  +0.008081] xhci-tegra 3530000.xhci: entering ELPG done
[ +16.831813] xhci-tegra 3530000.xhci: exiting ELPG
[  +0.052483] xhci-tegra 3530000.xhci: Firmware timestamp: 2017-03-31 06:21:56 UTC, Version: 55.07 release
[  +0.013151] xhci-tegra 3530000.xhci: exiting ELPG done
[  +0.218487] usb 1-2: new low-speed USB device number 28 using xhci-tegra
[  +0.120204] usb 1-2: device descriptor read/64, error -71
```

この出力結果から、以下のことが把握できる

1. カーネルはデバイスの認識をすることができている
2. **device descriptor read/64, error -71**のエラーからデバイスのデータの読み込みができていない(通信速度が違う・ポートorデバイスの故障・電力不足などが原因？？)
3. **xhci-tegra 3530000.xhci: tegra_xhci_mbox_work ignore firmware MBOX_CMD_DEC_SSPI_CLOCK request**のエラーと似たようなもの症状の対処が書いてある[フォーラム](https://forums.developer.nvidia.com/t/tx2-usb-3-0-otg-as-host-error-message/58113)を見つけた、要確認

これらのことから、システム面での解決は非常に困難そうなので、同時に回路方面からもアプローチしていただけると助かります

hiroki-0001, 2021-10-08:
この問題だが、JetPack4.3のJetson(現在166に使用中)のものにJ120を搭載した上で、lsusbコマンドでUSB Hubの認識を確認したところ、認識が確認された。

[image]

**Bus 001 Device 009: ID 1a40:0101 Terminus Technology Inc. Hub**が認識することができたデバイスである。
このデバイスは新規電源回路に搭載されているUSB HUBの**FE1.1S**を示すものである。

配線に関しては以下の画像のように行った。
[image]

J120のUSB HUBからUSB A to mini Bのケーブルを用いて、新規電源回路に接続しました。

以前は認識することができなかったのに、今回は新規電源回路のUSB HUBがしっかりと機能している原因として考えられるのは、

1. **回路側のミス**。今回使用した新規電源回路の中でも、久保寺くんが作成したプロトタイプの電源回路のもののため、そちらのほうがUSB HUBに関してはうまく認識できるように設計されていたという可能性。
 
2. **Jetson・J120側の問題の可能性**。これは前回認識の確認を行った際に使用したJetson・J120は167のものだった。こちらの不具合により、うまく認識がされなかった。

3. **JetsonのJetPackのバージョン違いによる影響**。前回の検証に使用した、JetPackのバージョンは**3.1**に対し、今回の検証で使用したJeTPackのバージョンは**4.3**である。このバージョン違いによる可能性。

が自分の考えた限りで思いつく可能性です。

1に関しての確認は自分ができないので、 @MasatoKuboteraまたは回路班の方に確認をお願いします。

それ以外の2, 3に関しては引き続き確認します。

hiroki-0001, 2021-10-08:
また本来の使用用途とはHUBの使い方が違うみたいなのがあれば、意見をお願いします。

MasatoKubotera, 2021-11-05:
171機体において同Ver2.0 MainBoardを使用しましたが，問題なくUSB-HUBが認識し，HUBを通して駆動するサーボモータの動作も確認できました．

MasatoKubotera, 2021-11-05:
171機ように作成した基板を167機（J120/TX2は163機のもの）に接続したが下図のように，FTR232Lが認識された．
そのため，前回検証した基板問題かと思われる．
[image]
[image]

MasatoKubotera, 2021-11-05:
現状，問題が解決されたためcloseする．

TakuyaNagamine, 2022-05-18:
## 220516調査結果
### 概要
・167機のRS-485変換基板が認識されない問題が発生したため調査を行った.
・USB-HUBは正常に動作していることを確認できたためUSB-HUBではなくRS-485変換基板側に問題があると考え,搭載されていたFT232RL-REEL,LTC1480CS8PBFを交換した.
### 結果
・交換後,lsusbコマンドを実行したところRS-485変換基板は認識されていたが,サーボモータを繋いでB3M_Managerを起動したところサーボモータは認識されなかった.
・以上の結果から,USB-HUBもしくはRS-485変換基板付近の部品に問題があると考えられるが,現在GanKenKunに使用されている部品の在庫不足などにより**167機は世界大会終了まで凍結とする.**

TakuyaNagamine, 2022-05-18:
今回発生した不具合は[167コックピット内基板修復#252](https://github.com/citbrains/SUSTAINA-OP/issues/252#)に類似していたため,そちらで調査報告を行う.よってこちらのissueはcloseとする.

closed 2022-05-18 by TakuyaNagamine