SUSTAINA-OP/SUSTAINA-OP #163: SideBumperの再設計

MasatoKubotera, 2021-09-24:


MasatoKubotera, 2021-09-24:
ケーブルの取り回しの関係で，Bumperの内側に格納した方が良いかもしれない．

aiba0921, 2021-10-02:
[image][image]
SideBumperにサーボケーブルを通せる穴を制作してみました。穴の大きさは7mm×13mmでサーボケーブルの端子より一回り大きくしてみました。

aiba0921, 2021-10-12:
@yasuohayashibara 
久保寺先輩と話し合ってver3.0で評価を行うことになりました。そのため、以前リアバンパーの評価で使用されたセンサーを使おうと考えているのですがお借りすることは可能でしょうか。

yasuohayashibara, 2021-10-15:
はい．
11号館3Fの準備室にあると思いますので，明日土曜日に貸し出します．

50Gと1000Gのセンサがありますが，転倒だと1000Gが良いですね．
ボディのどこかにガッチリ固定して，横に倒してバンパーがある場合と，無い場合で比較して下さい．
できればハイスピードカメラ（iPhoneなどは標準装備）で撮影して肩がどのように着地するのかを確認していただければと思います．
あまり，転倒させると壊れるかもしれないので，バンパ有り無しを２回づつぐらいで計測してもらえればと思います．
（メカニズムの担当の方に壊れないかを確認しながら進めて下さい．）

aiba0921, 2021-10-15:
ありがとうございます。
>あまり，転倒させると壊れるかもしれないので，バンパ有り無しを２回づつぐらいで計測してもらえればと思います．
（メカニズムの担当の方に壊れないかを確認しながら進めて下さい．）

了解です。おそらくメカ班の先輩と行うことになると思います。

MasatoKubotera, 2021-10-16:
@yasuohayashibara
対応ありがとうございます。

@aiba0921 
以前検証を行ったメンバがすでに卒業しているため、2年生以上も経験したことがありませんが一緒にやっていければと思います。

検証に使用する機体は、163機・165機など現状試合には出れない、RCAPで使用しない機体でhr/ready姿勢ができるものをで行っていきたいと思います。

yasuohayashibara, 2021-10-16:
加速度センサを計測用PCと共に貸し出しました．
一応ログを残しておきます．

MasatoKubotera, 2021-10-17:
加速度センサと計測用PC(windows7)の使いかについて、林原先生より教えていただきました。
[211016_加速度センサ使い方(動画)](https://drive.google.com/file/d/1pvrC_d0nVI82w570RhP0LoOBvXrFtGST/view?usp=sharing)

@aiba0921 
検証については自分が一緒に行うので希望日程などがあれば提示してください。

closed 2022-02-07 by MasatoKubotera