SUSTAINA-OP/SUSTAINA-OP #164: パウダーブレーキを用いたサーボ故障判定装置の作成

MasatoKubotera, 2021-09-25:


MasatoKubotera, 2021-09-25:
パウダーブレーキを使って，サーボモータのランク付けを行いたい．

KiyoshiroKawanabe, 2021-12-15:
世界大会の時に使用した装置を縦に使用するため，両脇に補助板をつける設計をした．
板は，はめ合で固定している．
[image]
[image]

[inventorのモデル](https://drive.google.com/drive/folders/1lv0lEeryHOShYUb555mTdLbCBs1bgvFy?usp=sharing)

KiyoshiroKawanabe, 2021-12-16:
いろいろ変更したので共有します．

yasuohayashibara, 2021-12-19:
気軽に図面が見れないため，上にアップした図に関して述べますが，斜交いを入れたほうが良いと思います．

KentaTsukino, 2021-12-24:
[image]
[image]

パウダーブレーキが完成しました

NaokiIwasawa, 2021-12-24:
# 最終形状について
設計は @KiyoshiroKawanabe にお願いしていたが、自分が一部変更をお願いし製作の過程でも一部応急手当をしたのでその詳細などを報告する。
設計データは[powder_brake](https://github.com/citbrains/OpenPlatform_ver3_0/tree/powder_brake) branchにある

## 設計の変更要求
先ず、林原先生から助言を頂いていたと知らなかったので筋交いを実装できず申し訳ありません。
自分が変更を要求したのは

- 2枚の板形状について四角形では板の使用面積が大きすぎると思ったので半分ぐらいの三角形へ
- パウダーブレーキ本体のはめ合いの隙間を0から0.1mmへ
- ネジの干渉が起きないかの確認と対応

## 製作過程での応急処置
大型の歯車とサーボ固定用ネジが通常のネジ頭ではどうしても干渉してしまうので皿ネジを利用できるように皿ザグリを追加工し皿ネジで固定した

MasatoKubotera, 2022-01-01:
装置の作成が終了しているためcloseしまう．
引き続き，装置を使った評価などは https://github.com/citbrains/OpenPlatform_ver3_0/issues/228 で行います

closed 2022-01-01 by MasatoKubotera

MasatoKubotera, 2022-03-01:
ブランチを整理するときに，コンフリクトが発生し統合できませんでした．
また，装置のモデルに使われているデータの著作権的にGitHubでの保管ができません．
そのため，CADモデルについては[Googleドライブ](https://drive.google.com/drive/folders/187Zg4EM0Fg6Qt1tToHC2n5ZOHi2IcwcX?usp=sharing)に移植しブランチを削除しました．