SUSTAINA-OP/SUSTAINA-OP #166: Ver4.0 設計変更に伴うルールベースの確認

MasatoKubotera, 2021-09-29:
ルール：[Virtual RoboCup Soccer Humanoid League Laws of the Game 2020/2021 April 15th, 2021](https://cdn.robocup.org/hl/wp/2021/04/V-HL21_Rules.pdf)

翻訳：[Google ドキュメント](https://docs.google.com/document/d/1Ms5g2x1dqoK6ML64X3x4QXYdfRD6G1bDnvPIPmgWV34/edit?usp=sharing)

MasatoKubotera, 2022-01-14:
http://humanoid.robocup.org/wp-content/uploads/RCHL-2020-Rules-Dec23.pdf

<blockquote>
# The Design of the Robots (new)

Robots participating in the Humanoid League competitions must have a human-like body plan, as shown in Fig. . They must consist of two legs, two arms, and one head, which are attached to a trunk.

(new:) Robots in KidSize must be equipped with a handle, to be picked up safely and with no harm to the robot and the handler.

[image]

Figure 3: Example of a humanoid robot body plan (left) and standing upright pose (right)

The robots must be able to stand upright on their feet and to walk on their legs. KidSize robots need to be able to recover from a fall (get back to a standing position). The only allowed modes of locomotion are bipedal walking, running and jumping.

All actions of the robots must be kinematically equivalent to humanoid motions.

Robots must be equipped with an emergency stop button that makes the robot immediately desist with all motions, or ideally go limp and/or cut power to the actuators. n addition to the emergency stop button, robots may only have up to two additional physical or virtual buttons: One to start the robot behaviour and one to stop the behaviour. The buttons must be clearly labeled. f the robot has more buttons that cannot be detached, they must be visibly masked during the games.

# Robot Height (new)

Based on Htop, the following size restrictions apply:

- 40 cm ≤ Htop ≤ 100 cm to play in the KidSize class,
- 100 cm ≤ Htop ≤ 200 cm to play in the AdultSize class.
 
Htop is defined as the height of the robot when standing upright (with fully extended knees, cf. Fig. right) and HCOM denotes the height of the robot's centre of mass, measured in upright posture. Htop is measured with the head of the robot oriented in such a way that it is tilted to either its maximum upwards tilt angle or the horizon line, whichever is lower.

# Weight Restrictions (new)
 

The robot's Body-Mass ndex (BM ) is defined as follows: BMI =	M 2 , where M is the mass of the robot in kg
top and Htop its height in meters. The following restriction applies:
- 5 ≤ BMI ≤ 30

# Size Restrictions (new)
 
All robots participating in the Humanoid League must comply with the following restrictions:

- Each foot must fit into a rectangle of area 1 (2.2 HCOM )2. A foot is defined as the minimum encapsulating rectangle covering all mechanical parts below the ankle joint. The encapsulating rectangle should be in a plane parallel to the bottom contact surface of the foot.
- The ratio between the longest and the shortest side of the encapsulating rectangle should be between 1.2 and 3.5
- The robot must fit into a cylinder of diameter 0.55 · Htop.
- The robot does not possess a configuration where it is extended longer than 1.5 Htop .
- The length of the legs Hleg, including the feet, satisfies 0. 5 Htop ≤ Hleg ≤ 0.7 Htop .
- The height of the head Hhead, including the neck, satisfies 0.1 · Htop ≤ Hhead ≤ 0.3 · Htop
- Hhead is defined as the vertical distance from the axis of the first arm joint at the shoulder to the top of the head.
- The leg length is measured while the robot is standing up straight. The length is measured from the first rotating joint where its axis lies in the plane parallel to the standing ground to the tip of the foot.
- The minimum length of the arm, measured from the first joint, is Htop − Hleg − Hhead.
</blockquote>

---

# ロボットのデザイン(新規)

ヒューマノイドリーグの競技に参加するロボットは、図に示すような人間に近いボディプランであることが求められます。2本の脚と2本の腕、そして1つの頭部が体幹に取り付けられていなければなりません。

(new:) KidSizeのロボットは、ロボットとハンドラーを傷つけることなく安全に抱き上げることができるよう、ハンドルが装備されている必要があります。

[image]

図3：ヒューマノイドロボットのボディプラン例（左）、直立ポーズ例（右）

ロボットは足で直立し、足で歩くことができることが必要です。キッズサイズのロボットには、転倒からの復帰（立ち位置への復帰）ができることが必要です。運動モードは、二足歩行、走行、ジャンプのみとします。

ロボットのすべての動作は、運動学的に人型の動作と等価でなければなりません。

ロボットは緊急停止ボタンを装備し、直ちに全ての動作を停止するか、理想的にはぐったりしてアクチュエータへの電力供給を停止させなければならない。1 つはロボットの動作を開始させるためのもの、もう 1 つは動作を停止させるためのもの。ボタンは明確にラベル付けされなければならない。ロボットが取り外すことのできないボタンをさらに持つ場合、ゲーム中は目に見えるようにマスキングされなければならない。

# ロボットの高さ (新規)

Htopを基準として、以下のサイズ制限を適用する。

- 40 cm ≤ Htop ≤ 100 cm であれば、キッズサイズクラスでプレーできます。
- 100cm ≦ Htop ≦ 200cmでAdultSizeクラスでプレイできます。
 
Htopは直立（膝を伸ばしきった状態、図右参照）したときのロボットの高さ、HCOMは直立姿勢で測定したロボットの重心の高さを示しています。Htopは、ロボットの頭を最大上方傾斜角か水平線のどちらか低い方に傾けた状態で測定されます。

# 重量制限（新規）
 

ロボットのボディマス指数（BM ）は以下のように定義される。BMI = M 2 , M はロボットの質量（kg）です。
top、身長（メートル）をHtopとする。以下の制限を適用する。
- 5 ≤ BMI ≤ 30

# サイズ制限（新規）
 
ヒューマノイドリーグに参加するロボットは、以下の制限を遵守する必要があります。

- 各足は、面積1（2.2HCOM）2の長方形に収まっていなければならない。足とは、足首の関節より下のすべての機械部分を覆う最小限の長方形と定義されます。内包する長方形は、足裏の接触面に平行な面でなければならない。
- 内包する長方形の長辺と短辺の比は、1.2～3.5であること。
- ロボットは直径0.55〜Htopの円筒に収まること。
- ロボットが1.5 Htopより長く伸びる構成を有しない。
- 足を含む脚の長さ Hleg は，0.5 Htop ≦ Hleg ≦ 0.7 Htop を満足すること。
- 首を含む頭部の高さ Hhead は 0.1 - Htop ≤ Hhead ≤ 0.3 - Htop を満たす。
- Hheadは、肩の第1腕関節の軸から頭頂部までの垂直距離と定義されます。
- 脚の長さは、ロボットがまっすぐ立っている状態で測定します。その軸が立っている地面と平行な平面上にある最初の回転ジョイントから足の先端までの長さを測定する。
- 腕の最小の長さは、最初の関節から測ると、Htop - Hleg - Hheadとなります。

Keita1Mori, 2022-01-15:
ルール説明[動画](https://www.youtube.com/watch?v=ZSOfX5TA5SI&t=111s)

Keita1Mori, 2022-01-19:
# 色
ガラエポの色を選定している際にロボットの色の指定があるとのことだったので追加で記載しておきます。
本文:Colours
 (new) Robots must be mostly black or of dark grey colour (i.e. RAL 7011 Iron Grey or darker) and non reflective. Robots may also be coloured in aluminimum-like silver, grey or white but then their feet must be coloured black. Any colour used for the field (green, white) or colours similar to the opponent team's team markers must be avoided. Arms, legs and bodies of the robot must be of solid shape appearance.
訳:カラー
 (新) ロボットは、大部分が黒色または濃い灰色（例：RAL 7011 Iron Grey またはそれ以上）で、反射しないものでなければならない。ロボットはアルミ調の銀色、灰色、白色でもよいが、その場合は足元は黒色でなければならない。フィールドに使用される色（緑、白）、または相手チームのチームカラーに類似した色。のマーカーは避けなければならない。ロボットの腕、脚、胴体は立体的な外観でなければならない。

MasatoKubotera, 2022-04-03:
サイズ制限
ロボットは直径0.55〜Htopの円筒に収まること。

について，腕を左右にわずかに広げる現状のHR姿勢では円筒に収まらないことがわかりました．
HR姿勢について見直す必要があるかもしれません．計測が脱力状態であれば問題ありません．

[image]

yasuohayashibara, 2022-04-03:
> サイズ制限
> ロボットは直径0.55〜Htopの円筒に収まること。

こちらは脇を締めた姿勢で構いません．

Keita1Mori, 2022-06-11:
設計変更によるルールベースの確認しました。
@MasatoKubotera 
# 2022/06/11のデータ（最新）
- Htop（身長）：650.608 mm
- Hleg（足の長さ）：315.264 mm
  - 最小：227.7128 mm
  - 最大：455.4256 mm
- Hhead（首を含めた頭の長さ）：108.708 mm
  - 最小：65.0608 mm
  - 最大：195.1824 mm
- ArmLength（腕の長さ）：327.500 mm
  - 最小：226.636 mm
- M（重量）：5031.7 g
- Hcom（重心位置）:323.5 mm
  - https://github.com/citbrains/SUSTAINA-OP/issues/169#issuecomment-1146883306
- BMI：11.8871007（kg/m）
  - 5<BMI<30
- ロボットが取れる動きの限度：975.912 mm
- 円柱の直径：357.8344 mm
- ゼッケン全体の面積：30130.99 mm^2
  - 25397.4462 mm^2 以上
- ゼッケン各方向の面積（前後左右）
- 前：7538.884 mm^2
- 後：8766.504 mm^2
- 左右：6912.801 mm^2
- 足裏の長辺と短辺の比率：175/80=2.1875
  - 1.2<x<3.5

# 懸念点
- ルールブック255行目にロボットに搭載できるスイッチの話がある。
どのスイッチがどのような役割をしているか明記する必要がある。
（例：エマスト、制御スイッチ）
SUSTAINAは二つ以上のスイッチを搭載しているため試合中隠す必要がある。
- サンドブラストしたところが少し白色すぎるかもしれないです。
  - ルールブックに提示されている理想の色：RAL 7011 Iron Greyもしくはこれより暗い色
- 一目でわかるようなゴールキーパーのゼッケンが必要
- 足裏の面積が足りない：14000mm^2
  - ルールブックによる数値計算の結果：15828.6528mm^2

MasatoKubotera, 2022-06-13:
@Keita1Mori 
>ルールブック255行目にロボットに搭載できるスイッチの話がある。
どのスイッチがどのような役割をしているか明記する必要がある。
（例：エマスト、制御スイッチ）

テプラで貼付して対応する予定です．
[image]

>SUSTAINAは二つ以上のスイッチを搭載しているため試合中隠す必要がある。

横側の電源ロッカースイッチ２個については，GankenKunの時から同設計なので見逃されると思っています．
見えるスイッチは，StartStopSwitchの２つです．

>サンドブラストしたところが少し白色すぎるかもしれないです。
ルールブックに提示されている理想の色：RAL 7011 Iron Greyもしくはこれより暗い色
https://github.com/citbrains/SUSTAINA-OP/issues/326　で対応しています．

>一目でわかるようなゴールキーパーのゼッケンが必要

検討中です．

>足裏の面積が足りない：14000mm^2
ルールブックによる数値計算の結果：15828.6528mm^2

足りないということは無いと思います．
ルール的に，「算出した規定面積に収まっている必要がある」と捉えています．
面積が大きい方が有利であり，算出値と同等にすることは困難だと思いますので翻訳ミスかと思いますが同でしょうか．

Keita1Mori, 2022-06-17:
@MasatoKubotera 
運営に確認したところ、足裏の面積は
1/32*(2.2*Hcom)^2より小さいことであることを確認しました。
なので足裏は現状問題ないです。

MasatoKubotera, 2022-08-08:
RoboCup2022にてルールに適応していることが証明されました．
下記，ロボット検査の様子です．
https://youtu.be/5BOK_6Y43pk
https://youtu.be/a25qXzwI7vU

ルールベースの確認が終了していますのでCloseします．

closed 2022-08-08 by MasatoKubotera