SUSTAINA-OP/SUSTAINA-OP #168: 脚サーボにヒートシンクを貼ってみる

MasatoKubotera, 2021-09-30:
以前の機体では脚など発熱するサーボ付近にファンが搭載されていたが、配線の必要性があるため、テープで簡単に貼り付けできるヒートシンクを導入してみる。

yasuohayashibara, 2021-09-30:
サーボモータの加熱が問題になるようであれば導入も検討したほうが良いと思いますが，そもそもは加熱しない領域で使用したほうが良いと思います．
モータを加熱させると減磁して最大トルクが下がります．
（ケースの前にモータのコアが加熱しますので，ケースが熱くなるというのはかなり危ない状態です．）

あと，おそらくメンテンナンスで取り付けない人が多くなるのと，外れたときに関節に噛み込んでモータのギアが欠けるおそれが生じます．
（実際に以前テープで止めていた部品が外れて，サーボモータを壊したことがありますので．．．）

参考ですが，GankenKun1.0で30分歩行させたときの温度が載っています．
https://docs.google.com/a/p.chibakoudai.jp/viewer?a=v&pid=sites&srcid=cC5jaGliYWtvdWRhaS5qcHxyb2JvdC1kZXNpZ24tYW5kLWNvbnRyb2wtbGFifGd4OjdmNmU3MGQ1YjRiNTU2ZjU

AD58-3104, 2021-10-06:
導入するかの議論とは別としてひとまずヒートシンクの効果を確認する為に実験を行う事を考えている。
20 * 20 * 15のヒートシンクを足首のサーボに1つずつ貼って、貼らない機体と温度を比較する。

hiroki-0001, 2022-02-24:
https://github.com/citbrains/citbrains_humanoid/issues/1613 でサーボモータに一定の負荷をかけ続けられることができる手法ができたので、外気温の条件等もなるべく揃えて、ヒートシンクあり場合となしの場合でどの程度温度上昇に変化があるかを調査していきたいと考えています。

MasatoKubotera, 2022-02-24:
>導入するかの議論とは別としてひとまずヒートシンクの効果を確認する為に実験を行う事を考えている。
20 * 20 * 15のヒートシンクを足首のサーボに1つずつ貼って、貼らない機体と温度を比較する。
現在161に貼り付け済みです。

ヒートシンクについて寸法しか記載がなく，型番が不明です．
今度導入となった時に型番が不明であると対応できないため，調査お願いします．

aiba0921, 2022-02-25:
@MasatoKubotera 
>20 * 20 * 15のヒートシンク

実際に161に張り付けてあるヒートシンクは20 * 20 * 10でした。寸法、色、材質的に該当すると思われるヒートシンクをアマゾンの下記リンクで発見しました。
https://www.amazon.co.jp/%E3%83%92%E3%83%BC%E3%83%88%E3%82%B7%E3%83%B3%E3%82%AF-%E7%86%B1%E4%BC%9D%E5%B0%8E%E6%80%A7%E4%B8%A1%E9%9D%A2%E3%83%86%E3%83%BC%E3%83%97%E4%BB%98%E3%81%8D-%E3%82%A2%E3%83%AB%E3%83%9F%E3%83%8B%E3%82%A6%E3%83%A0%E8%A3%BD-DIY%E3%82%AD%E3%83%83%E3%83%88-%E3%83%8F%E3%82%A4%E3%83%91%E3%83%AF%E3%83%BCLED%E3%82%A2%E3%83%B3%E3%83%97%E3%81%AB%E9%81%A9%E7%94%A8/dp/B08BKLX1DC/ref=sr_1_18_sspa?crid=10Q2I8DV8FCHA&keywords=%E3%83%92%E3%83%BC%E3%83%88%E3%82%B7%E3%83%B3%E3%82%AF+%E5%B0%8F%E5%9E%8B&qid=1645751356&sprefix=%E3%83%92%E3%83%BC%E3%83%88%E3%82%B7%E3%83%B3%E3%82%AF+%2Caps%2C185&sr=8-18-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExSFdXQThLWTUzTFVXJmVuY3J5cHRlZElkPUEwOTE4OTk2MVUzM1NZNEpHSVdIViZlbmNyeXB0ZWRBZElkPUEyOTNVVFhRMjFHSTJVJndpZGdldE5hbWU9c3BfbXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

aiba0921, 2022-02-25:
## 検証
実施日　２月２５日
## 参加者
@aiba0921 @KentaTsukino 
## 内容
ヒートシンクの効果を検証するためサーボにヒートシンクを貼った状態と貼っていない状態で行った。場所による変化が出るのかを確認するため同じサーボで貼る場所変えて計測を行った。計測で得られた値は[スプレットシート](https://docs.google.com/spreadsheets/d/1j8BY59EuR6Zf7hNudm2JH3eIATxWJ0ZC-HnG-4wTitY/edit#gid=0)にまとめた。
今回計測したサーボ
70-2108
70-2113
70-2114
掛かった時間：６時間
＊70-2113は貼る場所を変えて貼っていない状態と合計して３回検証を行った。
##結果
貼っていない状態のmotorが13℃、mcuが10℃上昇したのに対して貼った状態ではmotorが平均18℃、mcuが平均16℃上昇しておりあまり効果は見られなかった。
## 考察
ヒートシンクは本来cpuに直付けする用途が多いため外装に貼った状態では十分な効果を発揮できなかった可能性が考えられる。加えて70-2113は検証ごとに30分ファンで冷却していたとはいえ連続で使用していたため温度が上昇しやすくなっていた可能性もある。

AD58-3104, 2022-02-28:
@aiba0921 反応が遅れて申し訳ありません。実験に使用しているヒートシンクは上で貼って頂いたurlのもので間違いないです。

hiroki-0001, 2022-04-10:
こちらある程度余力ができたので再開したいと思います。
前回までのヒートシンクの効果の検証ですが問題点が2つほどありました

<1つ目の問題点>

[前回の計測時](https://github.com/citbrains/SUSTAINA-OP/issues/168#issuecomment-1050609761)とhttps://github.com/citbrains/SUSTAINA-OP/issues/168 で貼られているヒートシンクの場所が異なる

<2つ目の問題点>

[前回の計測時](https://github.com/citbrains/SUSTAINA-OP/issues/168#issuecomment-1050609761)ではヒートシンクによる冷却効率について検証されていない。

---

これらのことから、再度検証を行う際には適切な位置にヒートシンクを貼り付け、温度上昇したB3Mがある一定の温度まで温度が低下するまでにかかる時間を調査したほうが良さそうです。

yasuohayashibara, 2022-04-10:
まず，時間がそれほどかからないのであれば，こちら調査するのは良いと思います．

一方で，野口君が調べてくれたように最大トルクが低下しているモータが少ないということは，熱に関しては現状の使い方で問題無さそうということを示しています．
（むしろガタに関しては対策したほうが良いと思いますが．．．）

今後，より大きなトルクを発生しなければならない状況になるのであればヒートシンクをつけたほうが良いですが，そうでないとしたらメカニズムの複雑さは増やさない方が良いかなと思います．
再度になりますが，知見としてヒートシンクの効果を試してみるというのは良いと思います．

hiroki-0001, 2022-06-21:
こちらサーボモータの発熱が最近目立つので調査を行います。

ヒートシンクの貼る位置に関してですが久保寺くんと相談したところ、B3M內部のモータの搭載位置
(以下の画像は各B3Mのモータの取り付け位置を示す　この上にボトムケースがハマる)

[image] 

から考えるにモータからの熱伝導率が1番高そうなのは

**1. 型番が記載されたシールがあるサイド**

[image] 

**2. 1とは逆側のサイド**

[image] 

の2通りだと考えました。

またこの部分であれば、足付け根ロール以外の部位にヒートシンクを貼り付けすることがおそらくできそうです。

この部位にヒートシンクを貼り付ける方針でヒートシンクの効果について調査していきます。

yasuohayashibara, 2022-06-21:
発熱が問題になっているとしたら，まずはIゲインを０にするのと，歩行制御でリラックスモードを復活させるのが良いかもしれません．
そもそも発熱させないようにしたほうが良いと思います．

MasatoKubotera, 2022-08-23:
発熱に関して，検証前後の温度ログを取得していました．
https://github.com/citbrains/citbrains_various_log_data

RoboCup2022直前まで，発熱に関して観察していましたがヒートシンクを貼るほどではないという結論に至りました．

貼付したことによる定評的な評価はできていませんが，貼らなくてもRoboCup2022では問題になっていませんので一旦Closeします．

closed 2022-08-23 by MasatoKubotera