SUSTAINA-OP/SUSTAINA-OP #169: 重心位置の確認

yasuohayashibara, 2021-09-30:
重心位置ですが，直立時の脚のピッチ軸の上にあることが望まれます．
あまり議論されていないようですので，問題提起します．
背中が大きくなっているようですので，もしかしたら重心位置が後ろすぎるかもしれません．

重心が後ろにあると，直立しているだけでモータがトルクを出さなければならなくなり，さらに前に急加速すると後ろに転倒するおそれが高くなります．

yasuohayashibara, 2021-09-30:
参考　GankenKun ver.3.0の重心

[image]

MasatoKubotera, 2021-10-01:
クロスベアリングが反映された コミット [875d7ef281c1715ebcb90a73c6045eb17addf223](https://github.com/citbrains/OpenPlatform_ver3_0/tree/875d7ef281c1715ebcb90a73c6045eb17addf223)時点の重心位置を確認しました．

[image]「全身」
[image]「上半身の重心」

Ver4.0からJ120のUSBコネクタを保護するために，コックピットサイズが後方に10mmほど拡張されましが，バッテリの位置が股ヨー軸サーボ間に移動になったため，重心位置はver3.0と比較すると，前後方向にはほとんど変わらず，下方に10mm程度ずれています．
今回のモデルより，Jetsonやネジ等も含めてアセンブリを作成しているためver3.0よりもより正確なモデルを作成できるかと思います．

参考：[重心を表示する](https://knowledge.autodesk.com/ja/support/inventor-products/learn-explore/caas/CloudHelp/cloudhelp/2015/JPN/Inventor-Help/files/GUID-E8FFE71F-3F0D-4F9F-AAB7-2341B8BCAA6D-htm.html)

MasatoKubotera, 2021-10-08:
重心による足裏の制限について，Ver4.0では重心位置が異なっていますので，足裏の面積を変更する必要があります．

ルールベースにのっとっているかについては下記のスプレッドシートで確認をしていますが，現状モデルがコックピット内部が再現できていなく，バンパーも反映していない状態ですのでかなり暫定ですが，足裏の幅・長さをどちらか2.5mm程度縮める必要があります．

[211002_ルールベース確認](https://docs.google.com/spreadsheets/d/1HeOWMD17ResPVHYQDU05iXUUVeBqu2YhF_EmXf6jLDU/edit?usp=sharing)

yasuohayashibara, 2021-10-09:
力学的に重心の高さの変更が及ぼすであろう影響を示します．
林原が考えたものですので，何か間違いがあるかもしれません．
気になっていたのですが，分析が遅くなり申し訳ありません．

今回の変更は軽微ですので，それほど大きな影響は無いと思いますが，どのように影響するかは知っておいたほうが良いと思いますので，以下に示します．

- 横方向の加速度に関する影響
重心を低くすると，横方向により多く加速しないと姿勢が維持できなくなります．
ただし，陸上のスタートに見られるように，重心を低くすると加速はしやすくなります．
加速時に横方向に地面を蹴りますので，滑りやすくなるかもしれません．

[image]

図　重心の高さが及ぼす影響　（左：重心低，右：重心高）

- 股ロール軸に関する影響
地面から受ける力が大きくなるということもありますが，力までの距離が大きくなりますので，より大きな力が股ロール関節に加わります．
関節に加わるトルク　＝　地面から受ける力　ｘ　その力と軸との距離
股ロール軸の出力角のズレは歩行の安定性に直結しますので，従来以上に剛性を高める必要があるかもしれません．

[image]

図　重心の高さが及ぼす影響　（左：重心低，右：重心高）

参考までに，以下はHPR2になりますが，おそらくこのトルクを減らすためだと思いますが，股ロール軸が内側にあります．
[image]

軽微だとは思いますが，以上力学の観点から重心により影響を受けそうな内容を記しておきます．

yasuohayashibara, 2021-12-13:
現在ver. 4.1の設計を始めていると思いますが，ver. 4の評価を並行して行ったほうが良いかもしれないと考えています．

RCAP中にシュートモーションが作りづらいという話もありましたが，力学的な観点から，重心を下げると【より体を傾けないと】片足で立つことができませんので，その点が少し気になっています．（上図参照）

重量も増加していますので，モータへの負担の増大と，モーメント力の増加によるたわみで姿勢を維持することが難しくなっているかもしれません．
（時間があったら，私もモータに要求されるトルクがどの程度変化しているかを検証したいとは思いますが．．．）

試合などの検証も含めて，できればver.4を評価しながら次の設計を行うと間違いが無いと思います．

hiroki-0001, 2021-12-14:
図にするとこんな感じです。図のような姿勢(実際にはもっと上体は後ろにそらすが)にすると青の矢印の方向に機体が倒れていきます。
[image]

yasuohayashibara, 2021-12-14:
@hiroki-0001 
なるほど．
ピッチ方向に関しては重心位置はおそらく低いほうが良いので，重量増の影響が大きいかもしれません．

hiroki-0001, 2021-12-16:
RCAP中に起きた、キックモーション中に機体上半身が後ろに倒れてしまう現象を再度再現しようとしたところ。そのような現象は起きませんでした。

- 機体
- モーター
- 安定化電源(12V 12.5A)

上記の3つの要素はRCAP期間中と同様のものにしました。

このことから、キックモーション中に上半身がゆっくり後ろに倒れていく現象は

- ただの勘違い
- 長期間サーボを使用したことによるサーボの熱の影響によるトルク減少

の2つが原因であるような気がします。

MasatoKubotera, 2021-12-16:
重量について

GankenKun本体を測れる計量器が現状体重計しかなく（工作室に新規購入されるが現状未入荷），±100gの誤差はありますが，現状運用しているVer3の165機とVer4.1の171機はいずれも4.6kg(バッテリー無)でした．

※工作室の計量器が到着次第再度計測します．
※バッテリーの重量は，220gです．

コックピットの重量についてはVer3（RCAP仕様）で669g，Ver4.0（171機仕様）で636gで一番重くなっていると予想されていた部分は重くなっていませんでした．
[image]
「Ver 3」
[image]
「Ver 4」

これにより，増量した分は，ほとんどクロスローラー周りだということがわかりました．
胸以上も部位は，スラストベアリングを撤廃したので実施的には軽くなっているかと思います．

さらなる軽量化も行いますが，クロスローラー周りの軽量化はかなり限界があります．
そのためバッテリーの見直し等も必要になるかもしれません．
後継バッテリーの検討 #220
https://github.com/citbrains/OpenPlatform_ver3_0/issues/220

>足首ピッチのサーボに加わる負荷が強すぎて、上体を支えきれずにゆっくり倒れてしまいます。

について，サーボ本体の影響である可能性も考えられます．
この部位に使われているモータですが，管理シートによると新品ではなく中古モータ（修理済み品）が使われていました．
https://docs.google.com/spreadsheets/d/1bP7htNyyIz2KI1jB_ZaqXPhJWibj-PJ-OF8uCFB9jfw/edit#gid=824152206

2 (右足首ピッチ)	70-1981
11 (左足首ピッチ)	70-1977

修理内容
70-1977　ファイナルギヤ、3rdギヤ
70-1981　モータ

モータによる影響なのか明確にするために，パウダーブレーキを用いた実験装置を作成するissueを再始動しました．担当は　@KiyoshiroKawanabe です．

パウダーブレーキを用いたサーボ故障判定装置の作成 #164
https://github.com/citbrains/OpenPlatform_ver3_0/issues/164

また，電力不足ということも考えられます．
片足（つけ根ロール，膝上ピッチ，膝下ピッチ，足首ピッチ，足首ロール）には5個のモータが1つのXHコネクタから給電しています．
そもそものXHコネクタの定格は3Aであり，使用しているケーブルはAWG22であり，最大許容電流は７Aになります．
足首系がケーブルの末端になるので，電力が足りてない恐れもあります．
末端から電力の分配基板に電力線をループさせること，一つの線から並列に接続するのではなく，分配基板からそれぞれ接続するなどした方が安定するかと思います．

一旦，キック時のサーボモータの電流を計測してみたいと思います．

yasuohayashibara, 2021-12-19:
> 足首系がケーブルの末端になるので，電力が足りてない恐れもあります．

過去に議論になったことがあったような気がしますが，そのときには問題無しということになりました．
負荷をかけたときの電圧を計測してみると良いかもしれません．

B3Mマニュアル　P.7
『直列で 5 個以上接続する場合は、こちらからも External Power を入力してください。』
https://kondo-robot.com/w/wp-content/uploads/2014-B3M_series_manual_Ver.1.1.3.pdf

hiroki-0001, 2022-03-09:
重心の高さを計算しました。計測時の様子は以下の動画から見れます。
https://drive.google.com/file/d/1uBeiAtYpaky8DOhhzZ7ZsaN7zCfTdeEe/view?usp=sharing

### 計測方法

1. L字の角パイプと平らな木の板を使用し、最初に木の板の重心を計測する
2. 計測した木の板の重心と角パイプのL字部分を接点として角パイプの上に木の板を乗せる
3. 板の上に直立姿勢状態のロボットを乗せる
4. 板の上のロボットを移動し、木の板の傾きが水平になるまでロボットを移動させる
5. 傾きが水平になったら、木の板の重心からロボットの足先までの距離を計測する

上記の手順で重心を計測した。

参考にした動画(あとで載せておきます。)

**計測結果は足から320mm離れたところが重心の高さになります**

MasatoKubotera, 2022-05-01:
>Dear everyone,
The rule document has been updated in order to clarify how the Center of Mass will be officially measured. It now includes an appendix presenting how to build the official device and the procedure to follow for measuring Hcom.
You can check the latest version on the official website: https://humanoid.robocup.org/materials/rules/

>皆様へ
重心位置の公式測定方法を明確にするため、ルールドキュメントを更新しました。また、公式装置の作り方やHcomの測定手順などを付録として追加しました。
最新版は、公式サイト https://humanoid.robocup.org/materials/rules/ でご確認いただけます。

Current Rule Documents: [2022 Final Version](http://humanoid.robocup.org/wp-content/uploads/RC-HL-2022-Rules-1.pdf) ([with changes marked](http://humanoid.robocup.org/wp-content/uploads/RC-HL-2022-Rules-Changes-Marked-1.pdf))

MasatoKubotera, 2022-05-01:
http://humanoid.robocup.org/wp-content/uploads/RC-HL-2022-Rules-Changes-Marked-1.pdf
<blockquote>

**Center of mass measurement**

This sections presents the offcial procedure to measure HCOM , the height of the center of mass of the robot used in Law 4. It also provides instructions on how to build the measuring device used in the procedure.

**Construction of the measuring device**
The dimensions of the device are different for KidSize and AdultSize. In this section, Hmax denotes the maximum height allowed for the according league (see Law 4). W denotes the width of the measuring device (60cm for KidSize, 115cm for AdultSize).

**The required materials are:**
- 1 measuring tape of length Hmax
- An aluminium plate (Hmax + 5cm by W − 6cm)
- A wooden board - Plywood with:
  - Length: Hmax + 20cm for KidSize, Hmax + 40cm for AdultSize
  - Width: W
  - Thickness: 1.5cm for KidSize, 3cm for AdultSize
- Two acrylic plates with: the same length as the wooden board, 3cm width and 2-3mm thickness.
- 4 Aluminium pipe straps of same size

**Steps for building the measuring device**
1. Glue two acrylic plate on each side of the wooden board as shown in Fig 8a.
2. Draw a white line on the middle of the board, over the acrylic plates.
3. Screw 4 aluminimum pipe straps at the back-middle of the wooden board, aligning horizontally as shown in Fig 8b. This is to determine whether the robot weight is balanced.
4. Bend 4cm of the aluminimum plate to 90 degrees and tape the Hmax measuring tape as shown in Fig 8c.

**Measuring the center of mass**
1. Place the robot flat onto the aluminium plate holding, touching the bottom of the bended aluminium plate.
2. Ensure that the robot is in an upright pose14 as shown in Fig 9a.
3. Align the metal frame holding with the wooden board as shown in Fig 9b.
4. Slowly, slide the aluminium plate to towards the other end until the wooden board is balanced as shown in Fig 9c.
5. Record the reading shown by the white line in the middle of the wooden board pointing to the measuring tape, as shown in Fig 9d.

[image]

</blockquote>

MasatoKubotera, 2022-05-01:
<blockquote>

**重心位置の測定**
ここでは、法則4で使用したロボットの重心高さHCOMを測定する正式な手順を紹介します。また、その手順で使用する測定装置の製作方法を説明する。

**測定装置の製作**
測定装置の寸法は、KidSizeとAdultSizeで異なる。このセクションで、Hmaxは該当するリーグで許容される最大身長を示す（法4参照）。Wは、測定装置の幅を表します（KidSizeは60cm、AdultSizeは115cm）。

**必要な材料は以下の通り。**
長さHmaxのメジャー1本
アルミニウム板 (Hmax + 5cm x W - 6cm)
木の板 - ベニヤ板。
長さ：Hmax + 20cm for KidSize、Hmax + 40cm for AdultSize
幅：W
厚み：キッズサイズ1.5cm、アダルトサイズ3cm
アクリル板2枚：木の板と同じ長さ、幅3cm、厚さ2～3mm。
同じ大きさのアルミパイプストラップ4本

**測定器の作り方**
図 8a のように、木の板の両面にアクリル板を 2 枚ずつ接着剤で貼り付けます。
板の中央、アクリル板の上に白線を引きます。
図8bのように、木製ボードの背面中央部にアルミ製パイプストラップ4本を水平になるようにネジ止めします。これは、ロボットの重量のバランスが取れているかどうかを判断するためです。
アルミナム板の4cmを90度に曲げ、Fig.8cのようにHmaxの測定テープを貼る。

**質量中心を測定する**
ロボットをアルミ板押さえの上に平らに置き、曲げたアルミ板の底に接触させる。
図 9a に示すように、ロボットが直立した状態であることを確認します14。
図 9b に示すように、金属フレームを木製ボードに合わせます。
図9cに示すように、木の板のバランスが取れるまで、アルミ板をもう一方の端にゆっくりとスライドさせます。
図 9d に示すように、木製の板の中央部にある白線で示された測定値を、メジャーに記録します。

</blockquote>

MasatoKubotera, 2022-05-01:
prototypeにて https://github.com/citbrains/SUSTAINA-OP/issues/169#issuecomment-1062652355 で提示している方法で重心を計測したが，本番を想定してルールで詳細に提示された方法で計測したいと考える．

MasatoKubotera, 2022-05-01:
SUSTAINA(2022/5/1)のHmaxは643.863 mmです．

[image]

Hmaxを645mmとして，ルールブックで指定されている必要な材料は下記の通りです．

- メジャー ×1 (例はIKEAでもらえる紙製のメジャーのよう)
- アルミニウム板 695 x 540mm
- べニア板 845 x 600 x 15mm
- アクリル板 845 x 30 x 2~3 mm
- 同じ大きさのアルミパイプストラップ4本 [SANEI 配管部品 ステンレスサドルバンド パイプ固定用 内径25mm 5個入り R61-5S-20](https://www.amazon.co.jp/SANEI-%E3%80%90%E3%82%B9%E3%83%86%E3%83%B3%E3%83%AC%E3%82%B9%E3%82%B5%E3%83%89%E3%83%AB%E3%83%90%E3%83%B3%E3%83%89%E3%80%91-%E5%86%85%E5%BE%8425%E3%83%BB%E7%A9%B4%E5%BE%844mm-5%E5%80%8B%E5%85%A5%E3%82%8A-R61-5S-20/dp/B003PHSLKQ/ref=sr_1_5?keywords=%E3%83%91%E3%82%A4%E3%83%97%E5%9B%BA%E5%AE%9A%E9%87%91%E5%85%B7&qid=1651394683&sr=8-5)

MibuchiYuta, 2022-05-13:
[image][image]

MibuchiYuta, 2022-05-29:
[image]
[image]
[image]

MibuchiYuta, 2022-05-29:
@MasatoKubotera 
レシートです
[image]

MibuchiYuta, 2022-06-05:
[YouTube](https://youtu.be/vkD2AYEDcGg) 重心計測を行いました。　計測の結果、
323.5mm 

となりました。

@MasatoKubotera @mas

MasatoKubotera, 2022-06-30:
重心計測の方法に問題がありましたのでreopenします．
@MibuchiYuta @hiroki-0001 共有してください．

MibuchiYuta, 2022-07-01:
## 重心計測時のロボットの姿勢

> Laying robot upright on measurement device.

とルールブックに記載されており， @Keita1Mori から仰向けではないかという指摘が入りました．
こちらの "upright on 〜"の記述に関して，"直立させる" という意味が強いですが，現在その真偽を確認中です．

Keita1Mori, 2022-07-01:
@MibuchiYuta 
こちらに関してですが直立姿勢であっていると思います。今運営側にうつ伏せか仰向けかどちらでも計測するのか確認をとっているところです。

Keita1Mori, 2022-07-01:
確認したところ以下の返信内容が帰ってきた。
英文
I think in this case the configuration should be chosen that keeps the robot as parallel as possible to device
訳
この場合、ロボットができるだけ装置と平行になるような構成を選ぶべきだと思います。

hiroki-0001, 2022-07-07:
忘れないうちにメモします(後でしっかり書きます)

前回の計測と比べ変わった点

- 肩バンパー
- 足首ユニットのスペーサー(重量が軽くなった)
- 持ち手部分のスペーサー(重量が重くなった)

うつ伏せでの計測を行った結果

(足裏のスペーサーの高さは13mmを使用)

おもりなし

330mm

おもりあり(片方の肩で40gずつ、両肩合わせて80g)

333mm

＊この計測に関しては、メインボードにファンが乗っていない機体で行ったため、変動はあるかもしれません

MasatoKubotera, 2022-08-23:
実際のロボット検査の様子です．
[220712 day2 RobotCheck CIT Brains Fixed](https://youtu.be/a25qXzwI7vU)
[220712 day2 RobotCheck CIT Brains Move](https://youtu.be/5BOK_6Y43pk)

事前準備していた，肩おもりについてはRoboCup2022では一切使用しませんでした．

RoboCup2022で現地ボランティア＋レフリーが計測した結果は，動画内でも発言がありますが，**349mm**でした．
ちなみに検査時は，スパイク長が13ｍｍの時です．（期間中に歩行制御・脚への負荷軽減の要因により9mmに変更している．）

MasatoKubotera, 2022-08-23:
重心位置の計測についてRoBoCup2022でのレポートも記載しましたのでCloseします．

closed 2022-08-23 by MasatoKubotera