SUSTAINA-OP/SUSTAINA-OP #172: 板状アンテナの導入検討

MasatoKubotera, 2021-10-05:
現状使用している，[変換名人 無線LAN 対応SMAアンテナ ANT-SMA1](https://www.amazon.co.jp/gp/product/B004UQA7DU/ref=as_li_ss_tl?ie=UTF8&camp=247&creative=7399&creativeASIN=B004UQA7DU&linkCode=as2&tag=0002598481-22)が後方+-135度に首が回転したときにカメラに映り込み認識に影響が出ている．
これを最小限に抑えるために省スペースアンテナである板状アンテナの導入を検討する．

yasuohayashibara, 2021-10-06:
以前板状アンテナを導入したことがあり，電波を受信できなくて今のアンテナに変えた経緯があります．
おそらく金属に貼って使用することになると思いますが，電波の専門家からあまり良くないという指摘を受けました．
プラスチックに貼るのはOKと記憶しています．

ちなみに，今のアンテナの固定部は樹脂ですよね？
以前金属で固定したときに問題が発生していました．
通信はいつも問題になるところですので，受信しやすい状態を保ったほうが良いと思います．

MasatoKubotera, 2021-10-09:
>ちなみに，今のアンテナの固定部は樹脂ですよね？

固定部は，コックピットになりますのでPOM材になります．

---
現状使用している，アンテナは[変換名人 無線LAN 対応SMAアンテナ ANT-SMA1](https://www.amazon.co.jp/gp/product/B004UQA7DU/ref=as_li_ss_tl?ie=UTF8&camp=247&creative=7399&creativeASIN=B004UQA7DU&linkCode=as2&tag=0002598481-22)，[変換名人 内蔵アンテナ外付ケーブル ANT-CNB](https://www.amazon.co.jp/gp/product/B004UQA6V8/ref=as_li_ss_tl?ie=UTF8&camp=247&creative=7399&creativeASIN=B004UQA6V8&linkCode=as2&tag=0002598481-22)です．

---

アンテナについては故障（断線）頻度が高くなっているのが現状です．
理由としては，コックピット外に出ていることでコックピットを取り出す際に指などに引っ掛かり断線するというパターンが多いように思えます．
また，長さ的にハンドルを根元のねじ４本を外さなければハンドルに引っ掛かるという欠点もあり，せっかくの脱着が容易であるというコックピットの利点を損なってしまっているのもあります．

そのことを踏まえて，コックピット内部に収まる板状のアンテナ（例：[高性能5dBi 2.4GHz/5GHz 802.11a/b/g/n/ac対応 無線LANカード WIFI/Wimax/Bluetoothモジュール用アンテナ 2本セット (U.FL Cable=240mm)](https://www.amazon.co.jp/%E9%AB%98%E6%80%A7%E8%83%BD5dBi-2-4GHz-802-11a-%E7%84%A1%E7%B7%9ALAN%E3%82%AB%E3%83%BC%E3%83%89-Bluetooth%E3%83%A2%E3%82%B8%E3%83%A5%E3%83%BC%E3%83%AB%E7%94%A8%E3%82%A2%E3%83%B3%E3%83%86%E3%83%8A/dp/B06ZZC4WQG/ref=pd_lpo_2?pd_rd_i=B06ZZC4WQG&psc=1)）に変更することはメリットが大きいと考えますが，
>以前板状アンテナを導入したことがあり，電波を受信できなくて今のアンテナに変えた経緯があります．
おそらく金属に貼って使用することになると思いますが，電波の専門家からあまり良くないという指摘を受けました．
プラスチックに貼るのはOKと記憶しています．

を踏まえて慎重に検討したいと考えています．

[image]

MasatoKubotera, 2021-10-31:
@RyodaiArai 
検証機体の161機について，板状アンテナが１セットしか購入していなく171機に使用するため取り外し，棒状アンテナに戻します．

MasatoKubotera, 2021-10-31:
171機に搭載後に再度検証を行いたいと考えています．

MasatoKubotera, 2021-11-04:
Ver4.0コックピットには，板状アンテナを下図のように貼付しています．
検証の時と比較すると，場所・向きが変わっています．
[image]

MasatoKubotera, 2021-12-12:
１７１機体に搭載し，RCAPを通して検証を行ってきたが従来のWi-Fiアンテナと同様に使用できているため，今後すべての機体でこのアンテナを使用していきたいと思います．

closed 2021-12-23 by MasatoKubotera